openscad/openscad #3708: 2021.01 not working on macos: 11.2.2

stepheneb, 2021-03-07:
I recently upgraded from the last version of macos 10.x to Big Sur 11.2.2 and OpenScad v2021.01 which had been working fine seems to be broken.

Opening examples or choosing new results in a blank pane on the left. Looks like one problem is that the top of the window is positioned above the top of the window.

[image]

stepheneb, 2021-03-07:
And this morning it appears to be working -- can't think of any changes ???

[image]

closed 2021-03-07 by stepheneb

t-paul, 2021-03-07:
Thanks, good news. Without an active macOS developer we would have a difficult time tracking down issues specific to macOS.