openscad/openscad #3709: Build process cannot find lib3mf library (vv. 2.1.1 and 2.0.1)

GvMariani, 2021-03-07:
I'm trying to build openscad 2021.01 for Rosa Fresh Linux (a Mandriva derivative) on x86_64; on the system is installed the latest lib3mf from https://github.com/3MFConsortium/lib3mf (2.1.1), with the relevant devel files: headers in /usr/include/lib3mf, libs in /usr/lib64 and pkconfig file in /usr/lib64/pkgconfig. 

At cmake configure stage this is the console output:
********************************************************
Info: creating stash file /tmp/abf/rpmbuild/BUILD/openscad-2021.01/.qmake.stash
Project MESSAGE: If you're building a development binary, consider adding CONFIG+=experimental
Project MESSAGE: Using ccache
Project MESSAGE: Using QOpenGLWidget
Project MESSAGE: Using C++14
Project MESSAGE: cairo enabled
Package lib3MF was not found in the pkg-config search path.
Perhaps you should add the directory containing `lib3MF.pc'
to the PKG_CONFIG_PATH environment variable
No package 'lib3MF' found
Package lib3MF was not found in the pkg-config search path.
Perhaps you should add the directory containing `lib3MF.pc'
to the PKG_CONFIG_PATH environment variable
No package 'lib3MF' found
Package lib3MF was not found in the pkg-config search path.
Perhaps you should add the directory containing `lib3MF.pc'
to the PKG_CONFIG_PATH environment variable
No package 'lib3MF' found
Package lib3MF was not found in the pkg-config search path.
Perhaps you should add the directory containing `lib3MF.pc'
to the PKG_CONFIG_PATH environment variable
No package 'lib3MF' found
Project MESSAGE: 3MF Import/Export disabled
********************************************************

So, *first issue*: since 27/08/2020 (this means that any 3MF library release >= 2.0.1 should be affected by this issue) they changed the name of the .pc file for the 3MF library, renaming it from "lib3MF.pc" to "lib3mf.pc"; now the snippet "find_package(Lib3MF REQUIRED QUIET)"  (line 260 of the main CMakeLists.txt file) does not work anymore as intended, because it depends upon FindLib3MF.cmake and this in turn do "pkg_check_modules(LIB3MF lib3MF)" and looks for a now-non-existent lib3MF.pc file.

I tried to work around the issue by creating a symlink "lib3MF.pc" pointing to the real "lib3mf.pc" file, but without success; the console output became:
***************************************
Info: creating stash file /tmp/abf/rpmbuild/BUILD/openscad-2021.01/.qmake.stash
Project MESSAGE: If you're building a development binary, consider adding CONFIG+=experimental
Project MESSAGE: Using ccache
Project MESSAGE: Using QOpenGLWidget
Project MESSAGE: Using C++14
Project MESSAGE: cairo enabled
Project MESSAGE: 3MF Import/Export disabled
**************************************
... and still no 3MF support.

Commit 0d204d9e60513668358e84f43e64d247f0061d1b is no more enough, I fear...

t-paul, 2021-03-07:
Yes, the mentioned commit only introduces support for the 2.x API, there's no official build support for that as none of the environment have the newer library yet.

For the `qmake` build there will be no updates anymore, the dev snapshots are migrating to `cmake` so we can leave the issue open till the new library packages are available.

GvMariani, 2021-03-08:
> Yes, the mentioned commit only introduces support for the 2.x API, there's no official build support for that as none of the environment have the newer library yet.
AFAICT Fedora Rawhide have 2.00 and Gentoo have 2.1.0...

> the dev snapshots are migrating to cmake
While not being a cmake expert at all, I tried to use the snipped below in a Findlib3mf.cmake file, but no success... perhaps can be useful, though.
***********************************
# ----------
#
# FindModule for the 3mf library.
#
# This module can be used to detect lib3MF (releases < 2.0.0) or lib3mf (releases
# >= 2.0.0). It works by invoking pkgconfig and taking in account for the changes
# of library name between releases < 2.0.0 and >= 2.0.0.
#
# This module will define:
# LIB3MF_FOUND				Set to 1 the library exist and is usable
# LIB3MF_LIBRARIES			List only the libraries to be linked (without the '-l')
# LIB3MF_LINK_LIBRARIES		List the same libraries, with their absolute paths
# LIB3MF_LIBRARY_DIRS		List the paths where the libraries will be searched (without the '-L')
# LIB3MF_LDFLAGS			List all the required linker flags
# LIB3MF_LDFLAGS_OTHER		List all other linker flags
# LIB3MF_INCLUDE_DIRS		List the include flags (without the '-I')
# LIB3MF_CFLAGS				List all the required cflags
# LIB3MF_CFLAGS_OTHER		List the other compiler flags
# LIB3MF_VERSION			Contains the version of the library
# LIB3MF_PREFIX				Contains the "prefix" directory of the library
# LIB3MF_INCLUDEDIR			List the include directory for the libraries
# LIB3MF_LIBDIR				List the directory where to find the libraries
################################################################################# 

# The library releases >= 2.0.0 don't include anymore the COM interface: use this
# fact to find if we have a such library and adjust it's name accordingly.
# Suppose 2.x.x (name lowercase) by default
set(LIB3MF_LIB "lib3mf")
find_path(INCLUDEDIR NAMES NMR_DLLInterfaces.h PATHS /usr/include/lib3mf/Model/COM NO_DEFAULT_PATH)
if(NOT ${INCLUDEDIR} STREQUAL "")
	# 1.x.x library (name uppercase)
	set(LIB3MF_LIB "lib3MF")
endif

message(STATUS "Searching for ${LIB3MF_LIB}.")

pkg_check_modules(LIB3MF REQUIRED ${LIB3MF_LIB})

if (LIB3MF_FOUND)
	if (EXISTS "${LIB3MF_INCLUDEDIR}/Bindings/Cpp/lib3mf_implicit.hpp")
		set(LIB3MF_API_2)
		add_definitions(-DLIB3MF_API_2)
	endif()
	set(LIB3MF_CFLAGS "${LIB3MF_CFLAGS} -D__GCC -DENABLE_LIB3MF")
	message(STATUS "Found ${LIB3MF_LIB} in ${LIB3MF_LIBDIR}.")
else(LIB3MF_FOUND)
	message(STATUS "Could not find ${LIB3MF_LIB}.")
endif(LIB3MF_FOUND)
***********************************

t-paul, 2021-03-08:
I fear that's going to work across all platforms without becoming a mess. I think the only way to move forward is really separating the detection of 1.x and 2.x. That's probably only going to happen once packages become available or someone helps geting this moving.

PR #4023 (2021-12-20): Fix lib3MF version 2.0 build
Update the FindLib3MF.cmake to correct version 2.0 build. Previous change did not update necessary LIB3MF_FOUND used in version 2.0 check, and include file was missing from find_path.

Added secondary pkg_check_modules for lowercase lib3mf pkg-config file.

Tested with lib3MF 1.8.1 and 2.0.1.

May fix #3709, but was not able to test with 2.1.1.
closed 2021-12-20 by t-paul