openscad/openscad #3714: Change Request to Layout of Customizer: put the checkboxes on the left

mjk-at-sag, 2021-03-09:
# Currently
```
Customizer
[x] Automatic Preview (note: here the checkbox is already to the left of its label!)
...
check me                                                                         [x]
check me, too                                                                    [ ]
```
# Proposal
```
...
[x] check me
[ ] check me, too
```
# ... and while you are at it ...
Consider putting the spinboxes on the left, too.  Naturally, both these requests should be ignored for languages written right-to-left.

otsakir, 2024-02-28:
Is this issue valid ? 

Just confirming before taking some time to look into it. It seems an easy fix...

jordanbrown0, 2024-02-29:
The checkboxes and spinboxes should be on the same side, whichever side that is.

The advantage to putting them on the left is that then they can all be right-aligned in a column immediately next to their left-aligned labels.
The advantage to putting them on the right is that then the labels are on the left and so naturally read before you get to the space you must fill in.

{ I don't have a strong opinion one way or the other. ]