openscad/openscad #3716: `use <>` with certain command-line `-D var=a_var` causes spurious warning

MichaelAtOz, 2021-03-13:
As outlined in [the forum](http://forum.openscad.org/command-line-help-tp32114p32185.html)

With `use <aLibrary.scad>`, command-line with `-D variable=anotherVariable` results in
`WARNING: Ignoring unknown variable 'anotherVariable' in file aLibrary.scad, line <beyond-end-of-file>`.

A warning is generated for every call to a module in the library.

This is because `-D` appends `variable=anotherVariable;` to all source files, including the `use <aLibrary.scad>` file, in memory.
Each call to a library module, re-evaluates all expressions (in case $variables change).
It evaluates `variable=anotherVariable`, looks up `anotherVariable`, because `use` has a private scope, it is not found, hence the warning.

I don't think `-D` should append to `use <>` files.

nophead, 2021-03-13:
-D does need to affect used files though. I.e. when you call a module or a function from them the overridden variable has to be in scope.

MichaelAtOz, 2021-03-14:
> 
> 
> -D does need to affect used files though. I.e. when you call a module or a function from them the overridden variable has to be in scope.

@nophead Am I missing something? You can't override variables when the library is `use<>`'d.

nophead, 2021-03-14:
I can and do extensively in my library. For example I can set $extrusion_width and it affects every module I use.

If I set it on the command line it also needs to work but I think it will even if it only gets added to the main file.

MichaelAtOz, 2021-03-14:
I'm talking `use<>` not `include<>`

`use` has a private scope, it does not get to look at the main scope (except $vars), e.g.

```
a="A";
b="B";
var=a; // variable assigned another variable
$var="dollar";
// write() has echo(a=a,b=b,var=var,$var=$var);
use <write.scad>      echo("use");
//include <write.scad>  echo("include");
echo(a=a,b=b,var=var,$var=$var);
write(var);

Compiling design (CSG Tree generation)...
ECHO: "use"
ECHO: a = "A", b = "B", var = "A", $var = "dollar"
WARNING: Ignoring unknown variable 'a', in file write.scad, line 280. 
WARNING: Ignoring unknown variable 'b', in file write.scad, line 280. 
WARNING: Ignoring unknown variable 'var', in file write.scad, line 280. 
ECHO: a = undef, b = undef, var = undef, $var = "dollar"
Compiling design (CSG Products generation)...
```

So $vars do not need to be appended by `-D`, and normal variables are not accessible.

I don't think `-D` was envisaged as a backdoor to insert things into a `use<>` scope.

jordanbrown0, 2025-03-09:
`$` variables are available to use<>d files because they are scoped through the call stack.  Ordinary non-`$` variables should not be accessible to use<>d files; they should be considered to be part of the top-level file's lexical scope.