openscad/openscad #3717: Bug: two hull shapes becomes three

edwinm, 2021-03-13:
I have two hull shapes, both made up of four cylinders. The stl preview looks fine. After slicing, there appears a third hull shape.
I tried both PrusaSlicer and Ultimaker Cura and both gave this third shape so it doesn't look like a slicer problem.

This is de code:

```
$fn = 50;

bigWindowWidth = 40;
bigWindowLength = 30;
sideThick = 1.5;
boxWidth = 70;
mu = 0.2;
border = 3;

translate([boxWidth/2-bigWindowWidth/2, 0, 0]) {
    hull() {
        translate([-border, -border, 0]) {
            cylinder( sideThick, 2, 2);
        }
        translate([bigWindowWidth+border, -border, 0]) {
            cylinder( sideThick, 2, 2);
        }
        translate([-border, bigWindowLength+border, 0]) {
            cylinder( sideThick, 2, 2);
        }
        translate([bigWindowWidth+border, bigWindowLength+border, 0]) {
            cylinder( sideThick, 2, 2);
        }
    }
}

translate([boxWidth/2-bigWindowWidth/2, 0, sideThick]) {
    hull() {
        translate([mu, mu, 0]) {
            cylinder( sideThick, 2, 2);
        }
        translate([bigWindowWidth-mu, mu, 0]) {
            cylinder( sideThick, 2, 2);
        }
        translate([0, bigWindowLength-mu, mu]) {
            cylinder( sideThick, 2, 2);
        }
        translate([bigWindowWidth-mu, bigWindowLength-mu, 0]) {
            cylinder( sideThick, 2, 2);
        }
    }
}
```

After slicing and when printing, I get a third layer:

[image]

PROBLEM:

In the upper left, you see a small rectangular shape.

EXPECTED:

The small rectangular shape shouldn't be there since it's not in the source.

t-paul, 2021-03-13:
I don't think there's a bug. It's caused by the 3rd cylinder of the 2nd plate having a `mu` z-translation.

[image]

closed 2021-03-13 by t-paul

edwinm, 2021-03-13:
Thanks @t-paul !