openscad/openscad #3718: Extruded STL not manifold after difference

realflash, 2021-03-14:
This shape is a path I created in Inkscape. The path extrudes and then renders fine on its own. When I add a difference with a cube to take a chunk out of it, it previews OK but when rendering I get "ERROR: The given mesh is not closed!" or nothing at all. 

I made the cube significantly bigger than the path object so that there is a definite overlap.

Version 2021.03.09

[carpet_protector.zip](https://github.com/openscad/openscad/files/6137381/carpet_protector.zip)

Is the fact that the error message is not consistent another bug?

rcolyer, 2021-03-14:
While sometimes this error is from internal issues, in this case it reports that the mesh is not closed because the mesh is in fact not closed.  The error only appears when you go to combine with a cube because if no computational geometry steps need to be performed, OpenSCAD is happy to pass along the non-manifold input as non-manifold output.  But it requires the constraints to be satisfied to do calculations on it.

You can see the non-manifold issue pretty clearly if you align yourself directly above it in z using the top-down toolbar item, and select orthogonal viewing mode.  Then you can see that the max-y min-x part of your design has a self-intersection, where the 2D part crosses over itself.  This error therefore originated in inkscape, which has loose constraints and does not care about polygons self-intersecting their lines.

To fix, open the svg with a text editor, and change:
```
d="m -0.13604721,103.00066 c 1.79546131,0.0889 3.64551291,-0.24128
```
to
```
d="m -0.13604721,103.00066 c 1.79546131,0.0889 3.64551291,0
```
The -0.24128 dragged that line too far in y across the other line.

nophead, 2021-03-14:
Tiny 2D intersections can often be fixed by offset(tiny number). The
library used for offset() is good at fixing 2D geometry.

realflash, 2021-03-22:
Thank you. With your assistance I found the problem area. I was able to adjust it in Inkscape and then compile successfully. Are there relevant enhancment requests for this? I think I could have found this on my own if I had been alerted to the fact that the SVG shape was the source of the problem, and then not had to bother you. Things that come to mind are:

- I didn't know that a 2D shape could not be "manifold". I assumed that the extrusion of a 2D shape into a 3D shape should always create a manifold shape, otherwise the extrusion process is broken. I am thinking about manifoldness in the context of whether or not it has holes - perhaps this really about having some thickness somewhere? Having found the problem, I was (am still) confused as to why it was the cause; since the line in the SVG had thickness, the imported shape had thickness at some point everywhere too. I assumed the 2D shape once imported would follow the outside of the thickness of the line all the way around the outer extremity, thus creating a (somewhat lumpier than planned) but nonetheless defined shape.
- I feel as though allowing me to import a "bad" shape and extrude it is setting me up for problems later on. I didn't see a console warning or error without the the difference in play, and I can't easily tell by looking at the extruded shape that there is a problem.
- I was confused as to why sometimes I would compile and get an error, and sometimes get nothing, even if the code hadn't changed.

Clearly there is a continuum of expectations as to what users should have to learn before using the system. Perhaps there are others like me who could do with a little more help in this area. 

On the flip side, been using OpenSCAD for 5 years or so now (very little extrusion, as you can tell) and I am very grateful for all the excellent stuff I have been able to create. Thank you!

nophead, 2021-03-22:
If the 2D shape has self intersections then extruding to 3D will create a non-manifold shape because it will have zero thickness sections where the 2D outline crosses itself. For example, consider a figure of 8. Also an outline with a gap in it will extrude to a 3D shape with hole in the surface. 

The 3D shape will display OK in preview but if you have to do 3D operations on it like union and difference you will get a CGAL exception and parts will be missing. CGAL will not accept anything that isn't 2-manifold.

You only get the error message once though, which is why you sometimes don't see it but there are still bits missing.

A 2D outline with an inner hole is fine. For example extruding an O would get you a tube.

It would be nice if OpenSCAD checked all its imports. It would get rid of more than 90% of the queries in the mailing list.

lf94, 2021-04-01:
It looks like there's been sufficient explanation; if not please re-open this issue :slightly_smiling_face:

closed 2021-04-01 by lf94

realflash, 2021-04-01:
So is the line thickness in the SVG ignored, and is that right or wrong?

t-paul, 2021-04-02:
It's ignored for closed shapes: https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/SVG_Import#Open_and_closed_shapes