openscad/openscad #3719: Windows (at least) Snapshot badly broken - Main Window not displaying content

MichaelAtOz, 2021-03-14:
As per [forum](http://forum.openscad.org/Problem-with-latest-nightly-build-tp32199.html),  look like Qt is misbehaving. New build system??

MichaelAtOz, 2021-03-14:
My test for the forum was 2021.03.04, I just downloaded todays version 2021.03.12.ci7606 (git 4546119f2), still has transparent window content, click where the menus should be & they pop-up, Quit - crash.
Latest nightly on Debian works, version 2021.03.09.

nophead, 2021-03-15:
OpenSCAD 2021.03.12.ci7606 seems to work fine on my Win7 64 bit laptop.

MichaelAtOz, 2021-03-15:
Hmmm. Works on my other Win7/64 box.
I've asked the original forum poster for details.

To clarify, it is not transparent, it looked that way as the window 'captures' the background, and any pop-ups etc, but move the window and that image moves with it.
Doing Design/preview - crash.
Help/Library-info gets:

[image]

This is my VM hosted system accesses via FreeRDP, with S/W rendering via OpenGL32.dll, 
but removing the dll is a possibly clue, it behaves the same, but it does not complain about OpenGL version.
2021.01 for example without the dll gets:

```
Warning: Missing OpenGL capabilities for OpenCSG - OpenCSG has been disabled.
It is highly recommended to use OpenSCAD on a system with OpenGL 2.0 or later.
Your renderer information is as follows:
GLEW version 1.12.0
GDI Generic (Microsoft Corporation)
OpenGL version 1.1.0
```
This doesn't.

Command-line works, png & stl export worked, --info:
```OpenSCAD Version: 2021.03.12.ci7606 (git 4546119f2)
System information: Windows(TM) 6.1 SP 1.0 NTW 1 MSDN 724833 4 CPUs 8.00 GB RAM
User Agent: OpenSCAD/2021.03.12.ci7606 (git 4546119f2) (Windows(TM) 6.1 SP 1.0 NTW 1 MSDN 724833)
Compiler: GCC "10.2.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_60
Eigen version: 3.3.7
CGAL version, kernels: 5.2, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.15.2
QScintilla version: 2.11.2
InputDrivers:
GLib version: 2.50.2
lodepng version: 20180910
libzip version: 1.5.2
fontconfig version: 2.13.1
freetype version: 2.10.4
harfbuzz version: 2.7.4
cairo version: 1.16.0
lib3mf version: (not enabled)
Features: input-driver-dbus, lazy-union, vertex-object-renderers, 
vertex-object-renderers-indexing, 
vertex-object-renderers-direct, vertex-object-renderers-prealloc
Application Path: C:/temp/openscad-2021.03.12.ci7606
Documents Path: C:\Users\MeB\Documents
User Documents Path: C:\Users\MeB\Documents
Resource Path: C:/temp/openscad-2021.03.12.ci7606
User Library Path: C:/Users/MeB/Documents/OpenSCAD/libraries
User Config Path: C:\Users\MeB\AppData\Local/OpenSCAD
Backup Path: C:/Users/MeB/Documents/OpenSCAD/backups
OPENSCADPATH: C:\Users\MeB\Documents\3D-REPRAP\OpenSCAD\OpenSCAD Libraries;
C:\Users;C:\test;C:\test2;C:\test-uc-?
OpenSCAD library path:
  C:/Users/MeB/Documents/3D-REPRAP/OpenSCAD/OpenSCAD Libraries
  C:/Users
  C:/test
  C:/test2
  C:/test-uc-?
  C:/Users/MeB/Documents/OpenSCAD/libraries
  C:/temp/openscad-2021.03.12.ci7606\libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  C:/Windows/fonts
  C:/Users/MeB/.local/share/fonts
  C:/Users/MeB/.fonts

GLEW version: 2.1.0
OpenGL Version: 3.1 Mesa 19.1.4
GL Renderer: llvmpipe (LLVM 8.0, 256 bits)
GL Vendor: VMware, Inc.
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
GL context creator: WGL
PNG generator: lodepng
OS info:
```
lib3mf version: (not enabled)??

MichaelAtOz, 2021-03-16:
Further report by original forum poster.

> OK a few minutes ago I downloaded the latest "Development Snapshots" BOTH 64 and 32 bit windows variants :-
>
>    OpenSCAD-2021.03.12.ci7607-x86-32-Installer
>    OpenSCAD-2021.03.12.ci7606-x86-64-Installer
>
>They both behave as before on my laptop - a blank frame - no buttons or window structure.
>
>When I click where the buttons should be I get a drop-down menu but most of the entries do nothing (things like "show library folder" work - but that is a straight forward windows function).
>
>WIN7 64bit (up to date), on a Levono T410 with 8GB of memory.
>
>The last 64 bit release works fine.

I'm working back on older snapshots. Found one where Windows look normal but F5 crashed, 2021.02.21.ci7310.
It got the OpenGL warning (without dll).
[image]
Command-line --info v's newer 03.12 (with the bad Window_:
[image]
Command-line stl export worked, png export crashed (still without S/W render dll)
```
C:\temp\openscad-2021.02.21.ci7310>openscad -o l.png logo.scad
ECHO: version = [2021, 2, 21]
Compiling design (CSG Products normalization)...
Normalized CSG tree has 4 elements
Compiling highlights (1 CSG Trees)...
Framebuffer Object extension not found by GLEW
Can't create OpenGL OffscreenView. Code: -1.
Geometries in cache: 2
Geometry cache size in bytes: 725104
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0:00:00.123
```

With the dll it works. Older versions on this system without the dll would preview without crashing, but display wrong geometry, e.g. logo.scad shows a sphere. This may well be a different bug to the bad window.

Newer one which had the bad windows, 2021.02.27.ci7455.
Command-line library info comparison.
[image]
Qt version likely suspect? GCC is a big jump.
lib3mf version: (not enabled) again.

MichaelAtOz, 2021-03-17:
Command-line --info from original forum poster. Note Intel graphics.

```
C:\Program Files (x86)\OpenSCAD>openscad --info
OpenSCAD Version: 2021.03.12.ci7607 (git 4546119f2)
System information: Windows(TM) 6.1 SP 1.0 NTW 1 MSDN 724833 4 CPUs WOW64 7.80 GB RAM
User Agent: OpenSCAD/2021.03.12.ci7607 (git 4546119f2) (Windows(TM) 6.1 SP 1.0 NTW 1 MSDN 724833)
Compiler: GCC "10.2.0" 32bit
MinGW build: MingW32
Debug build: No
Boost version: 1_60
Eigen version: 3.3.7
CGAL version, kernels: 5.2, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.15.2
QScintilla version: 2.11.2
InputDrivers:
GLib version: 2.50.2
lodepng version: 20180910
libzip version: 1.5.2
fontconfig version: 2.13.1
freetype version: 2.10.4
harfbuzz version: 2.7.4
cairo version: 1.16.0
lib3mf version: (not enabled)
Features: input-driver-dbus, lazy-union, vertex-object-renderers, 
vertex-object-renderers-indexing, vertex-object-renderers-direct, 
vertex-object-renderers-prealloc
Application Path: C:/Program Files (x86)/OpenSCAD
Documents Path: C:\Users\<cut>\Documents
User Documents Path: C:\Users\\Documents
Resource Path: C:/Program Files (x86)/OpenSCAD
User Library Path: C:/Users//Documents/OpenSCAD/libraries
User Config Path: C:\Users\\AppData\Local/OpenSCAD
Backup Path: C:/Users//Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  C:/Users//Documents/OpenSCAD/libraries
  C:/Program Files (x86)/OpenSCAD\libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  C:/Windows/fonts
  C:/Users//.local/share/fonts
  C:/Users//.fonts

GLEW version: 2.1.0
OpenGL Version: 2.1.0 - Build 8.15.10.2622
GL Renderer: Intel(R) HD Graphics
GL Vendor: Intel
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
GL context creator: WGL
PNG generator: lodepng
OS info: Microsoft(TM) Windows(TM) 6 1 7601 0x4a9f580 x86
Machine: 586

C:\Program Files (x86)\OpenSCAD>
```

MichaelAtOz, 2021-04-07:
Just downloaded the new build from #3193.
Still getting bad window, capturing the background:
[image]
Windows snapshots are unusable.

MichaelAtOz, 2021-04-29:
I also just tried 2021.04.21-64 on my Windows-10 VM in boxes on Debian.
Window was empty white, menus pop-up iff  click in the region, design/f5 crashes.
Exception code: 0xc0000005 Fault offset: 0x0000000000000000 - prob classic reference to null object.

MichaelAtOz, 2021-04-29:
I tried to get help/library-info for the above, got:
[image]

kintel, 2024-03-31:
Is this still an issue?

kintel, 2025-04-06:
Closing as this hadn't resurfaced for > a year.

closed 2025-04-06 by kintel