openscad/openscad #3721: Polyhedron() should report faces list with non-lists

jordanbrown0, 2021-03-18:
In a `polyhedron` call, `faces` is supposed to be a list of lists - e.g. `[ [ 0, 1, 2 ], [3, 4, 5], ... ]`.

If you erroneously give it a simple list - as I just did when trying a trivial experiment:
```
polyhedron(points=[ [0,0,0], [0,1,0], [1,0,0] ], faces=[0,1,2]);
```
you get a mysterious error:
```
WARNING: PolySet has degenerate polygons 
```

I would have expected something more along the lines of what you get when you make a similar error in the `points` list:
```
polyhedron(points=[0,0,0], faces=[[0,1,2]]);
```
yields
```
ERROR: Unable to convert points[0] = 0 to a vec3 of numbers in file , line 2 
WARNING: PolySet has degenerate polygons 
```

redlizard, 2021-05-16:
This should be fixed in master.

thehans, 2021-05-17:
Fixed by #3745

example output:
```
ERROR: Unable to convert faces[0] = 0 to a vector of numbers in file , line 1
ERROR: Unable to convert faces[1] = 1 to a vector of numbers in file , line 1
ERROR: Unable to convert faces[2] = 2 to a vector of numbers in file , line 1
```

closed 2021-05-17 by thehans