openscad/openscad #3722: Give colour triplets/quadruplets the same privileges given to x-y-z triplets

mjk-at-sag, 2021-03-18:
Support `colour.r`, `colour.g`, `colour.b`, and `colour.a` in the same way that `vector.x`, `vector.y`, and `vector.z` are supported.

I see this as a low-priority enhancement -- but well above `"aliceblue".g`!

A much more useful enhancement would be `.h`, `.s`, `.l`, and `.v` to allow some users more comfortable input.
```openscad
c.h = 210; c.s = 79.1; c.v = 33.7;
echo(red=c.r, green=c.g, blue=c.b);
```

MichaelAtOz, 2021-03-18:
```
c.h = 210; c.s = 79.1; c.v = 33.7;
echo(red=c.r, green=c.g, blue=c.b);
```

That is more than just access to elements, that (converting to/from) is equiv. to a color class.

Linking to objects discussion #3088.

jordanbrown0, 2023-11-18:
Access to the first four elements as r, g, b, a, analogous to x, y, z, w, was added in #4010 .

closed 2023-11-18 by jordanbrown0