openscad/openscad #3724: 3D view update from joystick when the window isn't in focus

awsomechicken, 2021-03-19:
Could we get the 3D View to update from the Linux joystick input even when the window isn't in focus? This is specifically for the users, like me, who use an external editor and would like to use a joystick or 3D mouse. The implementation should also have an option under the Axes preferences rather than being set by default.

buzztiaan, 2021-03-19:
3dmouseuser says +1 for 'not set by default' lol

jordanbrown0, 2026-05-07:
HIDAPI on Windows is currently the worst of all possible worlds:  some, but not all, of the axes are active when OpenSCAD does not have focus.  (I think viewport-relative translate and buttons are active, and none of the others, but I haven't carefully walked through all of them.)

It seems to me that the default must be to follow focus - anything else seems surprising - but in those configurations that allow tracking these devices without focus, the "external editor" use case is pretty compelling.