openscad/openscad #3725: Black Window when right clicking

PRouzeau, 2021-03-21:
Since a few weeks, I experienced a bug  on Openscad snapshots. When right clicking on an image, as you do when searching the code associated with a model part (very nice feature, by the way), the list of codes open properly, but the underlying OpenScad windows became black, only the windows top bar remaining visible. The menu bar while invisible is still active and operating (meaning the menus open visibly and are active), the save is working, but it is impossible to recover the underlying OpenScad window. Docking/undocking have no effect. Trying to close/open a panel neither. The only way I found is to close/reopen OpenScad. As the save works, there is minimal damage (lost edit history) but personally as I right click accidentally quite frequently I met this bug a few times à day.
I am using Windows 64 bits version on up to date Windows 10 on a Dell Optiplex 3070. The graphic board is the Intel processor incorporated  UHD graphics 630 with drivers supplied by Dell.

MichaelAtOz, 2021-10-01:
Similar screen issues in #3719.

nophead, 2022-02-11:
I got this problem on my desktop PC when I was last at home but in my winter residence it seems to work properly on my Laptop. So perhaps it is graphics driver dependent.

Parkinbot, 2022-02-11:
It ALSO may be driver dependent or graphics card dependent and most common drivers might catch this condition. My universal NVIDIA driver doesn't. 

However, since I don't get this effect with any other application on my machine (and I have installed many that also use OpenGL and hardware acceleration), I am 99.9% sure that there is a bug in OpenSCAD and that has to do with the popup window showing the module trace.

dannypike, 2022-02-16:
I haven't examined all of the code to work out the correct solution, but I think that I've found a possible cause. This code in ```MouseSelector::select()``` in MouseSelector.cc:
```
  glDisable(GL_LIGHTING);
  glDepthFunc(GL_LESS);
  glCullFace(GL_BACK);
  glDisable(GL_CULL_FACE);
  glEnable(GL_DEPTH_TEST);
```
turns on the Z depth test. It never gets turned off again. That appears to be why the display disappears immediately afterwards - everything gets culled.

To prove this, I added a ```glDisable(GL_DEPTH_TEST)``` after it had extracted the color/index of the selected object and everything started behaving properly. As I said above, I'm not sure if this is the correct solution (or whether you want Z depth testing everywhere), but it should hopefully give you enough info to fix the problem.

Parkinbot, 2022-02-17:
You made my day. 
It works for me on Windows 8.1, 64 Bit.

t-paul, 2022-02-17:
Awesome, thanks @dannypike!

dannypike, 2022-02-17:
You're welcome.

closed 2022-02-17 by t-paul

OoDeLally, 2023-08-30:
I experience this as well on Ubuntu. What is the proper fix exactly? The link @t-paul mentioned does not work anymore.

Matthew0x, 2025-01-06:
> I experience this as well on Ubuntu. I notice it goes away when unchecking the box `Enable for OpenGL 1.x` in `Preferences > Advanced`.

I have the same problem on Fedora and this setting doesn't fix it for me.
[image]

When I right-click somewhere then the screen turns black.
Version 2021.01 from .RPM repository (the latest release on GitHub).

nophead, 2025-01-06:
I have the problem with mice that have keybounce. The right click is interpreted as double right click, which is is show the code that generate the pixel. That gives a balck screen with some of my graphics cards.

kintel, 2025-01-06:
This was fixed 3 years ago. Until we have a new stable release, use the latest development snapshot.

Matthew0x, 2025-01-08:
I assumed so, but I had checked the tags, GH releases, system repository and the OpenSCAD website without seeing anything newer than 2021 and so I simply thought it was not available.

Upon your suggestion I scrolled back a bit further on the website and then I found it. The "Development Snapshots" seem available and even pre-compiled. I will try it out, thanks! Sorry for commenting on an old issue.

Too sweeten the inconvenience: I compared Flatpak (which I had been using) with .RPM and for some reason I get this bug when double right-clicking after hitting the preview button only. In Flatpak it appears to be working fine, although both packages report the same version. The others might be better off with just the Flatpak release.

t-paul, 2025-01-09:
Yes, Flatpak ships a couple of additional patches for 2021.01, some needed to keep it building for the Flatpak platform.

PR #4125 (2022-02-17): Disable depth test after extracting pixel data.
Testing the fix for screen going black on context menu from https://github.com/openscad/openscad/issues/3725#issuecomment-1042413285

Fixes #3725