openscad/openscad #3726: Lost translation

PRouzeau, 2021-03-21:
Lost translation
Recent snapshots (Windows, 64 bits) lost most translation.
The only things which remain translated now is the panels titles (except the error-log panel, too recent stuff), and in customizer panel, the default set name, everything else stay in English (see the images) 
[image]
[image]
Below image is what it used to be. Note that I did’nt lost everything because in this old screen copy (2016), the customizer default set name was not translated, while it is now...
[image]

thehans, 2021-05-24:
I identified some problems with the translation configuration due to our switch from qmake to cmake.  I think it should be better with changes in #3787.
@PRouzeau Can you try the build artifacts from the PR to verify if it is better now?

I've only done a quick test of the win64 build through wine but it seemed to apply translations correctly.

thehans, 2021-05-31:
The PR mentioned above has been merged now, so the most recent dev snapshots should have those changes available for testing.

PRouzeau, 2021-06-02:
Tested the 31.05.2021 snapshot in Windows 32 bit (got troubles- with antivirus ?-  installing the 64 bit version), and translation is now back except that the default name set of customizer is yet back to english (while it was the only translated part on former versions) was 'valeurs de conception par défaut' and now '<design default>'.
[image]
[image]

closed 2021-07-26 by t-paul