openscad/openscad #3733: Feature request: edge shaping

nathanielstenzel, 2021-03-31:
Is there a chance of getting something like rotate_extrude but around the edge of a 2D shape? This would be similar to the result of using a hand router with cabinetry bits to make the fancy edges around the edges. Having the ability to add material to or remove material from the edge would be pretty powerful. You could start by following the edge and either at segment ends on the shape or every X distance you could place the desired shape. It is sort of a mix of a linear_extrude and a rotate_extrude. You could probably call it edge_extrude or router_extrude.

t-paul, 2021-03-31:
This sounds mostly like https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Transformations#offset

nathanielstenzel, 2021-03-31:
While that might perhaps be part of it, that is only regarding the edge following behavior and not the shape that is added to or removed from the linear_extrude results that you could get from the 2D shape. Imagine running a router bit along a board and that is basically what I am asking for. For any 2D shape, linear_extrude it to a certain height and then take a shape that you could rotate_extrude and use that as a router bit to cut the linear_extrude shape. This is essentially router work for OpenSCAD. I would also like the idea of adding or subtracting and that part affects the edge following path in the way that the offset feature works. I guess this might be best handled by starting with working with the newly linear_extruded shape in vertical slices that correspond to vertical slices between segments of the rotate_extrude shape. There is no ability to extract segments from the shapes so I can not make such a function by myself in OpenSCAD and even if there was, it would not be efficient in the least, so I figure adding a new feature would be best.

jordanbrown0, 2021-04-01:
* You can't extract a path from a 2D object.  Yeah, true.  As Torsten says, ref #301.
* Sweep is not built in.  Sure.  But it's also a feature that seems to have broad applicability and understanding.  If as I suspect sweep is a superset of the "router bit" functionality described here, it seems much more likely that sweep will be integrated into the core than that a lesser function will be.
* People don't know about libraries.  MCAD is included with every OpenSCAD distribution; if people don't know to use it then that's a documentation problem, not a problem with it being a library.  (MCAD's sweep seems to be in a strange state, ref openscad/MCAD issue #78, but it could clearly be fixed.) I have no idea whether the BOSL2 maintainers would be interested in bundling it with OpenSCAD, but that certainly could be done.

Critically, the set of people who can effectively work on OpenSCAD libraries is much larger than the set of people who can effectively work on OpenSCAD internals.

I have to wonder how much sweep's not-built-in-ness really hurts the performance of models using it.  The general rule is that it's the geometry processing that hurts, not the execution step per se.  If they both result in the same polyhedron being fed into the geometry processing, there might not be that much of a difference.  Data points:  Generating a 20,000-triangle torus using rotate_extrude() and circle() took maybe a tenth of a second. BOSL2's path_sweep took just over four seconds.  A more specialized function that I wrote takes just under two seconds.  Unioning that 20,000-triangle torus with a cube then took a further ~47 seconds in all three cases.

TLC123, 2021-04-08:
If it's all in  one plane its really really simple but painfully slow.

Take your shape and linear_extrude it a tiny amount.
Then minkowski it with a rotate_extrude spindle of your desired edge shape profile.
This will make your final body larger than your shape so you might want to negative offset if first of all.

For a more exact hand router effect
You first take subtract your shape from a larger bounding shape hull()+ offset()
Then minkowski that outer shape with a rotate_extrude of your desired edge shape profile.

Finally subtract the outer body from an linear_extrusion of your original shape.

Expect hours of rendering .

```
shape= [  for (i = [1: 4]) rands(-100, 100, 2)];

height = 5;
router_offset = 1;
epsilon = 1e-2;
$fn = 8;

router() {
    shape();
    cutterpin();
  } 
%  cutterpin();

module router() {
  color("green")
  difference() {
    linear_extrude(height, convexity = 20) children(0); //"shape()" basic shape
    minkowski(convexity = 120) {
      linear_extrude(epsilon, convexity = 20) difference() { // make inverted shape
        offset(delta = max(height, router_offset * 2)) hull() children(0); //"shape()"
        offset(router_offset) children(0); //"shape()"
      }
      children(1); // "cutterpin()"
    }
  }
}

module cutterpin() {
  //!
  rotate_extrude()
  intersection() {
    square([height + router_offset, height + router_offset]); // cut away all geometry in negative quadrant,
    union() {
      // insert your own profile here
      polygon([
        [0, 0],
        [router_offset, 0],
        [router_offset + height * .25, height * .5],
        [router_offset + height, height],
        [router_offset, height],
        [0, height]
      ]);
    }
  }
}

module shape() {
  //import your svg here 
  offset(1) polygon(shape);
}
```

[image]
[image]

nathanielstenzel, 2021-04-09:
A person has to be careful with hull operation and it has its limits even if you go child by child because you can not import any complex objects. Doing an offset function alone using the two slices is something good, but it would be quite a bit of computation to get things to work. Anything with Minkowski adds to Z so you would need to get the slice and then slice the added Z afterwards. Too bad there is no restriction for what dimensions to add in Minkowski. Back to the hull, if it was somehow limited to nearest triangles to where it could join the top surface and bottom surface without joining the shape, that would do the trick. Of course then, there was an old OpenSCAD forum post that I just found from 2015 that asked for a 3D offset command. Sadly, with what I have available now, the only stuff I could possibly come up with would be suitable for just manipulation of slicer layers for 3D printing or maybe layers in some CNC milling when you are not too worried about the smoothness between layers.