openscad/openscad #3734: Feature request: 2D_to_path and 3D_slice_to_path

nathanielstenzel, 2021-03-31:
I would like the ability to extract a path from a 2D object.

I would also like to be able to extract a slice (as a 2D object or a path) from a given Z height of a 3D object. There may be something from someone's custom SCAD libary (not actually a part of OpenSCAD itself), but having this as a built in OpenSCAD feature would greatly help efficiency and also help give the people who make such lovely libraries.

t-paul, 2021-03-31:
Please don't put multipe request in a single issue, that's impossible to handle.

First part is a duplicate of #301.

Second part exists already: [projection(cut = true);](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/3D_to_2D_Projection)

closed 2021-03-31 by t-paul

nathanielstenzel, 2021-03-31:
@t-paul you pointed out part 2 is handled by projection, but part 1 is still valid, isn't it?

t-paul, 2021-03-31:
Yes, valid and certainly something that would be extremely useful to have. But there's already at least ticket #301 and probably some more that are similar. The multi stage processing behavior of OpenSCAD makes this a bit difficult to do nicely.

nathanielstenzel, 2021-03-31:
@t-paul I do not think #301 is a good fit. That person seems to be more concerned with how to figure out a bounding box. That request was for parameters at first too. I am more concerned with getting a series of dots that are connected by line segments.

t-paul, 2021-03-31:
It might make sense to collect all the references basically asking to query geometry into a clear and specific feature request discussing how it would fit into a future version of the language. That could then cover #301 too. Otherwise it's a known limitation mentioned quite often, so it's not really needed to have just a reminder. With a tiny dev team scraping some spare time, those kind of topics unfortunately move at snail pace.

nathanielstenzel, 2021-03-31:
@t-paul I can imagine the dev team would be small. I tried to look into how to add a feature that I wanted once and could not make any sense of the code. I have programming experience in numerous languages but where to find things totally eluded me. Alright. No dice it is then. Thanks for taking your time to reply.