openscad/openscad #3735: 4000 times slower F6-render (compared to F5-render)

ofmetal, 2021-04-01:
Hi!

I'm trying to build and 3D-print a vase. It is a multilayer thing for FDM-printing. Basically a tube with n-gon bottom (say n=3) and m-gon top (say m=6), as z goes from bottom to top the n-gon morphes to m-gon, while twisting. Also it has holes to insert colored filament sticks inside.

F5 render is pretty fast, it takes less then 2 seconds to draw the whole thing.
But!
F6 render speed is somehow extremely slow, it takes 1h45min on 3.6GHz 16-core 32Gb machine with Radeon 580 8Gb onboard.

4000 times difference in speed!
It's the latest build from 24 March 2021.

[Here is the .scad project](https://pastebin.com/bAJNtQMS)

t-paul, 2021-04-01:
That seems to create a huge amount of slices and doing a union on that. This is going to be slow due to the large number of vertices to process. PR #3641 might improve on that, but preview exists for exactly the reason that mesh generation is quite a bit slower.

closed 2021-04-01 by t-paul

ochafik, 2022-02-03:
Bit of necromancy here (sorry!) but I can confirm that this file renders much faster w/ fast-csg (PR https://github.com/openscad/openscad/pull/3641): it takes just over a minute on my laptop (without fast-csg I couldnt get it to finish in over an hour).