openscad/openscad #3736: International native characters in the names of variables, functions, and modules.

GuramDuka, 2021-04-02:
Needed support for international native characters in the names of variables, functions, and modules. Is it possible?

t-paul, 2021-04-02:
No, there is no plan for supporting just any possible characters. OpenSCAD input is defined to be UTF-8 so that would enable just too many strange issues with identical looking characters, different normalization or even invisible characters in identifiers.

However it might be an option to add some limited support for a dedicated set of additional characters, e.g. I can understand how people would like to use some of the speciall math symbols.

jordanbrown0, 2021-04-02:
I believe JavaScript allows a wide range of non-ASCII characters.  You might consider stealing its rules.

I'm a native English speaker, so I don't feel constrained by ASCII-only languages, but I would think that those whose natural languages aren't covered by ASCII would like to be able to spell words correctly.

GuramDuka, 2021-04-06:
Help us at least determine the places in the code for creating a patch on your own.

t-paul, 2021-04-06:
To be clear, a patch just allowing all unicode characters will not be accepted.

It seems JavaScript targets a dedicated subset defined by the Unicode Standard using `ID_Start` and `ID_Continue` markers which might be a viable option if implemented correctly. It seems newer standards have a more stable `XID_Start` and `XID_Continue` tags which are supposed to be stable across new standard releases.

My suggestion would be looking at [C++ P1949](https://github.com/cplusplus/papers/issues/688) which proposes that kind of usage for C++ and has a good discussion of potential issues. It does seem to have a pretty strong chance to get accepted with no votes against the latest revision.

I don't know yet how to easily import those definitions, I don't see anything related in `glib` which is used for most UTF-8 handling currently. There's probably nobody who can directly tell which code places need changing, with some luck it might be restricted to the lexer.

lf94, 2021-04-14:
Not sure if useful but the C variant in Plan 9 allows this as well, probably using a simple implementation. http://code.9front.org/hg/plan9front/file/bc91b685ad60/sys/src/cmd all the "*c" are the C compilers. The code is not the easiest to read.

closed 2026-08-30 by t-paul