openscad/openscad #3737: acos() bug

powerpowerpower, 2021-04-03:
a=9;b=16;c=9; 

 A=acos((c*c+b*b-a*a)/(2*c*b));
 B=acos((a*a+c*c-b*b)/(2*a*c));
 C=acos((a*a+b*b-c*c)/(2*a*b));
 su=A+B+C;

echo(su); // returns 180. Really?.
echo(ceil(180));//returns 18 if su==180 then ceil(su) == ceil(180) == 180, but!
echo(ceil(su)); //181 
echo(su==180); //false 
echo(ceil(su)==180); //false
//>For angles near 0 and π, arccosine is ill-conditioned and will thus calculate the angle with reduced accuracy in a computer implementation (due to the limited number of digits).
echo(floor(su)); //180 	
echo(floor(su)==180); //true 
//that means 181>su>180

powerpowerpower, 2021-04-03:
ceil(su)==180||floor(su)==180  may can be used for code bugfixing

t-paul, 2021-04-03:
I don't know what you expect, but that's just a normal side effect of floating point numbers. OpenSCAD behaves just like other languages in that respect.

```
$ js
Welcome to Node.js v12.21.0.

a=9;b=16;c=9;

A=Math.acos((c*c+b*b-a*a)/(2*c*b));
B=Math.acos((a*a+c*c-b*b)/(2*a*c));
C=Math.acos((a*a+b*b-c*c)/(2*a*b));
su=(A+B+C)*180/Math.PI;

console.log(su);
> 180.00000000000003
console.log(Math.ceil(su));
> 181
```

```
ipython3 
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.20.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import math
   ...: 
   ...: a=9
   ...: b=16
   ...: c=9
   ...: 
   ...: A=math.acos((c*c+b*b-a*a)/(2*c*b))
   ...: B=math.acos((a*a+c*c-b*b)/(2*a*c))
   ...: C=math.acos((a*a+b*b-c*c)/(2*a*b))
   ...: su=(A+B+C)*180/math.pi
   ...: 
   ...: print(su);
   ...: print(math.ceil(su));
   ...: 
180.00000000000003
181
```

Detailed article on the topic: [What Every Computer Scientist Should Know About Floating-Point Arithmetic ](https://docs.oracle.com/cd/E19957-01/806-3568/ncg_goldberg.html)

closed 2021-04-03 by t-paul

MichaelAtOz, 2021-04-03:
> echo(su); // returns 180. Really?.

The missing piece is echo() outputs limited decimal places. 
```
echo(1.2345678901234); // ECHO: 1.23457
```