openscad/openscad #3738: HELP: Building for Amazon Linux 2; stuck on Boost library

musingsole, 2021-04-05:
I am having a problem with the boost library. I've tried using version 1.55 and 1.75, but both give me this error when running `make` on openscad:

```
/usr/local/include/boost/filesystem/operations.hpp:258: undefined reference to `boost::filesystem::detail::absolute(boost::filesystem::path const&, boost::filesystem::path const&, boost::system::error_code*)'
```

I believe I am not linking the boost library into the build correctly, but I'm not very experienced with this.

Scopeuk, 2021-04-05:
The qmake build system is currently phased out in favour of the cmake based system. 
This particular compile issue can come up if the c++ standard used for the library compile doesent match the project compilation settings https://stackoverflow.com/questions/35007134/c-boost-undefined-reference-to-boostfilesystemdetailcopy-file

musingsole, 2021-04-05:
Thank you! That seems to have resolved the undefined reference error. Though now I'm seeing a new error with the boost library that I can't intepret:

```
/usr/local/include/boost/variant/detail/apply_visitor_binary.hpp:175:32:   required from ‘typename Visitor::result_type boost::apply_visitor(const Visitor&, Visitable1&, Visitable2&) [with Visitor = notequal_visitor; Visitable1 = const boost::variant<UndefType, bool, double, str_utf8_wrapper, Value::VectorType, Value::EmbeddedVectorType, ValuePtr<RangeType>, ValuePtr<FunctionType> >; Visitable2 = const boost::variant<UndefType, bool, double, str_utf8_wrapper, Value::VectorType, Value::EmbeddedVectorType, ValuePtr<RangeType>, ValuePtr<FunctionType> >; typename Visitor::result_type = Value]’
src/value.cc:826:71:   required from here
src/value.cc:764:63: warning: unused parameter ‘op1’ [-Wunused-parameter]
   template <typename T, typename U> Value operator()(const T &op1, const U &op2) const { return true; }
                                                               ^~~
src/value.cc:764:77: warning: unused parameter ‘op2’ [-Wunused-parameter]
   template <typename T, typename U> Value operator()(const T &op1, const U &op2) const { return true; }
                                                                             ^~~
make: *** [objects/src/value.o] Error 1
```

What does it mean that qmake has been phased out? The process I was following to build was:
```
./scripts/uni-build-dependencies.sh
qmake openscad.pro
make
make install
```

Is there a different step that should replace the `qmake openscad.pro` statement?

t-paul, 2021-04-05:
The snippet you pasted seems to be just a warning, are you running parallel make (`make -j4` or similar)? The error might be quite a bit above in the logs.

Development builds move to:
```
mkdir build && cd build && cmake ..
```

Right now `qmake` still works, the transition is still in progress, but support for `qmake` will be removed at some point soon-ish. So basically, if you build the release version v2021.01, then use `qmake`, dev snapshots / master and upcoming releases should use `cmake`.

musingsole, 2021-04-05:
Ah, you were right! I found this error deeper in the log:

```
/usr/local/include/boost/variant/variant.hpp:1591:38: error: use of deleted function ‘Value::operator bool() const’
               initializer::initialize(
               ~~~~~~~~~~~~~~~~~~~~~~~^
                   storage_.address()
                   ~~~~~~~~~~~~~~~~~~
                 , operand
                 ~~~~~~~~~
                 )
                 ~
In file included from src/value.cc:37:0:
src/value.h:550:3: note: declared here
   operator bool() const = delete;
```

t-paul, 2021-04-05:
Is that boost 1.55?

Sounds like that's not compatible with latest OpenSCAD code anymore, so we may need to update docs to at least 1.56 which mentions:

https://www.boost.org/users/history/version_1_56_0.html
**Move-only classes now can be returned by visitors.**

If you can use a newer version, I have working builds with 1.74. I don't know yet about 1.75.

t-paul, 2021-04-06:
Meh, scratch that, I jumped to conclusion just looking at the "= delete" thinking it's the copy constructor, but it's obviously not. Anyway, oldest boost used on OBS seems to be 1.58 on Ubuntu 16.04, so there's actual builds with >= 1.58 and <= 1.74.

musingsole, 2021-04-06:
Yeap, this was with boost 1.55. I haven't tried boost 1.75 with the compile flag, so, that might work. I'm rebuilding the environment but will try a newer version once I'm up and running again. Thank you!!!

musingsole, 2021-04-06:
I was able to complete the build successfully using Boost 1.74. Thank you all for immensely the help!!!

musingsole, 2021-04-09:
For reference for any future code archaeologists, here's the Dockerfile I made to build openscad on Amazon Linux 2:

```
FROM public.ecr.aws/lambda/python:3.8
WORKDIR /root

RUN yum -y groupinstall "Development Tools" && yum -y install wget

RUN wget -O epel.rpm https://dl.fedoraproject.org/pub/epel/epel-release-latest-7.noarch.rpm
RUN yum -y install ./epel.rpm

RUN yum -y install amazon-linux-extras vim cmake git python3 python3-devel tar bzip2 bzip2-devel glew glew-devel openssl openssl-devel sed xz-utils cairo cairo-devel libxml2 libxml2-devel lib3mf lib3mf-devel double-conversion double-conversion-devel eigen3 eigen3-devel opencsg opencsg-devel mpfr mpfr-devel gmp gmp-devel glib2 glib2-devel fontconfig fontconfig-devel harfbuzz harfbuzz-devel libzip libzip-devel gzip

RUN git clone https://gitlab.kitware.com/cmake/cmake.git && cd cmake && ./bootstrap && make -j4 && make install

RUN git clone https://code.qt.io/qt/qt5.git && cd qt5 && git checkout 5.12 && perl init-repository
RUN mkdir qt5-build && cd qt5-build && ../qt5/configure -confirm-license -developer-build -opensource -nomake examples -nomake tests
RUN cd qt5-build && make -j4 && make install
RUN export PATH=/root/qt5-build/qtbase/bin:$PATH
RUN echo 'export PATH=/root/qt5-build/qtbase/bin:$PATH' > ~/.bashrc

RUN wget https://dl.bintray.com/boostorg/release/1.74.0/source/boost_1_74_0.tar.gz && tar -zxf boost_1_74_0.tar.gz && cd boost_1_74_0 && ./bootstrap.sh && ./b2 install

RUN wget https://github.com/CGAL/cgal/releases/download/releases%2FCGAL-4.9.1/CGAL-4.9.1.tar.xz
RUN tar -xf CGAL-4.9.1.tar.xz && cd CGAL-4.9.1 && cmake . && make -j4 && make install

ENV PATH="/root/qt5-build/qtbase/bin:${PATH}"
RUN LD_LIBRARY_PATH="/usr/local/lib:${LD_LIBRARY_PATH}"
RUN echo 'export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH' >> ~/.bashrc && echo 'export PATH=/root/qt5-build/qtbase/bin:$PATH' > ~/.bashrc

RUN wget https://www.riverbankcomputing.com/static/Downloads/QScintilla/2.11/QScintilla_gpl-2.11.tar.gz && tar -zxf QScintilla_gpl-2.11.tar.gz
RUN cd QScintilla_gpl-2.11/Qt4Qt5 && qmake qscintilla.pro && make -j4 && make install
RUN export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH && echo 'export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH' >> ~/.bashrc

RUN git clone https://github.com/openscad/openscad.git
RUN cd openscad && qmake openscad.pro
RUN sed -i 's/-mt//g' openscad/Makefile
RUN cd openscad && make -j4 && make install
```