openscad/openscad #3742: CGAL assertion with difference operation

Claghorn, 2021-04-11:
I have a model of the bottom of a ginger ale bottle created with scanner and lots of manual tweaking. This model has been smoothed a bit with meshlab and run through the netfabb online fix service to make sure it is a legit solid with no problems. When I try to difference this imported stl file with a couple of cylinders to make a bottle holder (so the $#@! things won't tip over all the time), I always get this error:

ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /usr/include/CGAL/Nef_S2/SM_const_decorator.h Line: 321

If I make the same cylinders in tinkercad and import the same stl file as a "hole", tinkercad produces the result I expect, so it also doesn't think there is anything wrong with the stl file.

[gingerale.zip](https://github.com/openscad/openscad/files/6291926/gingerale.zip)

The attached gingerale.zip file has the holder.scad source and the meshlab-smoothed_fixed.stl file it imports

MichaelAtOz, 2021-04-11:
Netfabb PC, so I assume online, does not automatically check for self-intersections. You do it manually:

 

 

Scans are often bad.

 

  _____  

 

 

This email has been checked for viruses by AVG.
https://www.avg.com

MichaelAtOz, 2021-04-11:
hmm, didn't like the image in email...

[image]

kintel, 2024-03-31:
Status as of 2024-03-24: Works when using Manifold, still broken with CGAL