openscad/openscad #3751: OpenSCAD-nightly behaves very oddly with temporary files

BlueNalgene, 2021-04-26:
## Short Description

The snap package of ``openscad-nightly`` cannot find some files stored in ``/tmp`` and instead finds ghost files which were deleted.

## Vitals

Ubuntu 18.04 Bionic
x86_64 Linux 4.15.0-142-generic
KDE 5.44.0 / Plasma 5.12.9
CPU: Intel Core i7-7500U @ 4x 3.5GHz
GPU: Mesa DRI Intel(R) HD Graphics 620 (KBL GT2)
RAM: 5602MiB / 7898MiB

I'm using two OpenSCAD versions to demonstrate the issue here.

**STANDARD**: OpenSCAD version 2021.01 (downloaded from the old versions repo)
**NIGHTLY**: OpenSCAD version 2021.04.14 (snap package)

## Detailed Description

This problem was first worked on over at the FreeCAD forums.  See [here](https://forum.freecadweb.org/viewtopic.php?f=3&t=57970&p=499522) for the background.  FreeCAD incorporates OpenSCAD code by the following steps:

1. Create a temporary ``.scad`` file in ``/tmp``
2. Convert the file via command line to create a new file in ``\tmp``.  This is usually a ``.csg`` file unless an ``.stl`` is requested.
3. Delete the old ``.scad`` file.

The steps use the binary OpenSCAD file pointed to in the preferences.  When I point FreeCAD to **STANDARD**, things work just fine as described in the steps above.  When I point FreeCAD to **NIGHTLY**, things fail after step 1 and an error is issued "Can't open input file '/tmp/fc-09626-452490-000001.scad'!".  

I tested this by manually opening the ``\tmp`` file using command line directly to the ``.scad`` file using each binary.  As before, when using **STANDARD**, the file opens and when using **NIGHTLY** the same error is issued in the GUI.

I tested opening another ``.scad`` file in my system NOT stored in ``\tmp``.  Both binaries are able to open the file.

I tested opening a file using the GUI rather than the command line.  This is where things get weird.  First, let me check what files are stored in ``\tmp``:

```
$ ls -l /tmp/*.scad
-rw-rw-r-- 1 USER USER 7 Apr 26 15:20 /tmp/fc-04286-844564-000001.scad
-rw-rw-r-- 1 USER USER 7 Apr 26 15:54 /tmp/fc-06590-048927-000001.scad

$ ls -l /tmp/*.csg
-rw-rw-r-- 1 USER USER 41 Apr 26 15:19 /tmp/fc-04142-836874-000001.csg
```

When I use the **STANDARD** GUI at this path I see these same files:

[image]

But, when I use the **NIGHTLY** GUI at ``\tmp`` I see completely different files:

[image]

Neither of those files currently exist in ``/tmp`` according the console.  Extra fun, I can actually open them -- and they are from a different project.

What in the world is going on here?

t-paul, 2021-04-27:
Snap packages are by default isolated from the system. See https://www.freecodecamp.org/news/managing-ubuntu-snaps/ for some details.

closed 2021-04-27 by t-paul

BlueNalgene, 2021-04-28:
Huh, I've been grudgingly using snaps for a while now and I just learned that it is a walled garden app.  How annoying.  I had assumed they weren't different from classic package management apps.  I can see the security and ease-of-development benefits for using Snap, but this scattered framework is difficult to work with for people who like to break things.

With a little more investigation, it looks like the snap package stores temporary files in ``/tmp/snap.openscad-nightly/tmp``.  The ``.csg`` files there have the same permissions as those in the default ``/tmp``, but they are protected by the ``/tmp/snap.openscad-nightly/`` directory which has root ownership.

It may be possible to modify the snap package to alter how it handles tmp files for cases such as mine, but that is outside the scope of this issue.  Good call on closing it.