openscad/openscad #3752: Failed to build on M1 macOS

LSChyi, 2021-04-28:
Hello, I want to make a native build on the M1 MacBook, but I failed with missing libraries during linking. I am not familiar with `qmak`, so I would like to share my build process and hoping to get some new ideas.

Here is my build flow, I take the `Homebrew` approach by running the `./scripts/macosx-build-homebrew.sh` command based on the `Building for Mac OS X` section. All dependencies were installed correctly. After that, I run `qmake openscad.pro`. The make successfully generated the `Makefile` file without any error:

```
Project MESSAGE: If you're building a development binary, consider adding CONFIG+=experimental
Project MESSAGE: Using ccache
Project MESSAGE: Using QOpenGLWidget
Project MESSAGE: Using C++14
Project WARNING: Qt has only been tested with version 10.15 of the platform SDK, you're using 11.3.
Project WARNING: This is an unsupported configuration. You may experience build issues, and by using
Project WARNING: the 11.3 SDK you are opting in to new features that Qt has not been prepared for.
Project WARNING: Please downgrade the SDK you use to build your app to version 10.15, or configure
Project WARNING: with CONFIG+=sdk_no_version_check when running qmake to silence this warning.
Project MESSAGE: cairo enabled
Project MESSAGE: 3MF Import/Export enabled API 1.0
```

Things are getting trick from this point. When I run the next command `make`, the compilation failed for not finding the library `src/AST.h:5:10: fatal error: 'boost/filesystem.hpp' file not found
#include <boost/filesystem.hpp>`. I realized that it is because the library installed path is changed with `Homebrew`, so I edited the `INCLUDEPATH`:

```
diff --git a/openscad.pro b/openscad.pro
index 20b39b553..f6487d4c5 100755
--- a/openscad.pro
+++ b/openscad.pro
@@ -96,6 +96,8 @@ macx {
   APP_RESOURCES.files = OpenSCAD.sdef dsa_pub.pem icons/SCAD.icns
   QMAKE_BUNDLE_DATA += APP_RESOURCES
   LIBS += -framework Cocoa -framework ApplicationServices
+  INCLUDEPATH += "/opt/homebrew/include"
+  LIBS += "/opt/homebrew/lib"
   QMAKE_MACOSX_DEPLOYMENT_TARGET = 10.9
 }
```

Now the missing .h files issue is solved, but the `ld` fails to find the library:

```
ld: library not found for -ldouble-conversion
```

I confirmed the `double-conversion` package is installed correctly, and I also tried to temporarily add the link option `-L/opt/homebrew/Cellar/double-conversion/3.1.5_1/lib` in the generated `Makefile`, and thhe missing of the double-conversion library is solved, but other libraries such as `boost_thread` are missing now with the `ld` linking. Is there any suggestions to fix this build issue? Thank you.

LSChyi, 2021-04-28:
After setting the `export OPENSCAD_LIBRARIES="/opt/homebrew"` and re-run the `qmake`, the issue of missing libraries during linking is solved. However, I got another issue:

```
ld: can't map file, errno=22 file '/opt/homebrew/Cellar' for architecture arm64
```

Scopeuk, 2021-04-28:
The M1 builds at this point are limited by the availability of a suitable CI target to allow automatic generation. this is an outstanding git hub issue https://github.com/actions/virtual-environments/issues/2187 
having full notes here on any issues encountered is probably a good stepping stone

LSChyi, 2021-04-28:
In the end, I only need to deal with the `Homebrew` path and setting the `OPENSCAD_LIBRARIES` path, and the latter seems to be related to the first one. Is there a good way to tell the system to add more library paths? I also test the build, everything works great. I can open a complicated model, output STL from the terminal. If there is anything I can help with, please tell me.

t-paul, 2021-04-28:
I would expect the build system to be able to compile universal binaries for both targets on any platform, so it might be possible to progress even before M1 builds are available as CI platform.

Getting the builds prepared would be great, but homebrew is not going to help here. So far homebrew builds work fine locally, but not across different macOS versions on other machines, so they are useless for distribution.

What we need is integration of M1/universal builds into the `macos-build-dependencies.sh` script. This is used to compile the application for >= macOS 10.9. (see [github release workflow](https://github.com/openscad/openscad/blob/master/.github/workflows/macos-release.yml)).

Any help is appreciated as currently there's nobody in the core dev team with access to M1 for testing and Apple does not allow running macOS on any other hardware via VM, so unless there's some additional help there's no native M1 support possible.

LSChyi, 2021-04-28:
Ok, so at least we need to wait for the VM support for M1. Here is the build I made In case someone needs it:

https://m1-local-build.s3.us-east-2.amazonaws.com/OpenSCAD.zip

The build is based on commit `d9d4a9757b4616df755c2219dfcff451f4e3c0a2` (latest commit for current master branch)

t-paul, 2021-04-28:
I don't have much options to try it out, but my understanding is that it would be possible build Intel + M1 universal app now on our Intel CI as it's using XCode >= 12.2. There's just no point in me spending time on that as I can't test run it and see if it works.

So the missing part is the update to both the dependencies build script and the `cmake` build for latest master source code... and being able to see if it's running.

Using `qmake` is not supported going forward, 2021.01 will be the last version to build with `qmake` and unless someone is really ready to spend the extra work, it's unlikely that 2021.01 will get an official M1 build. If all goes well, there might be a bugfix release later this year, preparing for that would be the best option I believe.

Scopeuk, 2021-04-28:
My apologies i mispoke over the build process m1 ci would only be needed for automated testing. It appears cmake has flags for generating universal binaries https://stackoverflow.com/questions/65157483/macos-build-universal-binary-2-with-cmake but i've not tested this.
Anyone in a position to attempt generating these builds and as importantly running and testing them sounds great.

LSChyi, 2021-04-29:
Ok, if I have any updates, I will first post them here then we can decide the next step. Also, the issue I have here seems not completely related to M1, but to the `Homebrew` path in Big Sur. As a result, any future build on Big Sur would need some change.

LSChyi, 2021-05-08:
@jeregrine from the first comment, I describe that I use homebrew to install the dependencies, but failed at complication time. In the end, I found that the build failed because that homebrew changes the path. If you want to make a build locally (also for the native arch), the easiest way for doing this is to set the `OPENSCAD_LIBRARIES` with the new homebrew path. By setting the `OPENSCAD_LIBRARIES`, I do not need to change any files and the build is working. I think there is no intention for configuring the homebrew script (as you can see, there is no parameter for that script), so for now you can not just configure that script and make the build.

>Its unclear from the post but openscad runs fine under Rosetta for now when installed via homebrew.

Yes, you can run Openscad installed by homebrew cask under Rosetta, but this post is for discussing how to make a build for the M1 ARM arch.

LSChyi, 2021-06-03:
I made some progress. I can build the universal binary, by using the `lipo` command to combine the build for `x86_64` and build for `amr64`.

```
$ lipo -create -output universal_app OpenSCAD.app/Contents/MacOS/OpenSCAD ~/Downloads/OpenSCAD.app/Contents/MacOS/OpenSCAD
$ file universal_app
universal_app: Mach-O universal binary with 2 architectures: [x86_64:Mach-O 64-bit executable x86_64] [arm64]
universal_app (for architecture x86_64):	Mach-O 64-bit executable x86_64
universal_app (for architecture arm64):	Mach-O 64-bit executable arm64
$ du -sh universal_app
 19M	universal_app
```

However, I am not sure how to do the cross-compiling on a different architecture. Also, if I simply changed the `EXPORT_VALID_ARCHS` in the generated `Makefile`, the build fails at linking stage:

When setting the build architecture as `x86_64` on the M1 machine, I got the error as shown below
```
ld: symbol(s) not found for architecture x86_64
```
, and I got another error as shown below when setting the build architecture as `arm64` on the intel machine:
```
ld: symbol(s) not found for architecture arm64
```

I think after we solve this issue, we should be able to just make two builds for each architecture, and link them into a universal binary.

IanSC, 2021-08-30:
@LSChyi. Thanks for your solution.

Folllowed your guide, which to summarize for noobs like me from the Windows world.  
In terminal, go to directory where you downloaded the source code, and enter the following, one line at a time.  
```
export OPENSCAD_LIBRARIES="/opt/homebrew" 
```
```
qmake openscad.pro
```
```
make
```
Unfortunately, I got additional errors:
```
src/parameter/parameterobject.h:3:10: fatal error: 'json.hpp' file not found
```
So, I edited the file **openscad.pro** and added these at the very start
```
OPENSCAD_LIBRARIES = "/opt/homebrew";
INCLUDEPATH += src/ext/json
```
then run again: &nbsp; ``` qmake openscad.pro ``` &nbsp; ``` make ``` . &nbsp; Now I'm getting:
```
Undefined symbols for architecture arm64:
  "export_param(SourceFile*, boost::filesystem::path const&, std::__1::basic_ostream<char, std::__1::char_traits<char> >&)", referenced from:
      do_export(CommandLine const&, RenderVariables const&, FileFormat, SourceFile*) in openscad.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```
Did I miss anything? I don't need a clean makefile. I just need it compiled and running.
Thanks

t-paul, 2021-08-30:
Please use `cmake` now, `qmake` is being phased out.

IanSC, 2021-08-30:
Thanks @t-paul 

I used ```make``` because it said in the ```README.md``` of OpenSCAD itself.  
I'm stating step by step of what I did, so it will be easier to pinpoint what is wrong.

I tried ```cmake```, need parameter. Tried ```cmake .```, got this:
```
CMake Error at /opt/homebrew/lib/cmake/Qt5Core/Qt5CoreConfig.cmake:14 (message):
  The imported target "Qt5::Core" references the file

     "/opt/homebrew/.//mkspecs/macx-clang-arm64"

  but this file does not exist.  Possible reasons include:
  * The file was deleted, renamed, or moved to another location.
  * An install or uninstall procedure did not complete successfully.
  * The installation package was faulty and contained
     "/opt/homebrew/lib/cmake/Qt5Core/Qt5CoreConfigExtras.cmake"
  but not all the files it references.

Call Stack (most recent call first):
  /opt/homebrew/lib/cmake/Qt5Core/Qt5CoreConfigExtras.cmake:56 (_qt5_Core_check_file_exists)
  /opt/homebrew/lib/cmake/Qt5Core/Qt5CoreConfig.cmake:227 (include)
  /opt/homebrew/lib/cmake/Qt5/Qt5Config.cmake:28 (find_package)
  CMakeLists.txt:367 (find_package)
```
I also tried the non-Homebrew option in the readme,
```source setenv_mac.sh```. 
**setenv_mac.sh** not found, I can't find it anywhere, closest is **scripts/setenv-macos.sh**. 
Next command of course did not work also ```./scripts/macosx-build-dependencies.sh```.

For those coming from VS Windows who wants to get involved, but don't know how to get started:
Download JetBrains CLion (never used it before).
Select the openscad folder. Open ```src/MainWindow.cc```.  
A mini prompt appeared on the upper right side showing cmake.  
Good thing @t-paul mentioned cmake so, I clicked that and chose ```CMakeLists.txt```.  
It created run configurations. Now Run works. Debug works. Breakpoints works.
Even the list of To-Do's are right there.

I also tried VSCode but don't know how to set it up.

Maybe I did not search hard enough, but a guide of recommended editors, extensions, IDEs would be very helpful.
Now I have 30 days to get it running in VSCode (or another IDE). VS Mac does not support c++.

LSChyi, 2021-08-31:
Sorry I am kind of busy recently. In the last discussion, I think a better direction would be not to try to build this directly on an M1 machine but to cross-compile on an Intel machine as it can be integrated into the CI. As far as I saw, for most Xcode projects, they just change the build option inside the GUI interface and make the universal release (a binary simply bundles two kinds of executables via `lipo`). As I am not familiar with the development of Xcode projects, I did not make much progress since then.

It is very welcomed to give any suggestions or feedback so that we can figure out how to build a universal release, and integrated it into the current build process.

IanSC, 2021-08-31:
Sorry to bother you. My main goal is to edit and hopefully contribute codes. Didn't know where to start, and your post was the solution to getting it built before doing any coding. Thanks again it set me in the right path.

IanSC, 2021-08-31:
To continue for those getting stated like me (point of view coming from VS Windows).

@GregNumworks discusses how to debug in M1 VSCode:
https://github.com/microsoft/vscode-cpptools/issues/6779

Download extension ```vadimcn.vscode-lldb```
CodeLLDB by Vadim Chugunov

add setting in ```.vscode/launch.json```
```
{
    "version": "0.2.0",
    "configurations": [
        //{ other configurations },
        {
            "name": "Launch",
            "type": "lldb",
            "request": "launch",
            "program": "${workspaceFolder}/build/OpenSCAD.app/Contents/MacOS/OpenSCAD",
        }
    ]
}
```
Use ```lldb``` instead of ```cppdbg```.
As for the executable:
In Finder it appears as an app, you have to right-click, show contents to get inside.
In terminal it's ```OpenSCAD.app``` --> ```Contents``` --> ```MacOS```.
In VSCode it appears as ```OpenSCAD.app/Contents``` --> ```MacOS```.
Didn't notice the Contents appended with the folder at first.
And yes, no .exe extension.

incanus, 2021-11-10:
FWIW, I just sat down to build on my M1 Mac mini since I was running the `2021.01` Intel build under Rosetta 2. Using a fully upgraded Homebrew & the dependency build script, then a normal `cmake` build took about 7 minutes on my machine to fully build the `.app` and it runs fine. 

[image]

thehans, 2021-11-12:
@incarnus That's great you are able to get the build working.  

I've been recently looking to recruit some mac users to help with testing/debugging of PR #3943 (with some discussion in Issue #3930).  The PR is a potential performance enhancement where initial tests on Linux showed about 35% improvement to render times.  The problem right now is that CI tests are crashing on mac, and I have no mac hardware to test on myself.  
If this sounds like something you would be interested/willing to help with, then please reply in #3930

tcurdt, 2021-12-03:
So one of the two should work to produce a build on a M1?

    sh scripts/macosx-build-dependencies.sh && cmake
    sh scripts/macosx-build-homebrew.sh

https://github.com/openscad/openscad/blob/master/scripts/macosx-build-dependencies.sh
https://github.com/openscad/openscad/blob/master/scripts/macosx-build-homebrew.sh

t-paul, 2021-12-03:
Homebrew build is confirmed to work, which is nice and works well for self-building. But it's not any use for distributing snapshot or release builds.

tcurdt, 2021-12-04:
Hope I am not missing anything - but this is how I built my openscad from master

```
git clone git://github.com/openscad/openscad.git
cd openscad
git submodule update --init
source ./scripts/setenv-unibuild.sh
./scripts/macosx-build-homebrew.sh
brew link --overwrite qt@5 # due to some conflict!?
PATH="$(brew --prefix qt5)/bin:$PATH"
mkdir -p build && cd build && cmake .. && cd ..
make
```
I was led astray by some old docs.

tcurdt, 2021-12-04:
Darn - but it has a lot of dependencies into homebrew. So cleaning those up breaks the app.

```
otool -L /Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD 
/Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD:
	/opt/homebrew/opt/boost/lib/libboost_filesystem-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/opt/homebrew/opt/boost/lib/libboost_system-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/opt/homebrew/opt/boost/lib/libboost_regex-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/opt/homebrew/opt/boost/lib/libboost_program_options-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/opt/homebrew/opt/harfbuzz/lib/libharfbuzz.0.dylib (compatibility version 30102.0.0, current version 30102.0.0)
	/opt/homebrew/opt/fontconfig/lib/libfontconfig.1.dylib (compatibility version 14.0.0, current version 14.0.0)
	/opt/homebrew/opt/glib/lib/libglib-2.0.0.dylib (compatibility version 7001.0.0, current version 7001.1.0)
	/opt/homebrew/opt/gettext/lib/libintl.8.dylib (compatibility version 11.0.0, current version 11.0.0)
	/opt/homebrew/opt/double-conversion/lib/libdouble-conversion.3.dylib (compatibility version 3.0.0, current version 3.1.5)
	/opt/homebrew/opt/opencsg/lib/libopencsg.1.dylib (compatibility version 1.4.0, current version 1.4.2)
	/opt/homebrew/opt/glew/lib/libGLEW.2.2.dylib (compatibility version 2.2.0, current version 2.2.0)
	/System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL (compatibility version 1.0.0, current version 1.0.0)
	/opt/homebrew/opt/libzip/lib/libzip.5.dylib (compatibility version 5.0.0, current version 5.4.0)
	/opt/homebrew/opt/freetype/lib/libfreetype.6.dylib (compatibility version 25.0.0, current version 25.0.0)
	/usr/lib/libxml2.2.dylib (compatibility version 10.0.0, current version 10.9.0)
	/opt/homebrew/opt/cairo/lib/libcairo.2.dylib (compatibility version 11603.0.0, current version 11603.0.0)
	/opt/homebrew/opt/lib3mf/lib/lib3MF.1.dylib (compatibility version 1.0.0, current version 1.8.1)
	/usr/lib/libz.1.dylib (compatibility version 1.0.0, current version 1.2.11)
	/System/Library/Frameworks/Cocoa.framework/Versions/A/Cocoa (compatibility version 1.0.0, current version 23.0.0)
	/opt/homebrew/opt/qt@5/lib/QtMultimedia.framework/Versions/5/QtMultimedia (compatibility version 5.15.0, current version 5.15.2)
	/opt/homebrew/opt/qt@5/lib/QtOpenGL.framework/Versions/5/QtOpenGL (compatibility version 5.15.0, current version 5.15.2)
	/opt/homebrew/opt/qt@5/lib/QtConcurrent.framework/Versions/5/QtConcurrent (compatibility version 5.15.0, current version 5.15.2)
	/opt/homebrew/opt/qt@5/lib/QtNetwork.framework/Versions/5/QtNetwork (compatibility version 5.15.0, current version 5.15.2)
	/opt/homebrew/opt/qt@5/lib/QtSvg.framework/Versions/5/QtSvg (compatibility version 5.15.0, current version 5.15.2)
	/opt/homebrew/opt/qscintilla2/lib/libqscintilla2_qt5.15.dylib (compatibility version 15.1.0, current version 15.1.1)
	/opt/homebrew/opt/qt@5/lib/QtDBus.framework/Versions/5/QtDBus (compatibility version 5.15.0, current version 5.15.2)
	/opt/homebrew/opt/qt@5/lib/QtGamepad.framework/Versions/5/QtGamepad (compatibility version 5.15.0, current version 5.15.2)
	/opt/homebrew/opt/gmp/lib/libgmpxx.4.dylib (compatibility version 11.0.0, current version 11.1.0)
	/opt/homebrew/opt/mpfr/lib/libmpfr.6.dylib (compatibility version 8.0.0, current version 8.0.0)
	/opt/homebrew/opt/gmp/lib/libgmp.10.dylib (compatibility version 15.0.0, current version 15.1.0)
	/opt/homebrew/opt/qt@5/lib/QtWidgets.framework/Versions/5/QtWidgets (compatibility version 5.15.0, current version 5.15.2)
	/opt/homebrew/opt/qt@5/lib/QtGui.framework/Versions/5/QtGui (compatibility version 5.15.0, current version 5.15.2)
	/opt/homebrew/opt/qt@5/lib/QtCore.framework/Versions/5/QtCore (compatibility version 5.15.0, current version 5.15.2)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 1200.3.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1311.0.0)
	/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit (compatibility version 45.0.0, current version 2103.13.10)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1853.0.0)
	/System/Library/Frameworks/CoreGraphics.framework/Versions/A/CoreGraphics (compatibility version 64.0.0, current version 1557.1.1)
	/System/Library/Frameworks/CoreServices.framework/Versions/A/CoreServices (compatibility version 1.0.0, current version 1141.1.0)
	/System/Library/Frameworks/Foundation.framework/Versions/C/Foundation (compatibility version 300.0.0, current version 1853.0.0)
	/System/Library/Frameworks/ImageIO.framework/Versions/A/ImageIO (compatibility version 1.0.0, current version 1.0.0)
	/usr/lib/libobjc.A.dylib (compatibility version 1.0.0, current version 228.0.0)
```

kintel, 2022-01-02:
Now that #4005 is merged to master, it should be possible to build M1 binaries using the `sh scripts/macosx-build-dependencies.sh && cmake` method.
Universal binaries are next, but it's a fair bit more work, see #4011

closed 2022-01-02 by kintel

kintel, 2022-01-03:
For people interested in following progress on and/or helping test universal binaries, please follow #4036