openscad/openscad #3753: 'Save' and 'Save As' do not work in some circumstances.

lar3ry, 2021-04-30:
If I bring up an instance of OpenSCAD, and either write a program or paste one in, when I hit Save or Save As, it closes without saving. 

I am running it on Linux MATE V1.24.0, and this happens in both: 
OpenSCAD version 2021.01 (git 41f58fe)  and  OpenSCAD version 2021.04.13.ai7733 (git d1c445d4c) 

However, If I open with a file name (in each version); 
openscad foo.scad 
OpenSCAD foo.scad 

Save works fine, but Save As causes it to close without saving.

lf94, 2021-04-30:
Potentially related to https://github.com/openscad/openscad/issues/3706

lf94, 2021-05-01:
Also potentially related: https://github.com/openscad/openscad/issues/3539

Marabellum, 2021-05-08:
I experience the same problem with Linux Mint Cinnamon 20.1.
The problem occurs with several Linux kernels and also beta versions of OpenSCAD.
I never experienced this problem with OpenSCAD 2019.