openscad/openscad #3754: Openscad failing to run after windows update

github4657, 2021-04-30:
I ran openscad 4 days ago with no problem.
Today when I try to run it splash screen comes up with list of past files but if I click on new or one of the files the program just crashes with no message.
Tried downloading and install new version  and get the same problem.
Tried an old version from 2015 and get the same problem.
The only change I know of is that windows did an update last night.
Running windows 10 home
Version 2H1
OS build 19043.964
I am going to run virus scan. Is there any thing else I should look at?

rcolyer, 2021-04-30:
Can you try deleting or temporarily moving away your entire config directory for OpenSCAD?  A reinstall alone does not touch this.

github4657, 2021-05-01:
I can not find any config file but I think the problem is with my windows update.
I am finding the same problem with most of my open source programs???
kicad meshlab slic3r and freecad show similar results so I will contact Microsoft
I will let you know if I find a solution but thanks anyway.

t-paul, 2021-05-01:
Did you reboot after the update? There's been cases already where the update broke all sorts of graphics things but the issues went away after reboot.

github4657, 2021-05-01:
Microsoft had me run some programs to clean files and problem has gone away.
First time using github help and Iam impressed with service

I assume Close with comment will close this request.

Thanks
George

closed 2021-05-01 by github4657

t-paul, 2021-05-01:
Thanks for the update, good news! Would it be possible to share some short details of what programs that were? In case those tools are publicly available, it might help other people running into similar issues.

github4657, 2021-05-01:
Microsoft support had me run System File Checker (SFC) commands and then Deployment Image Servicing and Management (DISM) commands but I did not see any errors reported.
They had me delete all files in my temp file under my user name and I think this is what fixed the situation as this folder had gotten huge and the problem with every program was when trying to open a file.