openscad/openscad #3756: Crash during rendering on macos 11.3

bartheerman, 2021-05-02:
Content of crash report:

Process:               OpenSCAD [747]
Path:                  /Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD
Identifier:            org.openscad.OpenSCAD
Version:               2021.01 (2021.01.31)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           OpenSCAD [747]
User ID:               501

Date/Time:             2021-05-02 21:40:55.194 +0200
OS Version:            macOS 11.3 (20E232)
Report Version:        12
Anonymous UUID:        AD197481-AA55-E93C-3057-7BB70F260E86

Time Awake Since Boot: 530 seconds

System Integrity Protection: enabled

Crashed Thread:        8  QThread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called
terminating with uncaught exception of type boost::wrapexcept<boost::bad_any_cast>: boost::bad_any_cast: failed conversion using boost::any_cast

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff2030632a mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff2030669c mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff20434c6f __CFRunLoopServiceMachPort + 316
3   com.apple.CoreFoundation      	0x00007fff2043334f __CFRunLoopRun + 1328
4   com.apple.CoreFoundation      	0x00007fff2043275c CFRunLoopRunSpecific + 563
5   com.apple.HIToolbox           	0x00007fff28654203 RunCurrentEventLoopInMode + 292
6   com.apple.HIToolbox           	0x00007fff28653f65 ReceiveNextEventCommon + 587
7   com.apple.HIToolbox           	0x00007fff28653d03 _BlockUntilNextEventMatchingListInModeWithFilter + 70
8   com.apple.AppKit              	0x00007fff22c2eb32 _DPSNextEvent + 864
9   com.apple.AppKit              	0x00007fff22c2d305 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1364
10  com.apple.AppKit              	0x00007fff22c1f679 -[NSApplication run] + 586
11  libqcocoa.dylib               	0x000000010a06887e 0x10a041000 + 161918
12  org.qt-project.QtCore         	0x000000010796c397 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 487
13  org.qt-project.QtCore         	0x0000000107970ffc QCoreApplication::exec() + 364
14  org.openscad.OpenSCAD         	0x000000010526ae1a gui(std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > >&, boost::filesystem::path const&, int, char**) + 2362
15  org.openscad.OpenSCAD         	0x000000010526f6b5 main + 15317
16  libdyld.dylib                 	0x00007fff20356f3d start + 1

Thread 1:: com.apple.NSEventThread
0   libsystem_kernel.dylib        	0x00007fff2030632a mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff2030669c mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff20434c6f __CFRunLoopServiceMachPort + 316
3   com.apple.CoreFoundation      	0x00007fff2043334f __CFRunLoopRun + 1328
4   com.apple.CoreFoundation      	0x00007fff2043275c CFRunLoopRunSpecific + 563
5   com.apple.AppKit              	0x00007fff22db4baa _NSEventThread + 124
6   libsystem_pthread.dylib       	0x00007fff2033b954 _pthread_start + 224
7   libsystem_pthread.dylib       	0x00007fff203374a7 thread_start + 15

Thread 2:: Qt bearer thread
0   libsystem_kernel.dylib        	0x00007fff2030c9e2 poll + 10
1   org.qt-project.QtCore         	0x00000001079bf7be qt_safe_poll(pollfd*, unsigned int, timespec const*) + 94
2   org.qt-project.QtCore         	0x00000001079c1042 QEventDispatcherUNIX::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 722
3   org.qt-project.QtCore         	0x000000010796c397 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 487
4   org.qt-project.QtCore         	0x00000001077ba1ac QThread::exec() + 140
5   org.qt-project.QtCore         	0x00000001077bdf80 0x107790000 + 188288
6   libsystem_pthread.dylib       	0x00007fff2033b954 _pthread_start + 224
7   libsystem_pthread.dylib       	0x00007fff203374a7 thread_start + 15

Thread 3:: AMCP Logging Spool
[... 559 lines cut ...]
__LINKEDIT                       508.6M       77 
__OBJC_RO                         70.2M        1 
__OBJC_RW                         2480K        3 
__TEXT                           384.7M      447 
__UNICODE                          588K        1 
libnetwork                         128K        8 
mapped file                      255.1M       38 
shared memory                      768K       17 
===========                     =======  ======= 
TOTAL                              2.5G     1695 
TOTAL, minus reserved VM space     2.4G     1695 

Model: MacBookPro11,2, BootROM 431.0.0.0.0, 4 processors, Quad-Core Intel Core i7, 2 GHz, 8 GB, SMC 2.18f15
Graphics: kHW_IntelIrisProItem, Intel Iris Pro, spdisplays_builtin
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343531533641465238412D50422020
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343531533641465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x134), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1680.7)
Bluetooth: Version 8.0.4d18, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0256F, 251 GB
USB Device: USB 3.0 Bus
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.1

bartheerman, 2021-05-02:
The problem occurs when rendering:

```
difference() {
  linear_extrude(height=10)
  polygon([[-25, 0], [-50, 60], [50, 60], [25, 0]]);
  color("purple") translate([0, 25, -5]) AugerThread(12, 13, 20, 2);
};
```

where AugurThread comes from the threads library
https://github.com/rcolyer/threads-scad

MichaelAtOz, 2021-05-02:
I did this to get a look at the threads
```
difference() 
{
  translate([0, 25, -5]) 
    AugerThread(12, 13, 20, 2);
  cube(40);
};
```
The preview looks really dodgy.
F6 crashed, on Windows 7 & 2021.01. Exception code: 0x40000015 Fault offset: 0x000000000188e897

I'll have a quick look at threads.scad, but it looks complex...

Regardless, OpenSCAD shouldn't crash.

MichaelAtOz, 2021-05-03:
@bartheerman I note the outer-thread is smaller than the inner-thread, causing the dodgy look, are you just reporting the crash, or do you want other assistance. With outer-thread larger it doesn't crash, exports STL, but with two small self-intersections.

bartheerman, 2021-05-03:
@MichaelAtOz, indeed, that was the problem.

Thank you!

ChrisCoxArt, 2022-01-16:
Reproduced this crash in the current build.

ChrisCoxArt, 2022-01-16:
For the any_cast, the operand content is NULL.  So, the real bug is somewhere in the CGAL data structures.
But we can do a better job of catching exceptions in CGAL threads.

ChrisCoxArt, 2022-01-16:
Yeah, need to catch std::exception (in addition to CGAL::Failure_exception) in applyOperator3D.  It would be nice of CGAL caught boost/stdlib exceptions and converted them in their code - but they don't, so we have to clean it up.

And it would be a good idea to also catch exceptions at the worker thread root level in CGALWorker::work() before they escape and crash.

kintel, 2024-03-31:
Closing as this no longer crashes, but fails with an error:
ERROR: exception in CGALUtils::applyBinaryOperator difference: boost::bad_any_cast: failed conversion using boost::any_cast

closed 2024-03-31 by kintel