openscad/openscad #3763: Naming model in 3mf export

freiser77, 2021-05-06:
Hello,

I create a model in openSCAD and export it in 3mf format. When I import it in PrusaSlicer, the model is shown there as "OpenSCAD model".

Is there any way to set a name for the models in OpenSCAD?

Like
```
name("bottom") cube(10);
name("top") cube(20);
```

Greetings

t-paul, 2021-05-06:
No, there's currently no way to give a specific name.

closed 2021-05-06 by t-paul

freiser77, 2021-05-06:
Hello @t-paul,

why do you close the issue then?

Wouldn't it be worth a feature request?
It shouldn't be such an effort, should it?

Greetings

t-paul, 2021-05-06:
What you posted was not a feature request but a question, so I answered. There's already feature requests that are very much covering the core topic (not the specific 3mf name though).
And yes, multiple objects are a bigger change as the current processing results in a single object.