openscad/openscad #3775: Consider dropping macOS 10.9-10.11 support to update Qt and enhance experience on supported macOS versions

TheJKM, 2021-05-16:
By reading old commit messages, I saw the macOS builds of OpenSCAD still use Qt 5.9.X to ensure compatibility with older versions of macOS, currently down to 10.9 (Mavericks).

With the 2019.05 version, there were huge problems on Macs with enabled dark mode (#2785). In 2021.01, dark mode is disabled, but there are still display issues, e.g. wrong text colors on the tab selection (see images).

I suggest to update Qt to a more recent version with support for up to date versions of macOS:
- Qt 5.12 LTS works great (I use a custom build made with it) and supports macOS 10.12 Sierra (even this version is already EOL) and newer. As this is an LTS release of Qt, it should get some more updates.
- Qt 5.15 also works great, many Linux-distributions package OpenSCAD with it and I also had a custom build. It works on macOS 10.13 (High Sierra, also EOL already) and newer. It supports exactly the same hardware as Qt 5.12, because any Mac running macOS 10.12 can be updated to macOS 10.13. But, as the Qt Company announced recently, LTS releases are commercial-only.

The downside is, this removes support for some very old Mac hardware, because a few of the currently supported machines (capable of running macOS 10.9) can only be upgraded to macOS 10.11. I tried compiling Qt 5.12 for macOS 10.11 by patching the deployment target, but It does not work.

In case you want to update the Qt version, I can supply a Pull Request as I have the workflow running in your CI pipeline working with Qt 5.12 and Qt 5.15 locally.

Personally, I’d go directly to Qt 5.15, as it will not drop more hardware than Qt 5.12 will. Because there are much more Macs running current OS releases, and there are also reports about problems on older OS releases (#3267), I think updating Qt and increasing the experience on macOS is worth the support removal.

Related issues which would be resolved by updating Qt: #3758, #2785 

Tab selection in 2021.01:
[image]

UI in dark mode with Qt 5.15:
[image]

Scopeuk, 2021-05-16:
To put some context on penetration (potential impact)
https://gs.statcounter.com/macos-version-market-share/desktop/worldwide
Back to mojave gets 80% user base roughly
High Sierra takes that up to about 90%

t-paul, 2021-05-16:
The main issue is that there's currently no active developer on macOS. Right now I can still *run* the builds on my old MacBookPro (which is on 10.11.6 / ElCapitan) but I can't compile anymore. Upgrading to a newer Qt version will mean I can't even validate build from the CI systems anymore.

Still we need to find a way forward, so starting a PR that goes to latest Qt5.15 would be a very good start.

Note that this means support for the `macosx-build-dependencies.sh` build script as currently there's no solution to get homebrew builds running as release on any (supported) system.

So my suggestion for now is:
- I'll create branch for maintaining this topic while not in master
- Qt5 update PR would go into that branch and we can setup CI for that too
- Maybe we can even get some interrest for native M1 build too?
- Meanwhile I'll try to find a sustainable way of going full in committing to newest Qt5 and macOS 10.13 / High Sierra

Do you think that could work?

TheJKM, 2021-05-16:
Sounds like a reasonable plan.

I made my custom build using `macosx-build-dependencies.sh`. There are two dependencies (iirc) not building on my system, probably because I'm on 11.3 while the CI seems to run 10.15.7 (took those two from Homebrew for my build). Will see how I can fix those, it might require updating those dependencies as well to compile on 11.X. Or I'll try on 10.15.7. There certainly must be a way.

t-paul, 2021-05-16:
I've created branch `qt5-upgrade-macos` which should also have the CircleCI snapshot builds and the Github Release build workflow on macOS enabled (so we probably need to be a bit conservative with pushing changes).

It should be fine building on a newer OS, the critical part are those `-mmacosx-version-min` flags and that we can verify everything is built and linked against those libraries (e.g. `otool -L` pointing to the correct files and such).

TheJKM, 2021-05-16:
Alright, thanks! It might take a few days or even weeks, I'll comment here if there are results.

gingerbeardman, 2021-05-17:
I've tried to build in the past but got nowhere (see a related issue of mine). 

But I'm happy to try again, and would volunteer myself to check builds regularly (at whatever cadence we decide).

I run Big Sur (macOS 11) and Mojave (10.14.6).

gingerbeardman, 2021-05-17:
OK, so I downloaded the latest `qt5-upgrade-macos` build from CI and it runs fine. However, the issues with the tabs are _not_ resolved.

[image]

TheJKM, 2021-05-17:
There is nothing changed in the branch yet, so you won't see any difference just yet as it still uses the old Qt. Will see if I can open the first pull request later today.

On my local build the tab issue is resolved (see the screenshot above).

TheJKM, 2021-05-18:
Now the `qt5-upgrade-macos` branch is using Qt 5.15.2, and the CI finished building successfully. If you want, you can now try the new build @gingerbeardman

gingerbeardman, 2021-05-18:
Thanks, tabs are much improved (but not perfect)

[image]

Also, there's a crash on quit. Happens ever time. I quit using Cmd+Q key combo.

<details>
<summary>Details (click to expand)</summary>

Process:               OpenSCAD [86375]
Path:                  /Users/USER/Downloads/*/OpenSCAD.app/Contents/MacOS/OpenSCAD
Identifier:            org.openscad.OpenSCAD
Version:               2017.03 (2021.05.18)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           OpenSCAD [86375]
User ID:               501

Date/Time:             2021-05-18 15:59:01.292 +0100
OS Version:            macOS 11.3.1 (20E241)
Report Version:        12
Bridge OS Version:     5.3 (18P4556)
Anonymous UUID:        79BC9B93-EBFD-DCDA-C6A1-CA4356DD4E23

Sleep/Wake UUID:       94DA31C7-B8EE-4C19-8A66-D18601E76826

Time Awake Since Boot: 340000 seconds
Time Since Wake:       4400 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [86375]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.qt-project.QtCore         	0x000000011262fbc7 QHashData::nextNode(QHashData::Node*) + 7
1   org.openscad.OpenSCAD         	0x000000010fd9f4f8 gui(std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > >&, boost::filesystem::path const&, int, char**) + 2408
2   org.openscad.OpenSCAD         	0x000000010fda4a05 main + 15317
3   libdyld.dylib                 	0x00007fff203eff3d start + 1

Thread 1:: com.apple.NSEventThread
0   libsystem_kernel.dylib        	0x00007fff2039f32a mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff2039f69c mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff204cdc6f __CFRunLoopServiceMachPort + 316
3   com.apple.CoreFoundation      	0x00007fff204cc34f __CFRunLoopRun + 1328
4   com.apple.CoreFoundation      	0x00007fff204cb75c CFRunLoopRunSpecific + 563
5   com.apple.AppKit              	0x00007fff22e4dbaa _NSEventThread + 124
6   libsystem_pthread.dylib       	0x00007fff203d4954 _pthread_start + 224
7   libsystem_pthread.dylib       	0x00007fff203d04a7 thread_start + 15

Thread 2:
0   libsystem_pthread.dylib       	0x00007fff203d0484 start_wqthread + 0

Thread 3:
0   libsystem_pthread.dylib       	0x00007fff203d0484 start_wqthread + 0

Thread 4:
0   libsystem_pthread.dylib       	0x00007fff203d0484 start_wqthread + 0

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000103  rbx: 0x0000600001b27680  rcx: 0x0000935f6bd87680  rdx: 0x0000000000650600
  rdi: 0x0000600001b27680  rsi: 0x0000000112bdfa00  rbp: 0x00007ffedfe83550  rsp: 0x00007ffedfe83550
   r8: 0x0000000000008011   r9: 0x0000000000000000  r10: 0x0000000000002200  r11: 0x0000000000003fff
[... 455 lines cut ...]
MALLOC                           246.2M       44 
MALLOC guard page                   48K       10 
MALLOC_MEDIUM (reserved)         720.0M        6         reserved VM address space (unallocated)
MALLOC_NANO (reserved)           384.0M        1         reserved VM address space (unallocated)
OpenGL GLSL                        256K        3 
STACK GUARD                       56.0M        5 
Stack                             10.0M        5 
VM_ALLOCATE                        492K       27 
__DATA                            16.9M      401 
__DATA_CONST                      17.1M      204 
__DATA_DIRTY                       789K      117 
__FONT_DATA                          4K        1 
__GLSLBUILTINS                    5176K        1 
__LINKEDIT                       508.8M       69 
__OBJC_RO                         70.1M        1 
__OBJC_RW                         2480K        2 
__TEXT                           301.9M      399 
__UNICODE                          588K        1 
mapped file                      572.9M       37 
shared memory                      744K       13 
===========                     =======  ======= 
TOTAL                              2.9G     1422 
TOTAL, minus reserved VM space     1.8G     1422 

</details>

jottr, 2021-05-18:
I'd like to suggest to switch the homebrew [formula](https://github.com/Homebrew/homebrew-cask/blob/HEAD/Casks/openscad.rb) to install the binraies from the `qt5-upgrade-macos`-branch, once it has been confirmed working.

I'm glad to see that openSCAD receives some support to improve MacOS support.  
There's also https://github.com/openscad/openscad/issues/1524 which is another of those issues that need some form of resolution on MacOS (my [duplicate](https://github.com/openscad/openscad/issues/3482) has a few more suggestions).

t-paul, 2021-05-18:
This forumula is not maintained here.

It has to be noted that the `qt5-upgrade-macos` branch is against the development version, so I would not recommend publishing that as "normal" openscad installation in homebrew. The official builds have a sparate "openscad-nightly" which is meant to be usable in parallel.

TheJKM, 2021-05-18:
I don't know how the release planing is, but it shouldn't be that much effort to backport the new Qt version. On the other hand, the 2021.01 release is far more usable than 2019.05 if you have dark mode enabled.

Either way that's future, as I can reproduce the crash on closing via CMD+Q (and only via this way). I don't have that much time, but I'll search if I can find the issue.

TheJKM, 2021-06-25:
A month has passed now and I'd like to ask the maintainers about Merge Request #3784. I know it's a rather brutal solution, and, personally, I'm not happy with it. If you say, you don't want the loop in question to get removed, I'm totally fine with that. But it would provide clarity so I can search for a different solution to get this crash, which is not limited to macOS (as you can see from the merge request comment), fixed.

TheJKM, 2021-07-25:
Another month later, I decided to update my merge request to the solution proposed by @icecream95 in #3837.

As the segfault is not limited to macOS on newer Qt versions, I see this change as finished and ready, as soon as #3845 is merged.

To the maintainer team, if you have any other issues with my changes, let me know. I hope you found a solution to verify the new version, so the improvements can reach the users. If you would like me to back port the newer Qt version to the 2021-01 release as proposed earlier here, please also let me know.

kapfab, 2021-10-05:
Would love to have a try at the `qt5-upgrade-macos` branch.
Is the segfault issue the reason why it is not up-to-date with the latest changes from `master`?

t-paul, 2021-10-05:
That bugfix is already in `master`. I've rebased `qt5-upgrade-macos` to include everything from `master`, lets see if we still have enough build credits for the macOS build to finish. There's also some updates for macOS in progress which may interfere.

The main issue with dropping old macOS is that I'm the only developer left with macOS environment but that's an old MacBookPro and I'm going to lose even the ability to run those builds if we go to >= 10.13.

If we can get a couple of people tracking the nightly builds and help with reporting issues so those can be fixed not too long after appearing, I would be ok to move this to `master` even without having found a solution for running things myself yet.

TheJKM, 2021-10-05:
That's probably the best idea, I'm sure there are multiple people testing the nightly versions (including myself). Sadly, there is no way to run macOS with full graphics acceleration other than having compatible hardware.

kintel, 2021-11-02:
I opened a PR bringing Qt and QScintilla up to date: https://github.com/openscad/openscad/pull/3966
See build artifacts here: https://app.circleci.com/pipelines/github/openscad/openscad/1942/workflows/2002c9b5-dcb1-46a8-a6c5-a7f62c806f66/jobs/9232/artifacts

Built for macOS 10.13+

I did some casual testing on 10.13, 10.14 and 11.6 (rosetta), and seems to work well, but I didn't look for dark mode issues.
If anyone here have time to take the binary for a spin that would be awesome!

kapfab, 2021-11-02:
Great !
Ran a few tests, preview and rendering are working fine (11.6, Rosetta 2).
And dark mode is correctly handled (main UI and native dialogs).

TheJKM, 2021-11-02:
Just tried the new build, everything seems to work correctly. Looks good in dark mode! Tested on Monterey 12.0.1, Intel native.

t-paul, 2021-11-04:
Closing via #3966, there's a couple of build issues pending, but the fixes and Qt upgrade are merged. Thanks for the patches and testing.

closed 2021-11-04 by t-paul

eduarcor, 2022-03-24:
Where can I download a macOS version with dark theme UI support? Thanks

t-paul, 2022-03-24:
https://openscad.org/downloads.html#snapshots

eduarcor, 2022-03-24:
Gotcha, thanks. When is it supposed that will be part of the stable version?