openscad/openscad #3776: Model not rendering the same in latest CI compared with last 2021.01 release

gingerbeardman, 2021-05-17:
- 2021.01
  [image]
- latest CI build (`qt5-upgrade-macos` branch)
[image]

MichaelAtOz, 2021-05-17:
Please see:
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#How_do_I_report_bugs? 

Particularly
> If possible, a trimmed down script reproducing the issue

It helps if you can try that first.
Screenshots of partial code snips are hard to reproduce...
Is that a preview or a render?
Does the console say anything of interest?

gingerbeardman, 2021-05-17:
Sorry that my initial report was brief.

- no errors in console
- screenshots show Preview
- but Render is also affected

Here's a trimmed script:

```
s = 30;
h = s/2;
d = s*2;
w = h*4.5;
bodyc = "#333";
trim = "#333";

rr = 0;
rl = 0;
rside = 4;

rf = 0;
rb = 0;
rfwd = 4;

translate([-h,-d,0])
car();

module car() {
    rotate([-rfwd*rf+rfwd*rb,-rside*rr+rside*rl,0])
    translate([0,0,-rfwd*rb+rfwd*rf])
    translate([0,0,40]) // monster rise
    union() {

        difference() {
            body();
            
            difference() {
                // front/back windows
                color("#FFF")
                translate([-11,-11,52])
                cube([52,100,26]);

                // boot bottom
                color(bodyc)
                translate([-h, h, d])
                rotate([35,0,0])
                translate([0, -37, -5])
                cube([s*2,h*0.4,h/2]);

                // windscreen bottom
                color(bodyc)
                translate([-h, d*1.82265, d*1.15])
                rotate([55,0,0])
                translate([0, -37, -5])
                cube([s*2,h/3,h/3]);
            }

            // side windows
            color("#FFF")
            translate([46,105-s,40])
            rotate([90,0,-90])
            linear_extrude(0,0,62)
            polygon(points=[[-11,12],[34,12],[34,38],[7,38]]);

            // rear side windows
            color(bodyc)
            translate([46,105-s*2.75,39.99])
            rotate([90,0,-90])
            linear_extrude(0,0,62)
            polygon(points=[[-15,12],[40,12],[40,48],[-15,48]]);

            // cut out
            color("#FFF")
            translate([-11,-11,40])
            cube([52,45,30]);
        }
    }
}

module body() {
    color(bodyc)
    union() {
        difference() {
            body_main();
            
            // windscreen angle
            translate([-s,s*1.75,s*3.5])
            rotate([-55,0,0])
            cube([s*3,s*2.2,d]);

            // bonnet angle
            translate([-s,s*3.01,s*1.7])
            rotate([-3,0,0])
            cube([s*3,d,d]);
        }
    }
}

module body_main() {
    // body box
    color("#FFF")
    translate([-h, -h, h])
    cube([s*2,s*5,s*2.2]);
}
```

t-paul, 2021-05-17:
I suspect a regression in `linear_extrude()`. Using as `linear_extrude(0,0,62);` assigns the values to the wrong parameters, specifically `height` gets the first 0.

Workaround till this is fixed is probably just `linear_extrude(62);`.

lf94, 2021-05-17:
I would suggest to always try to use the named parameters to prevent this happening in the future.

jordanbrown0, 2021-05-17:
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Using_the_2D_Subsystem#Linear_Extrude

says
> You must use parameter names due to a backward compatibility issue.

t-paul, 2021-05-17:
True, that would have prevented this specific issue, but still the positional parameters are a defined feature and must not randomly break. The problem here is that `linear_extrude()` tries too hard by detecting deprecated calls (using a file name as first parameter) while allowing for "normal" use with an integer as height as first parameter.

So while that general warning in the manual is probably useful, it's still a regression that we should try to prevent unless there's really no way out and then the documentation would need to be even more clear of what problems there are.

gingerbeardman, 2021-05-17:
I have a lot of models, I will go back through them at some point and use named parameters.

It would be good if the note about backwards compatibility mentioned the point at which things were broken. Obviously at some point since 2021.01

jordanbrown0, 2021-05-17:
I think the timeline is something like:
- At some point prior to 2012, linear_extrude had a set of positional parameters.  Exactly what those parameters were is no longer documented.
- At some point prior to 2012, those positional parameters were deprecated in favor of the current parameters, and the original parameters were removed from the documentation.
- In 2012, it was observed that you could not use the "obvious" positional parameters based on the documentation; you had to use the named parameters.  This note was added to the documentation.  Positional parameters based on the original syntax still worked, but were not documented.
- Recently (it seems) that backward-compatibility mechanism got broken and so those original positional parameters no longer work.

jordanbrown0, 2021-05-17:
I don't find documentation for those original parameters even going back as far as 2010, though maybe they were documented in some other book.

There are hints that linear_extrude might have been more tightly tied to DXF import; it and rotate_extrude are referred to as "DXF Extrusion statements".

gingerbeardman, 2021-05-17:
Thanks @jordanbrown0

redlizard, 2021-05-18:
Well, turns out I broke it when introducing the new and improved argument parser. linear_extrude() has some pretty quirky argument parsing special cases to remain bug-compatible with historic behavior, and I almost kept that intact but not quite.

Patch above should fix it.

PR #3779 (2021-05-20): Repair linear_extrude argument parser broken earlier during some overzealous simplification.
Fixes #3776.
closed 2021-05-20 by t-paul