openscad/openscad #3781: $variables don't work as they used to. Their scope seems to have changed.

nophead, 2021-05-19:
OpenSCAD 2021.05.11.ci7927  breaks my library. OpenSCAD 2021.03.24.ci7656  was the last snapshot I tried that worked.

In my `global_defs.scad` file that I include in every file in my library and project I have constants like these:

```
$_bom           = is_undef($bom)             ? 0      : $bom;             // 0 no bom, 1 assemblies and stls, 2 vitamins as well
$exploded       = is_undef($explode)         ? 0      : $explode;         // 1 for exploded view
```

Previously I could set `$bom` and `$exploded` on the command line and it would change the values of `$_bom` and `$exploded` throughout the project and used modules from the library. Now it doesn't seem to have the same effect on my library modules. Some BOMs are not produced. Also setting $explode at the end of my main file doesn't work but setting it at the top before my library is included does. Also this doesn't work anymore:

```
                let($explode = 1)
                    spring_washer(w);
```

It formally showed the spring washer exploded but now show it compressed. I.e. in the library module being called it doesn't see `$exploded` set to 1.

nophead, 2021-05-20:
My washer are drawn correctly with version OpenSCAD 2021.05.01.ci7809 but are wrong with version OpenSCAD 2021.05.03.ci7815. The only two commits between those two are d9491713914e37dbe04ec9f748c5111e7294bd9e and 9317cd98b15d3270d753ed7e6fecda024437cb3c, which are just additional warnings, so I can't see how they broke special variables.

nophead, 2021-05-26:
I tried to set up a simple test case to show this bug but found $variables are totally broken. With a simple test like this:

dollars.scad:
```
$v = 0;

module showv() {echo($v);}
```

test_dollars.scad:
```
use <dollars.scad>

showv();
```

The expect result and the result with OpenSCAD 2021.05.01.ci7809:
```
Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
ECHO: 0
```

With OpenSCAD 2021.05.03.ci7815:
```
Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
WARNING: Ignoring unknown variable '$v' in file dollars.scad, line 3
TRACE: called by 'echo' in file dollars.scad, line 3
TRACE: called by 'showv' in file test_dollars.scad, line 3
Execution aborted
 ```

So ```showv()``` can no longer see the $v defined in the same file when used by another file. This is a different bug to the one I first posted but shows $variable handling has completely changed between these two versions.

nophead, 2021-05-26:
Although github shows a linear list of commits I think the network view shows these two versions were on different branches before merging, so include a lot of changes. Possibly the context cleanup merge is responsible.

kim8823, 2021-06-05:
Just updated to 2021.01 (from 2015.something on this machine) and ran into a similar (same?) issue passing special variables to `children()`.

Variable assignments seem to use a different scope than other code.

```
$special = "default value";
module modifySpecial()
{
  $special = str($special, " modified");
  children();
}

modifySpecial()
{
  echo($special); // ECHO: "default value modified"
}

modifySpecial($special = "new value")
{
  echo($special); // ECHO: "new value modified"
}

modifySpecial()
{
  // the doc mentions echo being special to allow debugging recursive functions, so let's try to introduce another variable
  normal = $special;
  echo(normal); // ECHO: "default value"
}

modifySpecial($special = "new value")
{
  normal = $special;
  echo(normal); // ECHO: "default value"
}

modifySpecial()
{
  // maybe echo isn't really special, but its just how parameters work?
  let (normal = $special)
  echo(normal); // ECHO: "default value modified"
}

modifySpecial($special = "new value")
{
  let (normal = $special)
  echo(normal); // ECHO: "new value modified"
}
```

The reassignment of `$special` in the module is not necessary to show the issue, but it is what I ultimately want to achieve and it helps distinguish from the outer scope's default value. Using `let` for the reassignment doesn't change it. I feel like this only works because of the issue at hand, but I don't understand it, as its a parameter to `str` in any case...

If I make `$special` a parameter of the module, its obviously no longer able to be picked up from scope. Using it as a parameter still works fine, but the variable assignment doesn't even see it. Assigning the new variable with `let` still works, as it is in a parameter context again:

```
module modifySpecial($special = "default value")
{
  $special = str($special, " modified");
  children();
}

modifySpecial()
{
  echo($special); // ECHO: "default value modified"
}

modifySpecial($special = "new value")
{
  echo($special); // ECHO: "new value modified"
}

modifySpecial()
{
  normal = $special; // WARNING: Ignoring unknown variable '$special'
  echo(normal); // ECHO: undef
}

modifySpecial($special = "new value")
{
  normal = $special; // WARNING: Ignoring unknown variable '$special'
  echo(normal); // ECHO: undef
}

modifySpecial()
{
  let (normal = $special)
  echo(normal); // ECHO: "default value modified"
}

modifySpecial($special = "new value")
{
  let (normal = $special)
  echo(normal); // ECHO: "new value modified"
}
```

I've had some more trouble with the variable being gone while writing the test case (moving the default value and module to a different file to `use`), but I can't seem to reproduce it now. Maybe it was just the variable assignment in the child that threw me off before figuring it out, `use` doesn't make a difference with both variants.

nophead, 2021-06-05:
Yes I think that is a much older bug related to this one: #2200. The file scope variable is beating the dynamic scope one.

jordanbrown0, 2021-06-06:
This looks like a behavior I noticed last year.  Here's my message from 10/23/2020 ("interesting" scoping behavior for $ variables) analyzing it a bit.
---

When playing with my "tracking" transformation modules, I discovered an ... interesting ... scoping behavior of $ variables.  It seems undesirable, but I don't immediately see how to fix it.

What "should" the output of this program be?

    $x=1;

    module foo() {
        $x=2;
        children();
    }

    foo() {
        x = $x;
        echo(x=x, $x=$x);
    }

I expected:

    ECHO: x = 2, $x = 2

What it outputs:

    ECHO: x = 1, $x = 2

The reason is that the assignments for foo's child block are executed before any part of foo itself is executed, and then the modules for foo's children are executed when they are referenced with children(), after foo's own assignments and during its module execution.

This is a problem for my world-coordinate work, because it means that a reference to the local matrix inside the top level of a transformation's child block sees the transformation matrix *before* that transformation.  In my implementation:

[[ Note:  The following examples are based on an immediately-previous message where translate() has been replaced by an alternate version that records transformations in a $ variable. ]]

    translate([10,0,0]) {
        p = whereis([0,0,0]);
        echo(p);
    }

yields [0,0,0] instead of [10,0,0].  Revar's implementation suffers a similar fate:

    translate([30,40,50]) {
        p = local_translation();
        echo(p);
    }

also yields [0,0,0].

There's a workaround:  if the assignment is not in the top level of the child block then it doesn't get done until the children are referenced:

    translate([10,0,0]) union() {
        p = whereis([0,0,0]);
        echo(p);
    }

but I don't immediately see a way to work around it transparently, because the parent module doesn't get any chance at all to execute before the assignments in the top level of its child block are done.

I suspect that this would be tough to change, because (a) the child-block's assignments need to be executed before any of the children are executed and preferably only once, and (b) the parent module may do $ assignments during its own module-execution phase.  Perhaps the answer is that the child-block's assignments could be done on each call to children(), but that seems painful.  (But it also seems more correct, since different invocations of children() could have different values set into $ variables.)

I haven't a clue whether a "native" implementation of the world-coordinate-transformation-tracking mechanism could avoid this by doing its work before the child-block's assignments.

t-paul, 2021-06-06:
Dev snapshot outputs: `ECHO: x = 2, $x = 2`. That change likely introduced the issue @nophead reported here.

nophead, 2021-06-06:
Basically my library has worked for years and before that Mendel90 but something changed on May 3rd this year which means I can't use a snapshot later than that.

As well as the bugs raised in this issue there are at least two long standing bugs. One is file scope $variables in a used module beat dynamic scope ones, which is wrong, although now they seem to be ignored completely, which is also wrong. Copying a dynamic variable into a normal one fails in the top level of a child.

To get around @jordanbrown0's problem I have been using let. E.g.

`
$x=1;

module foo() {
    $x=2;
    children();
}

foo() {
    let(x = $x)
        echo(x=x, $x=$x);
}
`

nophead, 2021-08-22:
Is this ever going to be fixed? I think main thing that has regressed is that $variables are ignored in used files. That makes no because they are not ignored if the file is opened on its own, only when used by another file. Ordinary variables work in both cases. How can $variables having less scope than ordinary variables make sense?

jordanbrown0, 2021-09-05:
Q.  When *should* $ assignments in "used" files be processed?  Over what scope should those assignments be visible?

A case that's in some ways a little more interesting is:

1.scad:
```
use <2.scad>

$v = "1.scad file scope";

union () {
    $v = "1.scad union scope";
    foo();
}
```

2.scad:
```
$v = "2.scad file scope";

module foo() {
    echo($v=$v);
}
```

So when we execute foo, we have to decide which $v to use:  the one from the file, or the one from the caller.

2021.01 uses the one from the file.  The development snapshot uses the one from the caller.

Ignoring compatibility, I'd say that the new behavior is more right.  The rule for $ variables is that values are passed from your caller, and for a "used" file the file scope is not the caller.  It isn't clear to me whether $ variables set at the file scope in a "used" file should be used for anything other than perhaps other file-scope assignments, because they aren't in the call stack.

It seems to me that the only way to regard $ variables set in the file scope as being part of the call stack is to require that all file-scope assignments be processed as part of any call into a module in that file, and although that's what happens now I thought everybody was pretty agreed that it was a bad thing.

(It does surprise me a bit that $ variable assignments are somehow *excluded* from that behavior.)

Whether compatibility trumps that arguably-more-consistent behavior... I don't know.

niclash, 2021-09-05:
Yes, this is the punishment of ad-hoc programming languages compared to formally defined ones.

nophead, 2021-09-05:
Yes the value in 1.scad should override the one in  2.scad but if it is not present the value in 2.scad should be used. It makes no sense at all to ignore it completely and it totally breaks my library.

$variables don't just have dynamic scope up the call stack. They still have lexical scope like normal variables and $v is in scope when if you load 2.scad but not if you use it, which is bonkers.

jordanbrown0, 2021-09-05:
$v is in scope if you load 2.scad, because if you load 2.scad the file top level is part of the call stack.

nophead, 2021-09-05:
Yes but it should always be in scope inside 2.scad. It is in a block that lexically is in scope. So if the variable was called v it would be found. It is a bit late to decide $variables no longer have lexical scope when I have spend the last 10 years using them that way.

jordanbrown0, 2021-09-05:
I suspect that they never had lexical scope - that, instead, the file scope in the "used" file was sort of inserted as a level of the call stack.

As you say, compatibility is an excellent argument for maintaining that behavior.

You've also pointed out that performance for "used" files can be poor, because the file-scope stuff is re-evaluated on every call.  That issue seems like it might be directly tied to this one.

What should this produce?

--- 1.scad ---
```
use <2.scad>

$v = "1.scad file scope";

union () {
    $v = "1.scad union scope";
    foo();
}
```
--- 2.scad ---
```
x = $v;

module foo() {
    echo(x=x);
}
```
---

If you want it to produce "1.scad union scope", that means evaluating all of the file-scope stuff on every call.  Avoiding that would suggest that it should either produce "1.scad file scope" (processing them at "use" time, or something like it) or perhaps undef (saying that $v at file scope in 2.scad has no value, or that $v at file scope has no value since it wasn't yet defined at the time of the "use").

I don't have a One True Answer, but it seems like we have several semantics that individually are reasonable, that when combined produce undesirable results.

It would be interesting to know exactly what change caused the change in behavior, and exactly what the author's intent was.

nophead, 2021-09-05:
I think it was broken by @redlizard's commit https://github.com/openscad/openscad/commit/0ff3ca2fd89996a9b6fc1fccb411ef7585620a7a titled context cleanup. Basically refactoring.

Of course it should  echo "1.scad union scope". And yes that does require evaluating anything that depends on $variables that have changed but it has always re-evaluated all expressions regardless. How can it make sense to evaluate all the normal variables but ignore the $variables?

If this doesn't get fixed what am I to do? Forever use the last release and tell my library users the next release is not supported? Or should I fork OpenSCAD and try to get the latest fixes without context cleanup?

Currently I use the May 1st snapshot to get the VBO fast rendering that I need for big projects as that is the last one that works with my library but the customiser is broken and double right click produces a black window and I have to restart OpenSCAD. I don't use double right click but it seems all Microsoft mice have switch bounce on the right button because I am in the UK now but I had the same issue in my winter residence in Tenerife. Two different mice but probably the same model. No other applications I know of use double right click.

jordanbrown0, 2021-09-06:
It is indeed strange that it processes the file-level $-variable assignments (on each call!) and they are visible as it processes the other file-level assignments, but then those values are not visible to the modules as they execute.  I don't know beans about the implementation to know how that could happen.

But more generally...

I would not expect it to try to figure out which assignments to process based on what's changed.

If it was my decision, I'd probably accept the incompatible change and process all of the file-level assignments once only, approximately at "use" time, to address the performance problem.  There would be one-time pain for existing users, but everybody would then see the performance improvement.  There would then be a question of where those file-level $ variables fit into the call stack, whether they fit in somewhere near the other file-level $ variables or as another call layer between the caller from another file and the callee in this file.

nophead, 2021-09-06:
A performance increase is useless if it no longer processes the language. It would be a lot faster if cylinder was a no op but  useless for making geometry, Ignoring variables that have been used for a decade is just as bad.

Tell me how this makes sense:

[image]

[image]

I am surprised this regression has not been caught by the tests but I guess most are standalone rather than using other files.

MichaelAtOz, 2021-09-10:
I don't think this needs debate. It's broken, just needs to be fixed.
Should someone then want to propose an intentional change to behaviour, that needs to be discussed.

nophead, 2021-09-10:
Indeed. I will edit the title to be more explicit now I have narrowed down that $variables are simply ignored in used files.

It has been broken since May and I can no longer use later snapshots that have other bugs fixed and useful improvements like axis scales going to infinity.

rcolyer, 2021-09-11:
It seems to me that the previous behavior was actually a bug caused by the structural issues with the code that were being cleaned up.  The intended behavior of special variables seems pretty well specified in the manual and in all the discussions about them, as dynamically scoped down the call stack.  I think it wasn't in the testing suite because it wasn't supposed to happen.  But rather than try to maintain past-bug compatibility, here is a simple fix for obtaining the same behavior.  You simply need to bring the conditional into the dynamic scope where it is utilized.  This should be a simple formulaic swap, and results in similarly compact usage and code:

used.scad:
```
// Worked in old version:
$bar = is_undef($bar) ? 0 : $bar;
// Correctly scoped, works in all versions:
function baz() = is_undef($baz) ? 0 : $baz;

module foo() {
  echo($bar);
  echo(baz());
}
```

main.scad:
```
use <used.scad>
  
foo();
```

MichaelAtOz, 2021-09-11:
I believe @nophead smooth() example is the nail in that coffin.
The behaviour shown there is just wrong, and it should not need tricks to work-around it.

rcolyer, 2021-09-11:
I've implemented modules like smooth some time ago, but honestly it never occurred to me to try a file scope since dynamically scoped variables are not supposed to work like that.  Applying a default value should be done in module with either a direct value (fixing it inflexibly), or in a preset value dependent manner like shown here:  https://github.com/rcolyer/smooth-prim/blob/master/smooth_prim.scad

A third option is setting the $fa and $fs default values in the module parameters, but this has different logic in that it overrides the caller's call stack values unless explicitly overridden at the time of call.  To make a file scope setting dynamic would require reprocessing everything in a scope that a dynamic variable shouldn't even look at just to guarantee the file scope can also be dynamic to the call stack, which is horrendously unnecessary for a declarative language.  We in fact had quite significant performance bugs that were caused by this exact thing.  Working with the specified scopes for the two variable types is the right solution, and there seems to be a straightforward way to do that for every example.

MichaelAtOz, 2021-09-11:
I still have OpenSCAD 2013.06 installed, it uses $variables in the same way as 2021.01.

There are many years worth of .scad files out there with default values defined at the top level.
It is unknown how many of those would use $variables, but I would hazard a guess that good chunk would use one or more of the $fX variables.

Requiring the future use of each such library to be recoded is not the right path.
At least not in one step.
If you want to properly change entrenched behaviour you would start by depreciating such use.
Noting `assign()` is still there, been deprecated for an awful long time.
Or you could use a pragma, such a facility could be useful for other language feature transitions and library maintenance.
Or, I don't favour this approach, a Preference.
Historically, OpenSCAD has paid close attention to backward compatibility, I don't think breaking behaviour in one release step is warranted.

MichaelAtOz, 2021-09-11:
Though `use<>` has always been a outlier with clunky behaviour, perhaps it is time to deprecate it, and build a better mechanism to 'incorporate' modules and manage defaults.
There has been intermittent discussion of language features to enhance library versioning and package management.

rcolyer, 2021-09-11:
Well, the manual in 2013 described special variables with the identical language to today, as variables that strictly obtain their value from the call chain.  https://en.wikibooks.org/w/index.php?title=OpenSCAD_User_Manual/The_OpenSCAD_Language&oldid=2555425#Special_variables_2

I googled meticulously for OpenSCAD "file scope", to see how how prevalent this notion was.  But a little to my surprise, 100% of the pages I could find online were nophead describing it in some past discussions.  I think what happened was he found what should have been called a bug, but interpreted it as intended behavior a while ago, giving it a name, and thus it never ended up as a bug report until redlizard resolved this issue in his clean-up of that code.

A bit in jest, I'm not sure how many years you have to have an implementation bug before you can call it a feature.  I have a lot of concerns about this as a strategy when there was an announced, named, and intended behavior for the dynamic variables, which persisted in the documentation and language labeling them for a very long time.  I don't think it's neutral to try to reinsert the other behavior.  Aside from the extra repeated evaluations required to support it, it's overall not good for variables to have surprising and unexpected ways to pick up values.  There were actually quite a few cases of variables picking up out-of-spec values that were resolved by the #3745 clean-up (some with active and very old bug reports, such as #1968, ancient enough for people to have code that relied on those misbehaviors), and this was just one of them.  Surely we don't want a standard where we go back and reinsert all the surprise behaviors just because they previously existed.

Ultimately, I think we have a clear and simple way to write everything discussed here with the two variable types that matches the label on the can.  And we should be aiming for a self-consistent well-defined language.

nophead, 2021-09-11:
Unfortunately I have built a library and even ran a business on the old behavior that has been present since the start of OpenSCAD.  What am I supposed to do now? Have two versions for old and new OpenSCAD?

How does wrapping the variables in a function prevent the need to reevaluate all the expressions in the used code? Those expressions could use the same function that uses the $variable but stored in a normal variable.

How does it make sense that if you have
a = 5;
$b = 6;

function c() echo(a, $b);

You get $b not found?

$variables have always worked like normal variables in addition to propagating down the call stack. I don't see anything in the documentation that says file scope variables will be ignored in used files.

rcolyer, 2021-09-11:
There is no file scope behavior to ignore.

This has no output at all:
```
a = 5;
$b = 6;

function c() = echo(a, $b) 0;
```

On the other hand, this outputs 5, 6 not because of anything to do with where c() is defined, but entirely because of where c() is called:
```
a = 5;
$b = 6;

function c() = echo(a, $b) 0;
x = c();
```
Likewise for this case, it does not matter in the slightest what value $b has where c is defined.  It again, picks it up purely from where it is called:
```
a = 5;
$b = 6;

function d() = let($b = 20, c = function() echo(a, $b) 0) c;
c = d();
x = c();
```

And let us take one final case, just like the used file one:
```
a = 5;

function d() = let($b = 20, c = function() echo(a, $b) 0) c;
c = d();
x = c();
```
Here, we get the following output:
```
WARNING: Ignoring unknown variable '$b' in file file30GBNP.scad, line 3 
ECHO: 5, undef
```
Why in this last example was the $b = 20 completely ignored?  Because it did not exist at all in the call stack for c().  There was no fallback behavior to lexical scope, and the location of definition was completely irrelevant.  Identically, in the case of a used file, a $b = 20 in a used file where c() was defined would not exist at all in the call stack for c(), and should not be picked up there either.  It must remain undefined, just like the case of the function context.

There's no need to abandon the library, bifurcate the library versions, or collapse a business.  All you need to do is swap out the variables for dynamic function calls like the example I showed a few posts up, and it works on all openscad versions old and new.  Basically, just fix up that portion of the library code to follow the dynamic scope that always existed, get the same behavior out of it, and it's all resolved.

MichaelAtOz, 2021-09-11:
This is about special variables in the context of `use<>`.
OpenSCAD has no specification, historically the behaviour of the program has frequently been the basis adjustments to the wiki. When I joined the wiki was pretty rudimentary, with plenty of undefined situations and it was built up documenting the program behaviour. Anyway...

`use<>` has always had unique scoping.
Variables defined at the top level of a `use`'d file provide the default value within the scope of the file.

Now you are saying that normal variables at the top level are valid, but special variables are not.

Going to a situation where...
Oh, I was just going to say what Nophead just posted.

The correct way to do this is to leave the `use<>` behaviour as it was, (eventually) deprecate it, and replace it as part of changes needed to better support library versioning and packaging. `use<>` was always a kludge along with the '-D variable=val' causing all the variables redefinition confusion.

The work-around, filling a file full of `is_undef() : ?` for defaults is even a worse kludge.

rcolyer, 2021-09-11:
There's nothing "unique" about normal variables in a used file providing the default value.  This is just lexical scoping, like anywhere else.  They're picked up, lexically.  There's also nothing "unique" about special variables obeying dynamic scope now, as they do so everywhere else.  This issue report is simply highlighting one case where dynamic scoping rules were violated and now aren't.

I didn't introduce the "is_undef() ? :", so this is not a workaround.  The failure of the "is_undef() ? :" assignment result being picked up from a lexical scope was in the opening post of this issue.  I simply suggested that if the desired behavior is evaluating the is_undef sequence based on a special variable, then that desired behavior is correctly obtained with the same compactness by not assigning it to something in the outer lexical scope, but by doing so in a function call that will evaluate dynamically, and thus pick up dynamic scope.

nophead, 2021-09-11:
But that would require replacing nearly all variables in used files with functions. Lots of expressions depend on my global variables, either directly or indirectly. Consider something like extrusion width or layer height: nearly all printed part dimensions depend on those. So in a reusable printed part in the library it may have local definitions like length and width that are complex expressions with some dependency on the extrusion parameters. These now don't work if the file is used.

So basically all the variable assignments would need to be replaced with function calls and there would be a tree explosion of function calls to mimic the current re-evaluation of expressions but it would be done in user land, instead of C++, so be much slower.

MichaelAtOz, 2021-09-11:
The issue is that having $variables as defaults in a `use<>`d file, no longer work as a library.
You will be required to have them **all** defined in the calling file as well.
You may as well introduce .h files.

I'll reread all this tomorrow in case I missed something. It's getting to that time of day when my head is changing scope.

MichaelAtOz, 2021-09-11:
> So basically all the variable assignments would need to be replaced with function calls 

`:)` functional language...

nophead, 2021-09-11:
Actually when copied into a normal variable its value is captured and accessible. Why can't it simply search the $variables in the used file scope after all the others, i.e. just before giving the not found error? I can't see how that breaks anything. It seems more logical and is more backwards compatible. 

People will still get a surprise from the smooth sphere behaviour changing. That is because the built in $variables always have a default value, so will override any default given in a used file.

rcolyer, 2021-09-11:
> But that would require replacing nearly all variables in used files with functions.

I think that's a bit hyperbolic.  Lexical scope is still picked up exactly the way it always has been, lexically.  It is just that if you want to pick up dynamic scope, you have to do it dynamically.  But this has always been true everywhere else in scad code as well, where in-module code has been the way to make a modification down the chain for one module call, while functions have always been the go-to method for making a value modification in many locations following a formulaic process.  If you want to make a value modification in multiple locations of the code in different dynamic scopes, then a function call is the logical way to do that.  This is a functional language where this should be the norm.  Getting a modified value from a function in a functional declarative language is not a usage hazard, it's the intended process.

> Actually when copied into a normal variable its value is captured and accessible. Why can't it simply search the $variables in the used file scope after all the others, i.e. just before giving the [...]

It's not more logical to grab from the file.  In fact what it breaks is the logic of dynamic scoping.  There's a clear, consistent, and predictable expectation of dynamic scope: Values come from the call chain, always.

> People will still get a surprise from the smooth sphere behaviour changing. That is because the built in $variables always have a default value, so will override any default given in a used file.

That's simply not how to make a smooth sphere, but I demonstrated with an example of doing it that there are trivial ways to make a smoothed sphere.  In fact, it's disruptive to code architecture to even have that behavior with the $f built-ins, precisely because if you were to wedge a bonus file scope look-up in between the dynamic scoping, you suddenly end up with $f values getting picked up in a used file that you cannot override with the dynamic scope of the caller!  The whole point of dynamic scoped variables is for the caller to control the values up to the point of entering a module or function, so that you can do things like have the caller specify the desired smoothness for all objects down the chain unless it is overridden somewhere IN the chain.  In fact this should have been submitted as a bug report, but was missed for a while.

This is not neutral, and a fallback of wedging in a pass through file scope violates the premise of the dynamic scope in ways that can be harmful.  It is acceptable practice to throw demos, like testing suites, into a used file, which should not impact use<> usage as only the modules and functions are picked up.  Here is an example of surprising and wrong behavior in 2021.01, which is fixed in the master branch:

main.scad:
```
use<translator.scad>
  
$fa=1;
$fs=0.4;

DoubleTranslator([0, 10, 0]) sphere(20);
```

translator.scad:
```
// Translates children by twice the vector.
module DoubleTranslator(v) {
  translate(2*v) children();
}

// Pentagon demo:
$fn = 5;
DoubleTranslator([10, 0, 0]) cylinder(r=10, h=2);
```

In this example, DoubleTranslator, which does nothing but translate, turns the smooth sphere from main into a blocky mess in 2021.01!  This happens even though DoubleTranslator is not supposed to do anything of the sort, it generates no geometry objects of its own, and none of the call chain does anything of the sort.  It's a bug.  Fortunately, it works okay in the master branch.

nophead, 2021-09-11:
> Getting a modified value from a function in a functional declarative language is not a usage hazard, it's the intended process.
In C++ I wouldn't hesitate wrapping a constant in a function because I know the compiler will optimise it away. But  OpenScad is effectively an interpreted language and wrapping constants in functions makes it is slower. I know because I have tried do that in the past to mitigate all constants being re-evaluated and found it gets much slower.

>Values come from the call chain, always.
Says you but that has never been the case for the last 10 years. How can such a breaking semantic change be simply a merge of tidyup with no notification? Why is it not OpenScad 2 like Python 3?

>precisely because if you were to wedge a bonus file scope look-up in between the dynamic scoping, you suddenly end up with $f values getting picked up in a used file that you cannot override with the dynamic scope of the caller! 

That is why I use the `is_undef()` guard.  So I can have defaults in my library that can be overridden by the caller.

>It's a bug. 

No it works as OpenScad is designed as far as I am concerned. If you want it to work as intended do:
```
let($fn =5) DoubleTranslator([10, 0, 0]) cylinder(r=10, h=2);
```

Tell me how the following make sense and is there any other computer language that has purely dynamic scope variables?

callee.scad
```
a = 1;
$b = a + 1;
c = $b + 1;

module x() {
    echo(a, c);
    echo($b);
}
```
caller.scad
```
use <callee.scad>

let($b=42) x();

x();
```
```
ECHO: 1, 3
ECHO: 42
ECHO: 1, 3
WARNING: Ignoring unknown variable '$b' in file callee.scad, line 7
TRACE: called by 'echo' in file callee.scad, line 7
TRACE: called by 'x' in file caller.scad, line 5
Execution aborted
 ```

rcolyer, 2021-09-11:
> How can such a breaking semantic change be simply a merge of tidyup with no notification?

I don't consider it a breaking semantic change any more than the other bug fixes we do on edge cases of variable lookups, because special variables are specified to be dynamic in scope, and have behaved like this in all contexts except for some places where there were bugs found and fixed.  It's unfortunate you have some code that was using this out-of-scope behavior as a feature, but you can definitely fix up that code with the dynamic scope approach shown above, and it will still work the same.

While earlier today looking into the performance issues that resulted from this bug, we dug up this 2014 issue #782 where kintel remarks, "This is not intended - top-level expressions in use'd libraries should only be evaluated once.  Whether models in the wild actually depend on this is a good question."  Your code and examples you are showing are specifically relying on this automatic repeated top-level evaluation bug which should never be happening in a declarative language, and is on the agenda to hunt down and purge now that the scopes are behaving properly.  This should speed up ALL usages of use<>.

> Tell me how the following makes sense

It is doing exactly what dynamically scoped variables do.  It sincerely makes perfect sense to me.  The variable does not evaluate to anything because it was not set anywhere within the call stack.

> and is there any other computer language that has purely dynamic scope variables?

Yes, many.  For example there are the lisps, and there's a nice bash example on this wiki page, which also lists other example languages:  https://en.wikipedia.org/wiki/Scope_(computer_science)  What I can't think of off the top of my head is any language that puts a lexical lookup layer in the middle of dynamic scope lookups.  As far as I know this is not done anywhere, because it is inconsistent, and would result in very surprising side-grabs of values like the DoubleTranslate example.

nophead, 2021-09-11:
Yes it was a bug to put the lexical lookup in the middle, it should be at the end though. Why can you see c that depends on $b but not $b?

rcolyer, 2021-09-11:
Because it is an entirely different $b run at a different time, just like if it were a $b inside of a function used to set c.  I think you built up an instinct to visually check that lexical scope for dynamic variables after writing a pile of code using this, which is why you are thinking it should check there.  But not only should the program not have to check there, the programmers should not have to either, when they are trying to determine where an erroneous value came from.  Programmers should be able to look purely in the call stack to find variables specified as dynamic scope.  This is why we have scopes, so that there's no global hunt all over for things causing errors.

nophead, 2021-09-12:
This really does break everything:
[image]

When making STLs I use the file that defines them and instantiate the module to make the STL. That no longer works because $fn, etc are not set but they are in the used file. It really doesn't make sense that when you use a module you have to set up exactly the same environment to get the same results. The default should be the same and you have to override it to change it.

MichaelAtOz, 2021-09-12:
As I said, I would reread all of this. And I did my deep dive today, I completed me review, including my ~10 year detailed involvement in this project.

 

I will report tomorrow, as it is 2200+hours here.

 

 

  _____  

 

 <https://user-images.githubusercontent.com/566149/132984308-09be9e1d-18de-4a6c-b08c-61ad4cf93bb4.png> image

 

This email has been checked for viruses by AVG.
https://www.avg.com

rcolyer, 2021-09-12:
> When making STLs I use the file that defines them and instantiate the module to make the STL. That no longer works because $fn, etc are not set but they are in the used file. It really doesn't make [...]

While you could choose the function approach described above and get exactly identical behavior, your use case there sounds like $fn should be part of the module arguments, to have a forced default inserted that can be explicitly changed.  This doesn't make sense as part of the file's lexical scope, because that's not in the call stack, but you can certainly put that in the module signature so it's explicit that there are defaults set that can be explicitly overridden.  If you want a centralized default value for a file or a subset of modules in a file, then module MyThing(..., $fn=default_fn) { ... } and grab it from lexical scope.  This sounds like the behavior you are looking for, and it's still easy and straightforward to do.  This approach is better because it makes clear what is happening and what can be changed.  There are no mysterious values entering the call chain from afar.

The behaviors you want seem to all be implementable in a clear and simple way in the scad files.  The only thing I actually entered this discussion for was to explain that all these behaviors are still scad implementable without OpenSCAD needing to auto-reparse the entire top-level used file on every function or module call.  If you need more tips about how to implement a specific case, I'm happy to provide it.

nophead, 2021-09-12:
Yes but this breaks the interface of the library. For 10 years it has been possible to define defaults in file scope and be able to override them explicitly if desired. This is how libraries have worked for all this time.

Another use case I have is setting $variables in my main file with the customiser to influence aspects of a multi-file project. Again that no longer works. I have to change all those variables to normal and then copy them all to $variables in the  main module. Not a big problem but all my projects have been broken by this change and all need to be fixed. I expect once this becomes a release there will be many projects affected. 

Never before has there been such a breaking change in OpenScad until you got involved. Where is @kintel these days?

nophead, 2021-09-12:
Also if $fn, etc need to be passed to every function, why are they dynamic at all? They have always been traditionally set at the top of the file and only explicitly set to make hexagons, etc.

t-paul, 2021-09-12:
The personal attacks will stop right there!

Also the issue has nothing to do with @rcolyer, he's provided great suggestions and explanations so far which I find extremely useful. It seems this issue was introduced long time ago and with the long release cycles only found much later. That's unfortunate. Putting up demands is not going to help. To keep OpenSCAD alive for the future, we need to find a way to solve the problem, ideally without need to keep bugs forever. My personal motivation to look into this issue is somewhere below zero because of how it's discussed. So far other positive things have convinced me it's still worth it to spend time on OpenSCAD. I don't know how long that will work.

If there's a separate issue with $variables regarding customizer, please check if there's a ticket for that and if not create one, there's no point in circling 2 issues at once going nowhere. This "oh, and there's also that issue" has successfully destroyed lots of discussions on the mailing list leading to no results.

So can we please go back to the specific issue of $variables declared at top level in used scripts? My summary at this point is:
- from documentation and other references, $variables are described to have *only* dynamic scope, so the current behavior is probably "correct"
- however it breaks existing code for the use case described here where those $variables are set to declare defaults => this is a bit fuzzy as it's not documented but the effect of a buggy implementation in OpenSCAD code handling $variables
- this "buggy" behavior also produces strange and surprising side effects for modules that don't directly use $variables
- OpenSCAD always tries to keep backward compatibility even though lots of people complain about this

This seems to lead to 4 options:
1) re-introduce bug, which may even mean the recalculation performance issue people complain very loudly about is not going to be fixable - we'll be stuck with that with no way forward
2) introduce some kind of pragma/flag thing as @MichaelAtOz mentioned, quite complicated as that will need to be on file level but probably doable, this would somehow allow both ways to coexist for some time
3) collect some help for updating affected libraries and setup documentation on how to do that
4) development stops, no releases anymore

nophead, 2021-09-12:
I don't see how this fixes the re-evaluation problem. File scope variables are still re-evaluated. If they cease to be that makes a huge problem because they can depend on $variables directly or indirectly through other expressions or through functions. I had always assumed this would be fixed by tagging them and only re-evaluating those where a used $value had changed. If they simply stop being re-evaluated all code will have to be re-written, so that all $variables are wrapped in functions, so are always referenced from the call stack.

In the current behavior the file scope variables of the main file are considered to be the bottom of the call stack but the file scope variables of a used file are sort of ignored, You can't look them up from a function or module but they can be mixed into other file scope expressions and referenced that way.

Previously the file scope variables of a used module would win over the call stack, which was a bug, but was easily worked around `is_undef()` and before that `if($x == undef)`.

Since all my code depends on `$extrustion_width` and `$layer_height` I would need to make those functions and any expression using them. So a lot of file scope expressions would become functions and every call would re-evaluate all of them, so doing in userland what OpenSCAD was doing in C++. But it wouldn't re-evaluate the truly constant expressions. So perhaps it would be a time win, But the only way to catch errors would be to make it illegal to reference $variables outside of functions and modules.

If the file scope of a used module was searched after the call stack but before the file scope of the main file then I think it would be largely backwards compatible but it fix the bug where it took precedence of the call stack unless you used `if_undef()` guards.

nophead, 2021-09-12:
The big problem with turning all the file scope expressions into functions is they are then public, when that is not appropriate. File scope variables are currently private when the file is used.

nophead, 2021-09-13:
The more I think about this the more broken it is. There isn't a separate problem with the customiser it is simply now a bug to set a $variable at file scope anywhere, including the main file, if they are going to be ignored.

My framework makes STLs and assembly diagrams by finding modules with the right name, using that file and instantiating a module from it. It can't include the file because that would instantiate it's objects . But when it uses it the $variables are ignored, so any bespoke ones like my customiser ones are undefined.

When you use a module and it uses others the call search context stack includes all the file scopes as well as local variables of functions and modules. Either it also has to include the $variables or ir has to be illegal to set them at file scope. Otherwise you get horrible undetected bugs. The sphere above is a simple example. That could be a printed part in a file that is part of a multi file project. It might set $fa, or $fs to get a smooth curve or it might set $layer_height and $extrusion_width and that will change the wall widths and vertical quantisation, etc.

If you open that file in the GUI  it will look fine and if you make an STL and export it, it will be correct, but when it is used by another file it will be wrong. So when my framework exports the STLs they will be wrong and opening any other file that has a subassembly using the part will get the wrong result. It makes no sense that using  file gives a different result to opening it in the GUI.

rcolyer, 2021-09-13:
> The more I think about this the more broken I realise it is. There isn't a separate problem with the customiser it is simply now a bug to set a $variable at file scope anywhere, including the main file, if they are going to be ignored.

This is a misunderstanding.  The special variables are processed in every scope in which they exist.  In the main file, they automatically become part of the execution call stack because the call stack starts at the top-level of the main file.  In execution of the main file, to satisfy the semantics, used file top levels should be only processed at the time of use<> to determine the value of the lexical variables which form the lexical context of the functions and modules in the used file.  Special variables can be a part of those calculations, and should continue to be set and made use of at this time, such as a variable at the top level which calls a function which makes use of a special variable at the top level.  This is all fine.  It just follows from the call stack.

> It makes no sense that using a file gives a different result to opening it in the GUI.

It's specified that this will be different, as use only brings in the functions and modules, and is specified to omit all geometries and to not bring variables from the used file into the context.  This is what differentiates it from include, which can be used if identical content is desired.

> Since all my code depends on $extrustion_width and $layer_height I would need to make those functions and any expression using them. So a lot of file scope expressions would become functions and [...]

Yes, exactly.  This is the way to go about it.  It's basically equivalent calculation for the places where this evaluation is needed, grabbing from the dynamic scope and doing so, but all of the lexical stuff gets the benefits of the declarative language and is fixed in value, which can boost performance significantly.  Moreover, most users of the language are probably assuming that the language is already evaluating these only once, and expecting the performance that accompanies this, because we describe the language this way all over the place.  So I'm certain there is a lot of code that is getting significant performance hits doing expensive things in used files, but where people simply don't know why their code runs so slow.  For modules this is usually not such a big deal, but for functions in used files, which can end up called in heavy loops, this surprising side-effect overhead builds up REAL fast.  List comprehension code executions of functions that don't even access the top level variables can go from a fraction of a second to hours just from switching it from include to use, which is a pretty big problem.

But if we stick to the variable specifications, and only dynamically evaluate the things that are dynamic (and convenience functions can help), then we can have both a clean traceable data flow of lexical constants and call stack dynamics, and predictable overall much faster performance.

nophead, 2021-09-13:
>It's specified that this will be different, as use only brings in the functions and modules, and is specified to omit all geometries and to not bring variables from the used file into the context. This is what differentiates it from include, which can be used if identical content is desired.

Yes of course, what I mean is instantiating modules in a used file should give the same results as opening that file in the GUI and instantiating them. It did but now it doesn't if there any file scope $variables because they get ignored. They either give an undefined error, or if they are a built in get their default value.

With the multi-tab editor there isn't really a main file anymore. You can have a used file open in another tab, which I did with the sphere example. Hit F5 in one tab and you get a completely different result to hitting F5 in the other tab, despite them both instantiating the same module. $fn was a a trivial example but it could equally be $extrusion_width which would give very subtle changes.

> In execution of the main file, to satisfy the semantics, used file top levels should be only processed at the time of use<> to determine the value of the lexical variables which form the lexical [...]

If a $variable is used in file scope calculations and those are stored in file scope normal variables which are used in the modules and functions of a used file then if you call one with a different value $variable in the call stack then they all need to be recalculated. That is why they are I presume. Not a bug but a deliberate design decision to make it work.

In the posted picture of my example project you can see all the images are a little different. This is because $fa and $fs are ignored when file is used and the modules are instantiated by my framework.

I think either all file expressions that use a $variable need to be recalculated every time a module or function is called or you make it illegal to use file scope $variables. But if you do that I can't have dimensions depending on $extrusion_width in a file scope constant. I would have to put them in function and that would pollute the namespace because they would all be global.

rcolyer, 2021-09-14:
> Yes of course, what I mean is instantiating modules in a used file should give the same results as opening that file in the GUI and instantiating them.

Definitely not, as this is not guaranteed in any version of OpenSCAD I've ever seen.  These have fundamentally different behavior, so whether or not you get the same result definitively depends on how you have implemented the main file and the used file.  I could make a set of failing examples (e.g., module Foo(){sphere(5);} in used, $fn=100; Foo(); in main), but if you play around a bit and try to make it fail, you'll see this was never guaranteed, and was easy to have mismatched under normal use.  And yet, it is also possible to write code that follows the scope principles and provides the same result in both modes of operating in all versions of OpenSCAD going back many years through the master branch.

> I think either all file expressions that use a $variable need to be recalculated every time a module or function is called or you make it illegal to use file scope $variables

That recalculating is expensive and out of the call chain.  But there is no need to make it illegal, as they still follow a clear process where they are, and could be part of very involved calculations of lexical values.  (Or used for other applications of a file that happens to be used, such as the use/demo pattern.)

> I would have to put them in function and that would pollute the namespace because they would all be global.

Fortunately, this does not need to happen.  Here's a simple solution which takes advantage of function literals not being captured by a use, as they reside in the variable namespace:

main.scad:
```
use <used.scad>
  
$extrusion_width = 0.2;  // Optional.  To override default.
echo(extrusion_width());  // WARNING: Ignoring unknown function 'extrusion_width'

Foo();
```

used.scad:
```
extrusion_width = function() is_undef($extrusion_width) ? 0.4 : $extrusion_width;

module Foo() {
  echo(extrusion_width());
}
```

nophead, 2021-09-14:
>That recalculating is expensive and out of the call chain. 

I don't understand. It is currently very expensive because all file scope variables are re-evaluated on every call when 99% don't change. This still happens if if didn't it would break even more semantics. I.e. the call stack can have a new value for a $variable that is used in the file scope calculations, so hey need to be recalculated and all the ones that depend on them, either directly or via functions. This change doesn't alter that.

I don't see any way to make my framework work with this new semantics. I have to be able to use a file and instantiate a random module from it with the command line and get the same result that I would opening the file in the GUI and instantiating the same module. For example I have modules `<name>_stl()` that my framework instantiates to generate all the STLs it finds on the BOM. That doesn't work now because the $variables that may be set in those files are not defined, or default to the wrong value.

rcolyer, 2021-09-15:
> I don't see any way to make my framework work with this new semantics.

I came here primarily to try to demonstrate how, so I'm happy to try to help with this.  I looked in NopSCADlib, and it looks like you were already in the process of resolving many of these issues, but I couldn't find the &lt;name&gt;_stl() modules you mentioned.  If you can point me to an example of this to inspect the situation, I can try to give an example fix.

nophead, 2021-09-15:
An example of a module that creates an STL is https://github.com/nophead/NopSCADlib/blob/master/printed/handle.scad#L48. It announces its presence on the BOM by calling `stl("handle")` so `handle.stl` gets added, My Python framework then searches for a scad file that has a module called `handle_stl()`. It can be in the client project or the library. To make the STL it generates a temporary  stub file that uses the file it found and instantiates `handle_stl()`.

What I have done to botch around projects not building is include my [`global_defs.scad`](https://github.com/nophead/NopSCADlib/blob/master/global_defs.scad) file in the stub file I generate when making STLs. That avoids $variables being undefined in used files and gives $fa, etc their default values for a library project because every file of the project also includes` global_defs.scad` . But if a file overrides `$fa` at local scope I think it will get the value from `global_defs.scad` instead.

Similarly `$extrusion_width` and `$layer_height` are given a defaults that match how I print things. Other library uses most likely override those, so I think there generated STLS will be incorrect.

The other change I made to `global_defs.scad`  is removing two $variables and pasting the expression they were assigned into the functions that used them: https://github.com/nophead/NopSCADlib/blob/master/utils/core/bom.scad#L46. All the other $variables are captured in file scope normal variables, so they are still defined but will get defaults instead of any file scope overrides.

They do respond to overrides on the stack though. So I don't see how this change has gone anyway at all to preventing all the file scope variables being revaluated, It just breaks my library by making used files different to opening them in the GUI.

I think I will have to use the last release for command line processing and the newer version for the GUI to get fast rendering. eventually I will have to freeze my versions and stop using the latest OpenSCAD as I can't do the things I do in this library with the latest version because the shifting semantics make it less powerful.

rcolyer, 2021-09-16:
I checked every usage of $fa in the library and couldn't find any that looked like it would differ between the two versions, so I don't see anything to fix there.  The linked handle.scad file also does not appear to contain any special variables so that file cannot be directly affected by this issue.  I checked some other files not mentioned, and I do see others using extrusion_width and layer_height, but searching through a pile of code I wasn't able from your description to find what would be a failing example.  Can you point to or provide a specific file or usage instructions which result in a specific example of failure to achieve the desired objective with this codebase?

nophead, 2021-09-16:
If I set `$fa` to 1 in my main file then in the GUI all the corners get smoother than the default, which is set to 6 in my global defs. But with master my STLs are not changed because I have to now include my `global_defs.scad` in the stub that makes them. If I don't then `$fa` has the OpenSCAD default 12 and that overrides the value in main. So it is a manifestation of the smooth sphere example above. Used files don't respect their local values of `$fa`, etc.

https://github.com/openscad/openscad/issues/3881#issuecomment-920095101 shows that `$layer_height` and `$extrusion_width`  will go back to their defaults if a printed part comes from a used file. That used file will include my `global_defs.scad` which should see the dynamic value of  `$layer_height` and set layer_height to it. I think instead it will now see `$layer_height` undefined in a used module or function call and so set `layer_height` to the default.

martinbudden, 2021-09-16:
>My personal motivation to look into this issue is somewhere below zero because of how it's discussed. So far other positive things have convinced me it's still worth it to spend time on OpenSCAD. I don't know how long that will work.

@t-paul , unfortunately it's human nature that things that work well don't tend to get noticed and are taken for granted - people only tend to speak out when things change or break.

So, to do a small bit to redress that balance, I'd like to say I've been a heavy user of OpenSCAD for a number of years and very much appreciate the work that you and others have put into that project.

nophead, 2021-09-16:
Well of course I appreciate all the work that goes into OpenSCAD. I have done some myself in the past. It is the nature of bug reports that they are only posted when things aren't working, not necessarily human nature to only complain.

martinbudden, 2021-09-16:
@nophead , there was no implied criticism of you. This is a serious issue that needs to be resolved.

martinbudden, 2021-09-16:
@nophead , this is a complicated issue and I haven't got my head fully around it yet, and won't have the time to do so for a while. Can you clarify something, which is the situatio, 1 or 2?

1. This change breaks NopSCADlib, and there is no work around (aside from using an earlier version of OpenSCAD)
2. This change breaks NopSCADlib, but can be worked around in a way that severely impacts performance (ie by using functions instead of $variables and perhaps other work arounds)

[11 more replies]

closed 2023-11-27 by kintel