openscad/openscad #3785: Wiki: Tutorial code examples contain invalid content

piranha771, 2021-05-24:
Example: https://en.wikibooks.org/wiki/OpenSCAD_Tutorial/Chapter_1

See first code example.
Expected:
`cube(10);`
 
Actual:
`cube(10);

{{BookCat}}`

Conclusion:
This  seems to have something todo how code is quoted in the wiki. Once you know it's not part of the code it is obvious but this might scare any newcomers away.

jordanbrown0, 2021-05-24:
On 5/24/2021 1:13 AM, David Asmuth wrote:
> Example: https://en.wikibooks.org/wiki/OpenSCAD_Tutorial/Chapter_1
> <https://en.wikibooks.org/wiki/OpenSCAD_Tutorial/Chapter_1>
>
> See first code example.
> Expected:
> |cube(10);|
>
> Actual:
> ``cube(10); [...]

Huh.  I edited that a while back (and need to go back to it, because
consensus is that the way I structured it is too complicated), and it
was working then.

... And I see that somebody made a change later that introduced this
problem.

Right now I'm not remembering what that {{BookCat}} was supposed to do,
but it was supposed to do something useful.

I'll see about fixing it.

jordanbrown0, 2021-05-24:
Does anybody know who the WikiBooks user "Anachronist" is?

t-paul, 2021-05-24:
No, at least I don't know that user.

t-paul, 2021-06-30:
Look like it's fixed now.

closed 2021-06-30 by t-paul

jordanbrown0, 2021-07-01:
It renders OK again, but there are still outstanding issues that have had me keeping it as an open item in my mailbox.  The new template works fine for trivial examples, but I don't know how well it will work for examples with, for instance, braces in them.  More experimentation needed - and once we decide it's OK, the rest of the examples should be changed over.

t-paul, 2021-07-01:
Yes, that would be nice. I'm not sure if there's a good place for tracking those kind of wishlist / improvement items. I suppose we could add a category for that in the tracker here.