openscad/openscad #3786: Retina/hi-dpi icons

gingerbeardman, 2021-05-24:
I'm happy to draw a new set of icons if this would be helpful.

They would be as vector and I would contribute the artwork and any exported versions with no licence attached.

Current icons look:
- pixelated on retina/hi-dpi displays
- different styles as they were drawn at different times

gingerbeardman, 2021-05-24:
Some examples from a spare few minutes

[image]

t-paul, 2021-05-24:
We have some (most?) of the icons in some branch as OpenSCAD code. The plan was to move to SVG icons at some point which might be even better than the need for different pixel sizes.

I'll have a look where those icons are stored...

gingerbeardman, 2021-05-24:
Moving to vector makes total sense.

I'll stop now and wait to hear from you.

t-paul, 2021-05-29:
I've merged the script and icons from the old `svg-icons` branch. They are not actually used directly yet but were used to generate the current pixmaps.

The SCAD script is https://github.com/openscad/openscad/blob/master/resources/icons/svg-default/icons.scad

List of icons available so far:
[image]

t-paul, 2021-05-30:
Added some more icons. I'll try to post a PR actually using the icons.
[image]

Those will need some more polishing to create a bit of a border (Linux/4k display):
[image]

gingerbeardman, 2021-05-31:
Cool.

My issue with these icons, and with the pixmaps, is that the icons seem to fight against each other due to their relative sizes. 

I think all icons should be as wide/tall as the "angled cube" with the exception of any elements that like the arrows and hourglass. This would give much more pleasing and less claustrophobic toolbars.

t-paul, 2021-05-31:
Yes, that's what I mean with the todo item in #3795. Right now they just scale to fit based on bounding box. I think the easiest solution is to normalize to e.g. 100x100 in the SCAD and then patch the SVG to use a viewbox of maybe 140x140 to also keep some additional margin.

gingerbeardman, 2021-05-31:
Ah, right, great!

I should have read that rather than doing this mockup.

[image]

t-paul, 2021-05-31:
Yes, that's much more like I would want it to look like.

closed 2021-06-10 by t-paul

gingerbeardman, 2021-06-21:
Just a quick note that whilst SVG icons change colour in dark mode the chrome around them does not and this makes things less readable: white on silver.

PR #3795 (2021-06-10): Enable SVG icons
Enable use of SVG icons (fixes #3786).

- Set SVG icons directly on actions
- Remove icons from GUI file as they are set in code based on dark-mode detection
- Remove image files that are replaced with SVG generated from `icons.scad`

PR has MacOS CI-build enabled.

ToDo:
- [x] Convert all existing toolbar/menubar icons to SVG
- [x] Fix display, right now icons are just maxing out based on their bounding box, idea is to declare a specific size and patch that into the SVG header to prevent uneven scaling and allow for some empty margin.
- [x] Add QtSVG to AppImage Docker image  [... cut]