openscad/openscad #3791: Errors and inconsistent results when hulling two circles in OpenSCAD 2021.01

nyponen, 2021-05-28:
Sometimes the sample code produces the correct results, sometimes lines are missing with no error messages, sometimes lines are missing with error messages like these:

WARNING: GeometryEvaluator::applyHull2D() during CGAL::convex_hull_2(): CGAL ERROR: postcondition violation! Expr: is_ccw_strongly_convex_2( res.output_so_far_begin(), res.output_so_far_end(), ch_traits) File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Convex_hull_2/ch_akl_toussaint_impl.h Line: 340 

The results are inconsistent, but this is what usually happens. If I run the sample code as is, it produces the correct results with no error messages. If I change radius to 25, one line is missing, and the above error message is issued. If I use radius=20, and change line_width to 0.5, two lines are missing and two error messages are issued. If I use radius=20, and line_width=1.5, two different lines are missing. Different combinations of parameters result in different lines missing, sometimes with and sometimes without error messages.

`sides = 5;
line_width = 1;
radius = 20;

for (i=[0:sides-1]) {
    for (j=[i:sides-1]) {
        if (i!= j) {
            hull() {
                rotate([0, 0, i*(360/sides)])
                    translate([radius, 0, 0])
                        circle(d=line_width, $fn=sides);
                rotate([0, 0, j*(360/sides)])
                    translate([radius, 0, 0])
                        circle(d=line_width, $fn=sides);
            }
        }
    }`

OpenSCAD Version: 2021.01
System information: Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833 8 CPUs 39.87 GB RAM
User Agent: OpenSCAD/2021.01 (Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833)
Compiler: GCC "7.4.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_60
Eigen version: 3.3.7
CGAL version, kernels: 4.11, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.13.0
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.50.2
lodepng version: 20180910
libzip version: 1.5.2
fontconfig version: 2.13.1
freetype version: 2.10.1
harfbuzz version: 2.4.0
cairo version: 1.16.0
lib3mf version: 1.8.0
Application Path: C:/Program Files/OpenSCAD2021.01
Documents Path: C:\Users\nypon\Documents
User Documents Path: C:\Users\nypon\Documents
Resource Path: C:/Program Files/OpenSCAD2021.01
User Library Path: C:/Users/nypon/Documents/OpenSCAD/libraries
User Config Path: C:\Users\nypon\AppData\Local/OpenSCAD
Backup Path: C:/Users/nypon/Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
C:/Users/nypon/Documents/OpenSCAD/libraries
C:/Program Files/OpenSCAD2021.01\libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
C:/WINDOWS/fonts
C:/Users/nypon/.local/share/fonts
C:/Users/nypon/.fonts
GLEW version: 1.12.0
OpenGL Version: 4.6.0 - Build 27.20.100.8854
GL Renderer: Intel(R) HD Graphics 630
GL Vendor: Intel
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
[... 217 lines cut ...]
GL_KHR_blend_equation_advanced_coherent
GL_KHR_context_flush_control
GL_KHR_debug
GL_KHR_no_error
GL_KHR_shader_subgroup
GL_KHR_shader_subgroup_arithmetic
GL_KHR_shader_subgroup_ballot
GL_KHR_shader_subgroup_basic
GL_KHR_shader_subgroup_clustered
GL_KHR_shader_subgroup_quad
GL_KHR_shader_subgroup_shuffle
GL_KHR_shader_subgroup_shuffle_relative
GL_KHR_shader_subgroup_vote
GL_KHR_texture_compression_astc_hdr
GL_KHR_texture_compression_astc_ldr
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control

nophead, 2021-05-28:
Yes it seems to be some numerical instability when you rotate pentagons by the their angle and then hull them. A side of the pentagon has vertices that exactly inline with the edge of the hull but with angles being irrational when expressed in radians I expect they don't exactly line up but are incredibly close.

[image]

I guess that leads to some indecision of whether the point is inside the hull or not. Presumably including it makes the result very very slight concave, when it should be convex by definition.

A workaround seems to be to rotate one end slightly more. E.g. it works with:
'''
rotate([0, 0, i*(360/sides) + 0.001])
'''

That makes the point clearly inside or outside but also probably makes the result have a few extra vertices in the outer perimeter.

nyponen, 2021-05-28:
FWIW, the problem only occurs in 2D. If you change the circles to cylinders, the hull works correctly.

My circumvention was to add scale([1, 0.99999]) before the circles. It's a kludge, but it avoids the error while not adding any extra vertices to the perimeter.

kintel, 2024-03-31:
Status as of 2024-03-24: Still broken with both CGAL and Manifold

pca006132, 2024-09-05:
Removing the manifold tag because this is only something related to 2D hull.

pca006132, 2024-09-05:
And error message from CGAL:

```
CGAL error: postcondition violation!
Expression : is_ccw_strongly_convex_2( res.output_so_far_begin(), res.output_so_far_end(), ch_traits)
File       : /nix/store/g9spz3sbkig9l4kyyr3ks6w9nmfcsjzd-cgal-5.6.1/include/CGAL/Convex_hull_2/ch_akl_toussaint_impl.h
Line       : 330
```