openscad/openscad #3797: Geometry missing on render.

gabrielrobinson000, 2021-06-01:
Geometry missing in render but is visible in preview. 
When the for loop is set to `[0:0]` it's visible again.
```
linear_extrude(1)
    for (Hole = [0:1])
        rotate([0, 0, 360/2 * Hole])
            polygon([[0,0],[1,0],[1,1]]);

cylinder(2, r= .25);
```
It was test on the current master version 1a9430fba435c3b3758988dbf80f94cb2f0423c6 .

MichaelAtOz, 2021-06-01:
The two polygons present a shared edge, that is non-manifold.
Fix is to move it a fraction from the origin, e.g. `polygon([[0,0],[1,0],[1,1]]);`
You should not have edges or points that exactly touch, either make them
two distinct objects or embed on clearly in the other.

MichaelAtOz, 2021-06-01:
Oops make that `polygon([[1/256,0],[1,0],[1,1]]);` (small number should be a power of two to prevent binary float inaccuracy)

closed 2021-06-01 by MichaelAtOz

unDeadHerbs, 2021-06-02:
Interestingly, if you wrap that in a module it fails to generate a warning.

```
module bad()
    linear_extrude(1)
        for (Hole = [0:1])
            rotate([0, 0, 360/2 * Hole])
                polygon([[0,0],[1,0],[1,1]]);

bad();
cylinder(2,r=.25);
```