openscad/openscad #3798: Error Suppressed by Module Declaration

unDeadHerbs, 2021-06-03:
In #3797 the example code produces an error correctly describing what's happening.
However, if the code is wrapped in a module the error is missing, possibly making this very hard to track down or invisible to the user.

```
module bad()
    linear_extrude(1)
        for (Hole = [0:1])
            rotate([0, 0, 360/2 * Hole])
                polygon([[0,0],[1,0],[1,1]]);

bad();
cylinder(2, r= .25);
```

thehans, 2021-06-03:
Your example code produces an error message for me.

You are likely encountering #454, which is that warnings don't get displayed on *subsequent* renders.  The result geometry gets cached on first run, so 2nd run skips over re-processing the code that encountered the warnings the first time.

One workaround is to click  "Design -> Flush Caches"

closed 2021-06-03 by thehans

unDeadHerbs, 2021-06-03:
Yep, my apologies.