openscad/openscad #3799: assertion at model generation time

lcominelli, 2021-06-06:
Hello,

I am Lionel from offshore french island, 

I am using openscad to design foils, boomrangs and various airfoils.
Yesterday, I fall on an unworkaroundable issue at my level, may be, I did something wrong. 
Then I ask for any help to get out my deadlock ! 
I try a lot of scenarios without any positive result.

Bellow the code issuing the assertion and brake F6 generation.
stl file is in data.zip or can be dawnloaded at http://eruptiontrek.com/covid19/entrepot/rhomboiide_REARbody.stl

Please, did someone knows whats happen and workaround ?

Kind Regards
Lionel

The openscad console message is 
Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e_below != SHalfedge_handle() File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_3/SNC_FM_decorator.h Line: 426 
Geometries in cache: 4
Geometry cache size in bytes: 1653152
CGAL Polyhedrons in cache: 2
CGAL cache size in bytes: 22688
Total rendering time: 0:00:07.564
   Top level object is a 3D object:
   Simple:        yes
   Vertices:        8
   Halfedges:      24
   Edges:          12
   Halffacets:     12
   Facets:          6
   Volumes:         2
Rendering finished. 

[data.zip](https://github.com/openscad/openscad/files/6603732/data.zip)

// Main openscad file //
//
module full_body () {
  //difference () {
    import ("rhomboiide_REARbody.stl");
  //  translate([500,0,0]) 
  //  cube([1000,500,600],true);
 // }
};

full_body();
translate([1000,0,0])  cube([1000,500,600],true);

lcominelli, 2021-06-06:
Sorry, forgot to say that the preview F5 is working well.

Kind Regards

MichaelAtOz, 2021-06-06:
Firstly, general help questions should be initially raised at the [Forum](http://forum.openscad.org/Registration-and-Subscription-Info-READ-THIS-FIRST-tp21660.html).

But the issue is here, so...

Generally such CGAL Assertions are from bad stl files, or badly formed polyhedron.
You don't get an issue in Preview as the geometry is not evaluated by CGAL.

However the stl appears to be OK.
The next thing that can happen, is what I call the small number problem.
OpenSCAD has a fine grid that is applied to coordinates, sometimes where there are
many points in a small space, one can get popped into another grid, making the geometry
non-manifold.
Given the many long thin triangles I suspect this has occurred here.

There is litlle you can do, you could try resampling the stl.
Or if you generated the stl with OpenSCAD there are tips to avoid this.

lcominelli, 2021-06-07:
Dear Michael,

oups, sorry I was thinking the assertion was issuing a bug, this is the reason why I post here.
Nevertheless, hundred thanks, you put the finger on ! 

Some points in my foil cross section 2D shape were too near.
You where right, even no error or warning messages, some polyhedrons should be corrupted at morphing time.

Next time I will directly ask to the correct forum.  
Thanks a lot.
Kind Regards

closed 2021-06-07 by MichaelAtOz