openscad/openscad #3800: Add option to reset window layout

t-paul, 2021-06-06:
This should cover 2 issues:

- [ ] Fix the initial screen layout to a sane state, right now it's a mess if no config file exists
- [ ] Add menu entry to reset to the initial layout, e.g. `Window` -> `Reset Windows`.

MichaelAtOz, 2021-06-06:
Is it worth a cmd-line switch?

t-paul, 2021-06-06:
Hmm, yes. That could cover the cases where the window position is messed up that bad that the main window does not even show up.

jordanbrown0, 2021-06-07:
One might think that if it is so bad that the main window doesn't show up, there could be a sanity test that would fix it automatically.

DanielRunningen, 2023-07-10:
Is there any manual way to do this at the moment? My customizer is stuck under the preview and since the preview can't be re-sized, it's almost impossible to use.

t-paul, 2023-07-12:
On a Linux system the config file would be `~/.config/OpenSCAD/OpenSCAD.conf` and that file has a `[window]` section with some binary blobs describing the state:

[image]

Removing that section should revert everything back to defaults.