openscad/openscad #3801: Customizer metadata output (CLI feature request)

Irev-Dev, 2021-06-06:
The CLI already has a flag for getting the customizer output `-p`. Though there is a usecase for more rich data about the customiser values, (whether it's a string or range or spinbox etc) so that other platforms can use this information to display their own customizer UI.

I'm specifically interested in it to add a customizer UI for [CadHub](https://github.com/Irev-Dev/cadhub). Originally we had planned on making a [javascript parser](https://github.com/Irev-Dev/cadhub/issues/320) that would attempt to extract these values from the `.scad` code (possible reusing some of @kintel's [previous work](https://bitbucket.org/shapefactory/scad-web/src/master/)), but @t-paul stepped in a suggested it be built into OpenSCAD. Which would be great since OpenSCAD already understands the customizer comments syntax. Would be a much more robust approach.

Irev-Dev, 2021-08-07:
I think it should all be able to fit into what's been outlined [here](https://github.com/Irev-Dev/cadhub/discussions/437), so I put together a little example . Although it's neglecting groups and that the reports for the time being.

```typescript
// OpenSCAD
const openscadValues = `
// slider widget for number with max. value
sliderWithMax =34; // [50]

// slider widget for number in range
sliderWithRange =34; // [10:100]

//step slider for number
stepSlider=2; //[0:5:100]

// slider widget for number in range
sliderCentered =0; // [-10:0.1:10]

// spinbox with step size 1
Spinbox= 5;

// Text box for string
String="hello";

// Text box for string with length 8
String="length"; //8

//description
Variable = true;
`

const openscadConverted = [
  {type: 'int', name: 'sliderWithMax', caption: 'slider widget for number with max. value', initial: 34, step:1, max: 50},
  {type: 'int', name: 'sliderWithRange', caption: 'slider widget for number in range', initial: 34, step:1, min: 10, max: 100},
  {type: 'int', name: 'stepSlider', caption: 'step slider for number', initial: 2, step:5, min: 0, max: 100},
  {type: 'decimal', name: 'sliderCentered', caption: 'slider widget for number in range', initial: 0, step:0.1, min: -10, max: 10},
  {type: 'int', name: 'Spinbox', caption: 'spinbox with step size 1', initial: 5, step:1},

  {type: 'string', name: 'String', caption: 'Text box for string', initial: 'hello'},
  {type: 'string', name: 'String', caption: 'Text box for string with length 8', initial: 'length', maxLength: 8},

  {type: 'boolean', name: 'Variable', caption: 'description', initial: true,},
]
```

closed 2021-08-24 by t-paul

MichaelAtOz, 2021-08-24:
@t-paul you should claim the bounty.
I note the comment in #3817 
>Unfortunately Bountysource seems to be defunct, a lot of people are complaining and the site is unstable.

It may be a good test to see if it working, I've been thinking maybe the parallel bounty should be claimed and funds extracted. But I haven't had time to checkout Bountysource.

t-paul, 2021-08-24:
Oh, right, let me try.

MichaelAtOz, 2021-08-24:
Twitter says

>Bountysource@Bountysource
Jul 6
Nicolas le Herissier is nominated General Manager of Bountysource. New team. New ambitions. 
The Blockchain Group.

Plenty of IBM Bounties being placed.
Web site seems to work.
Maybe nothing to worry about.

coderforlife, 2022-10-26:
This feature was added and works, but doesn't mention it in the `--help` or the docs on the customizer. I think it needs to be added to the list in the string at https://github.com/openscad/openscad/blob/1da8cca1048fd01544feaf8aee7814fb65265502/src/openscad.cc#L943

commit 1e382f42 (2021-08-24): Add customizer parameter export (closes #3801).