openscad/openscad #3804: Resolution support for SVG import

sidneycadot, 2021-06-08:
Hi,

When using the import function with an SVG file containing a path with many ARC-type segments, I noticed that the discretization into an OpenSCAD polygon is quite crude.

I hoped that it would be possible to improve the conversion by setting eg. the $fn variable, much like it does with text(), but unfortunately this doesn't seem to affect the conversion.

It would be great if the SVG importer honors the resolution flags like $fn, $fa, etc.

ChrisCoxArt, 2021-12-23:
This has been implemented in the current development builds.

sidneycadot, 2021-12-23:
Interesting, will have a look. Christmas holiday coming up ....

t-paul, 2021-12-23:
Thanks for the heads-up, and huge thanks for adding this feature @ChrisCoxArt. Closed via #3994.

Download and info about dev snapshots at https://openscad.org/downloads.html#snapshots

closed 2021-12-23 by t-paul