openscad/openscad #3805: ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron.

quirxi, 2021-06-09:
-> Openscad Version: OpenSCAD-2021.01-x86_64.AppImage

-> Operating System: 
    * Linux 5.6.0-1056-oem #60-Ubuntu SMP Tue May 4 06:40:10 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
    * Ubuntu 20.04.2.0 LTS _Focal Fossa_ - Release amd64 (20210209.1)

-> What is the intended goal:
 * I wanted to render a kind of puzzle shaped bowl, that is construced by cutting out an inner puzzle piece of an outer puzzle piece in the same shape (see awaited.png and cutout.png in the download link).

-> Steps to reproduce error:

1. Download and uncompress the example zip file [here](https://drive.google.com/file/d/1qXDnFUjQXm__zBLSt4k35HPhlAgevuPx/view?usp=sharing).
2. Open the provided file "JFAS-Logo_Bug.scad" in openscad.
3. Press the Preview button. You will get a preview image that looks allright: see "awaited.png"
4. If you press the "Render" button (F6) only a closed shape of the outer puzzle piece will be rendered without the cutout. (see "rendered.png"). 
5. I got the following error message: 
`ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron.`

-> What was the awaited result?
    * It should be generated a puzzle piece shaped bowl, but instead a closed shape puzzle piece was generated as if the difference function would have no effect. I have rendered the single pieces into an stl file which worked flawlessly but when I tried to difference the two stl-files I have run into the same problem again. The mesh itself is generated by openscad itself from two imported svg files, so they should be allright. Opening the single pieces in meshlab did not show any error in the mesh aswell.

-> This is the minimal code I have used for the example:

`totalHei=10;
baseHei=2;

difference()
{
linear_extrude(height=totalHei, center=false, convexity=10)
import("PuzzleTeilAussen.svg");

translate([0,0,baseHei+0.1])
linear_extrude(height=(totalHei-baseHei+0.1), center=false, convexity=10)
import("PuzzleTeilInnen.svg");
}
`

MichaelAtOz, 2021-06-09:
PuzzleTeilInnen.svg is bad.
[image]

MichaelAtOz, 2021-06-10:
You can work-around the bad svg via offset(),

```
offset(delta=0.001)
        import("PuzzleTeilInnen.svg");
```

closed 2021-06-10 by MichaelAtOz

quirxi, 2021-06-10:
Woow! Thank you very much for your fast reaction! Now generating the puzzle piece works but when I try to insert letters I get an assertion. I will open a new bugreport for that.

Thanks again for your fast help.