openscad/openscad #3808: ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e_below != SHalfedge_handle()

quirxi, 2021-06-10:
-> Openscad Version: OpenSCAD-2021.01-x86_64.AppImage

-> Operating System:
* Linux 5.6.0-1056-oem #60-Ubuntu SMP Tue May 4 06:40:10 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
* Ubuntu 20.04.2.0 LTS Focal Fossa - Release amd64 (20210209.1)

-> What is the intended goal:
 * I tried to render a kind of stamp where the outside is puzzle shaped and the inside are some letters (see intended.png).

-> Steps to reproduce error:

1. Download and uncompress the example zip file [here](https://drive.google.com/file/d/1NDZAy7ZE33RQfpCGZn1KhPaHx4z0sbed/view?usp=sharing).
2. Open the provided file "JFAS-Logo_Bug.scad" in openscad.
3. Press the Preview button. You will get a preview image that looks allright: see "intended.png"
4. If you press the "Render" button (F6) only a closed shape of the outer puzzle piece will be rendered without the letters in it. (see "rendered.png").
5. I got the following error message:

> ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e_below != SHalfedge_handle() File: /usr/include/CGAL/Nef_3/SNC_FM_decorator.h Line: 426 

-> This is the minimal code I have used for the example:

> totalHei=10;
> baseHei=8;
> 
> ////////////////////////////////////////////////////////////////////////
> scale([0.3,0.3,1])
>   makeStamp();
>  
> ////////////////////////////////////////////////////////////////////////
> module makeStamp()
> {
>   union()
>   {
>     translate([5,-2,0.01])
>       scale([0.85,0.85,1])
>         import("3d-text.stl");
> 
>     difference()
>     {
>       linear_extrude(height=totalHei, center=false, convexity=10)
>         import("PuzzleTeilAussen.svg");
> 
>       translate([0,0,baseHei+0.1])
>         linear_extrude(height=(totalHei-baseHei+0.1), 
>                        center=false, convexity=10)
>           offset(delta=0.001)
>             import("PuzzleTeilInnen.svg");
>     }
>   }
> }
>

CaptainFalcon92, 2023-08-14:
I'm in a similar situation where any difference operation performed on an imported .stl results in CGAL error. As soon as the imported stl is not the most basic cube, it fails.

I think scad or underlying library is completely incapable of making operation on imported meshes.
Althout if fixed the stl in many software for potential errors.

kintel, 2024-04-01:
Status as of 2024-03-24: The majority of issues like these are now working when using the Manifold backend (Preferences -> Features -> Manifold). Since the STL isn't available any longer, I'm closing this, as we have dozens of very similar issues open.

closed 2024-04-01 by kintel