openscad/openscad #3809: SVG rendering incorrect for a path with multiple simple subpaths.

DavidPhillipOster, 2021-06-10:
The python qrcode package pyqrcode generates SVG using a series of short horizontal and vertical line segments.

Example of creating the .svg attached.

```
import pyqrcode
from pyqrcode import QRCode
getqrcode = pyqrcode.create("A")
getqrcode.svg("qrA.svg", scale=1)
```

OpenSCAD version 2021.01, given the following .scad file:
```
import("qrA.svg");
```
does not parse this correctly compared to Inkscape and Google Chrome. `qrA.svg` could be worked up into a unit test.

_Originally posted by @DavidPhillipOster in https://github.com/openscad/openscad/issues/1740#issuecomment-858815675_

but reduced to a smaller file, with (once I can attach them) expected and actual results.

MichaelAtOz, 2021-06-10:
Note that QR codes seem to frequently contain points which become non-manifold.
e.g. http://forum.openscad.org/Preview-OK-Render-gets-only-half-tp33306p33315.html 
That won't be the cause of the above issue, but could be your next problem.
As shown, offset() is a work-around.

DavidPhillipOster, 2021-06-11:
I don't think this is quite the same problem, since although the file renders incorrectly, it previews and renders without error (log attached: [log.txt](https://github.com/openscad/openscad/files/6635199/log.txt) )

One workaround for QRCodes generated in Python anyway: In the python, you can have it generate a `.png` file just as easily as it generates `.svg`, then bring the `.png` into OpenSCAD.

DavidPhillipOster, 2021-06-11:
Thank you. Yes: 

```
linear_extrude(1)offset(0.001)import("qrA.svg");
```
reports as valid. Without the `offset()`  I get:

```
ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron. 
```

But the rendering is still badly wrong: the  [image] shapes look nothing like they are supposed to.

commit 099d200d (2024-03-10): SVG import: Fix default stroke-linecap

Was "square", now is "butt" as stated as default value in SVG spec:
https://www.w3.org/TR/SVG11/painting.html#StrokeLinecapProperty
Fixes #3809
closed 2024-03-10 by t-paul