openscad/openscad #3815: resize() drops the color of "imported" module, scale() keeps the nested colors working

RustyJoeM, 2021-06-15:
Scenario:

file `colored.scad`:

```
module green_10_box() {
    color("green")
    cube(10, center = true);
}
```

file `core.scad`:

```
use <colored.scad>:

// shown in green, as expected
green_10_box();

// coloring dropped, displayed in default "yellowish"
translate([20, 0, 0])
resize([20, 20, 20])
green_10_box();

// shown in green, as expected
translate([50, 0, 0])
scale(2)
green_10_box();
```

I'm not sure if this is correct/documented behavior.
My expectation was that scaling/resizing will not affect the coloring of module...

jordanbrown0, 2021-06-15:
Always try to reduce reports to the simplest possible example.

This one can be reproduced by
```
resize(1) color("red") cube();
```

resize() runs its children through CGAL rendering, which loses color information.

RustyJoeM, 2021-06-15:
Got it & apologies, i will reduce potential future reports, i didn't think it through...

Thanks as well for CGAL explanation/reminder, it is indeed mentioned in `render()` documentation.
I didn't connect this implicitly with color drop, which is obvious for preview vs render display...

closed 2021-06-15 by RustyJoeM