openscad/openscad #3817: Let customizer accept arbitrary nested vectors

50an6xy06r6n, 2021-06-17:
Customizer vectors are limited in that they (seemingly) can only contain strings and doubles. This seems like an artificial limitation, since OpenSCAD vectors can contain arbitrary combinations of types, including other vectors, and it seems relatively easy to encode the result in the customizer JSON. 

I've traced the issue to `VectorParameter` and `VectorParameter::importValue`, which seem to treat all vector parameters as only containing doubles (I haven't yet figured out why strings are also handled correctly). Augmenting this to accept any valid SCAD input seems pretty doable. I'm willing to take a stab at this, as I have a use case for it, and it sounds like at least a few other people do as well, based on [this forum thread](http://forum.openscad.org/search-a-solution-converting-special-string-to-array-for-customizer-td33139.html). Would just like to get confirmation that this is a change that makes sense, since this is my first time diving into the OpenSCAD codebase.

Metadorius, 2021-06-20:
Need this issue as well. Was designing customizable V-Slot covers with dynamic amount of holes for bolts and stumbled upon this issue when trying to describe bolt holes with arrays of position and diameter.

t-paul, 2021-06-20:
This needs some clear distinction between:
1) The customizer meta data collected from comments, this is supposed to follow the [Thingiverse docs](https://customizer.makerbot.com/docs)
2) Implementation of the GUI part - which allows users to modify values and store multiple value sets

Changing 1) to something incompatible is not acceptable, however that does not mean it's impossible to extend customizer support. It simply means we need a native way to specify customizer information in addition to the comment style. Some ideas for that can be found in the [wiki](https://github.com/openscad/openscad/wiki/Meta-Data-Use-Cases)

Improving the GUI is certainly a good thing, even just for the comment style customizer part there's lots of room for improvement, going to extended meta data would allow even more features.

50an6xy06r6n, 2021-06-20:
@t-paul this will not require changes to either the annotation syntax or the GUI. If you try to use a nested vector with the customizer it will successfully render in the text box UI and be saved to the JSON, but any changes made are always reverted to the file defaults because they can't be parsed. All I want to do is improve the parsing behavior.

50an6xy06r6n, 2021-06-22:
Oh I see that support for unconstrained vectors was actually removed in b374ae74e25a87ce442508b9b314b03b7e67ed2d. I've just been running 2021.01, so I guess this change wasn't in there. @redlizard do you have any thoughts on adding support back in a more robust way? I genuinely think this is a very useful feature, as it makes it much easier to generate more complex geometry based on the output of external programs, for one thing.

nobodyinperson, 2021-08-13:
This is definitely an important feature. For example when designing a part where features like holes or pins should be located at specific different locations depending on the parameter set - like a generic mount for electronics of different versions where the dimensions are equal but just the holes are slightly off.

Unfortunately Bountysource seems to be defunct, a lot of people are complaining and the site is unstable. If there is another way of posting bounties for OpenSCAD issues, I am willing to do so for this issue.

nobodyinperson, 2021-08-17:
For those really relying on loading arbitrarily nested vectors with the Customizer (or from JSON parameter files), it is still possible to reshape one-dimensional vectors (which the Customizer **can** handle as OP also points out).

This might already be implemented in some fancy library somewhere else, however I made two functions for this (available in [my OpenSCAD utils collection](https://gitlab.com/-/snippets/2147983)):

### `group()` - if the input vector has arbitrary length

```c
// a one-dimensional vector for an arbitrary amount of hole positions that is editable and loadable in the Customizer
hole_positions = [0,0,2,2,3,4];  // [-5:0.1:5]

// group every two elements in a vector together
hole_positions_ = group(vector_editable_in_customizer,2);
// result:
// hole_positions_ = [[0,0],[2,2],[3,4]];

// we can now loop over the hole positions:
for(pos = hole_positions_) translate(pos) cylinder();
```

### `reshape()` - if the shape is fixed

```c
// a one-dimensional vector representing a 2x2x2 matrix (fixed shape) that is still editable and loadable in the Customizer
matrix = [1,2,3,4,5,6,7,8]

// reshape one-dimensional vector to 2x2x2 matrix
matrix_ = reshape(matrix, [2,2,2])
// result: 
// matrix_ = [[[1, 2], [3, 4]], [[5, 6], [7, 8]]];
// elements can be accessed as desired: matrix_[0][0][0] == 1
```

For me this is enough for now to have the Customizer store and load nested vectors. Still, having the Customizer **load** nested vectors properly would be nice - **storing** them works fine.

nobodyinperson, 2021-08-20:
> Oh I see that support for unconstrained vectors was actually removed in b374ae7.

I just realized what @50an6xy06r6n means by that, namely that in the development snapshot of OpenSCAD not even one-dimensional unconstrained vectors (which work totally fine in OpenSCAD 2021.01) can be used in the Customizer anymore.

Reverting b374ae7 is not possible without conflicts, though they don't look all that complicated to resolve.

For the sake of backwards compatibility I would keep support at least for one-dimensional unconstrained vectors, not just drop it entirely. In my previous comment I provide a solution to still get nested functionality out of this.