openscad/openscad #3818: Feature Request: new transformation 'negative'

humanoid2050, 2021-06-18:
I'm not entirely sure if this proposal is compatible with OpenSCAD internals, but I think it should be based on the way OpenSCAD advertises it self as a scripting/compiling solution for 3D modeling.

Normally OpenSCAD behaves like there is a sort of implicit union of all top-level objects added to a project.  As such, I think it would be useful to be able to mark an object (or a collection of them from a module) as 'negative', causing it to be subtracted (like a difference operator) from all the positive objects constructed prior.

So instead of 

```
difference() {
  cube(5);
  cylinder(h=5,r=2);
}
```

you have

```
cube(5);
negative() cylinder(h=5,r=2);
```

The benefit is that you don't have to do ad nauseam  adjustment of nested unions and differences as you decided which shapes should be removed from which other shapes. Simply resequencing the positive and negative objects would be sufficient. In fact, I think you could do away with the difference and union operators completely by creating a scope operator from `{}`, implicitly unioning the contents of the scope, which would also serve to restrict what the 'negative' shapes can touch.

rcolyer, 2021-09-18:
> causing it to be subtracted (like a difference operator) from all the positive objects constructed prior.

I will note there is no notion of "prior" in OpenSCAD, so this would need to be "from all the positive objects constructed within that scope."  There is an obvious remapping to a difference that could be done to any explicit or implicit union with a negative() in so this would not require any deep changes in the OpenSCAD code except for one significant issue, which is children().  Currently, the behavior of children() is forced to be syntactic due to the #3143 issue, and the meaning of this in any scope that would have a negative() falls apart significantly into some chaos.

humanoid2050, 2021-09-19:
@rcolyer Thanks for taking the time to respond. After I made the original post, I did a little more digging and found a statement in the documentation that I should have realized sooner, which is that OpenSCAD is a functional language. My original comment was written from the (incorrect) perspective that it was an imperative language like C++ or Python and manipulating a managed collection of objects in memory. As such, I'm not sure that my comment really has value, but I thought I would leave it for discussion. Feel free to close if you think this is beyond what OpenSCAD can/will support.

jordanbrown0, 2025-03-09:
See BOSL2's `diff()` mechanism: https://github.com/BelfrySCAD/BOSL2/wiki/attachments.scad#module-diff