openscad/openscad #3820: Segfault when opening files

leonschoorl, 2021-06-20:
I've just installed OpenSCAD 2021.01 through snapcraft.
The "Welcome to OpenSCAD" dialog open just fine.
But as soon try to open a file or start a file new it crashed with a segmentation fault :(

The system log says:
```
jun 20 20:31:00 t470 audit[13129]: AVC apparmor="DENIED" operation="open" profile="snap.openscad.openscad" name="/proc/sys/dev/i915/perf_stream_paranoid" pid=13129 comm="openscad" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
jun 20 20:31:04 t470 kernel: show_signal_msg: 73 callbacks suppressed
jun 20 20:31:04 t470 kernel: openscad[13129]: segfault at 24 ip 00007f62d98e6c34 sp 00007ffd440d7e00 error 6 in iris_dri.so[7f62d8da3000+e54000]
```

The system is running Linux Mint 19.3 Tricia, which is based on Ubuntu 18.04.
And it does have Intel graphics, but not an Iris variant.  It has an 7200U CPU with HD620 graphics.

Is there anything I can do to (help) debug this?

leonschoorl, 2021-06-20:
Also the openscad-nightly (version 0+git.2b1d660ab) has the same problem

t-paul, 2021-06-20:
Looks like the graphics driver gets denied access to some part of the system. I don't see what OpenSCAD can do, that seems to be more a generic Snap / System issue.

M4rc3lv, 2021-06-26:
Related to 3523.
I have the same bug (reported December last year and I put a bounty on it).
Workaround is written in that issue:  https://github.com/openscad/openscad/issues/3523