openscad/openscad #3825: Spiro spline for better curve quality

stefano2734, 2021-06-28:
Circular Arcs are simple but not the best curves in nature and in technics.
Bezier Splines are limited in precision and mathematics.
spiro splines are Derivats of spirals.

See phd Levien

https://levien.com/phd/phd.html

spiro splines are implemented in Inkscape.

t-paul, 2021-06-30:
You've already commented that on the existing feature request #3607, that seems enough to track the pointer.

closed 2021-06-30 by t-paul

stefano2734, 2021-06-30:
There are two things 
1. Improvement of bezier splines in other issue:
In this PhD are some pages for better bezier splines 
See page 135 and more with error picture in page 141.

2. This issue is a new spiro spline from Levien:
The spiro spline is more than a better bezier spline.

See also https://www.levien.com/spiro/

with some code for his phd and his font program.
and for more clearness: 
https://raphlinus.github.io/curves/2018/12/21/new-spline.html

The code in Inkscape with the spiro path tool was improved in last years.
So Inkscape source code is here perhaps a better source with spiro splines also for this project.

nophead, 2021-06-30:
Just code it in OpenSCAD. Why does it need to be built in?

stefano2734, 2021-06-30:
Why is better better than good?
This is a core feature from my view as mechanical engineer.

jordanbrown0, 2021-06-30:
Write it in OpenSCAD and put that in MCAD, so it's available in a default installation.

Writing things in C++ is usually a great deal harder and riskier.

t-paul, 2021-06-30:
I believe I've looked at this some time ago already. It's certainly interesting but there's an important word at least in the title of the paper: "Optimal Techniques in **Interactive** Curve Design". The notion of "iteractive" and control points make it much hard to use in code CAD than interactive GUI tools. I've ended up looking at catmul-rom splines which work with minimal options, and importantly have the line go *through* all the points given.

Normal (quadratic and cubic) bezier support pretty much exists internally as it's needed for at least `text()` and SVG. Exposing this to the language is useful in my view. Still this is covered by the more general issue linked earlier.