openscad/openscad #3826: Simple Matthek triangle method for simple optimization of edges without FEA or CFD

stefano2734, 2021-06-28:
See 
http://bibliothek.fzk.de/mattheck/mattheck_postersammlung.pdf

and in Detail

https://publikationen.bibliothek.kit.edu/220061912/3814714

so an Optimization of edges is here simple possible.

Perhaps this is a good function for better real world models of openscad with real forces and real stress.

t-paul, 2021-06-30:
Sorry, this is a bit too unspecific to keep around as open issue.

closed 2021-06-30 by t-paul

stefano2734, 2021-06-30:
Look in the poster and you see the triangle method with 3 Triangles to construct spline points.
With this points a good edge curve is the result without FEM for beams, shells and solids or CFD for fluids.

there is a pdf Poster in german with stress Relation of matthek spline by triangles to circle arc . 
https://publikationen.bibliothek.kit.edu/220076489