openscad/openscad #3827: No details for camera debug: --debug=camera -> camera: Camera()

Irev-Dev, 2021-06-30:
I'm not sure what `-debug=camera` is supposed to do exactly, but I would have expected info about it's position etc. Is that correct?

Currently the output looks like the following:
```
Debug on. --debug=camera
Camera: Camera()
Compiling design (CSG Products normalization)...
Normalized CSG tree has 2 elements
Camera: Camera()
Camera: Camera()
Geometries in cache: 2
Geometry cache size in bytes: 3328
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0:00:00.413
```

i.e. `Camera: Camera()` seems wrong to me.

My environment is a dockerised ubuntu:20.04

Replication steps:
- clone https://github.com/Irev-Dev/cadhub
- checkout `kurt/openscad-camera-debug`
- `cd ./app/api/src/docker`
- `docker-compose up --build`
- open a shell into the openscad preview or stl container
- run `echo "cube(10);" | xvfb-run --auto-servernum --server-args "-screen 0 1024x768x24" openscad --debug=camera`

Should see
```
# echo "cube(10);" | xvfb-run --auto-servernum --server-args "-screen 0 1024x768x24" openscad --debug=camera
Debug on. --debug=camera
Camera: Camera()
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
```

thehans, 2021-06-30:
All the camera position data is shown in the status bar at the bottom.
The `--debug=`  option just enables any debug logging statements (`PRINTD` and `PRINTDB` mostly) for source filenames which match (case insensitive).

You can see them in the source code here:
https://github.com/openscad/openscad/blob/master/src/Camera.cc
There are currently only debug statements in the constructor, and for `viewAll` for Camera.

The name before the colon is not the class name, but the *filename*, without extension (pretty sure its intentionally shown that way, since it could come from matching `.h` or `.cc`, etc).

`PRINTD` and `PRINTDB` are macros defined here:
https://github.com/openscad/openscad/blob/32c5c18e1cac173b79fc37b85dca51f4e8d57e45/src/printutils.h#L107-L108
which basically just call out to `PRINTDEBUG` function defined here:
https://github.com/openscad/openscad/blob/32c5c18e1cac173b79fc37b85dca51f4e8d57e45/src/printutils.cc#L126-L140

Irev-Dev, 2021-07-01:
Thanks @thehans,

I think I follow, I was able to get it to put out some extra information using the `--viewall` flag. So I guess that means there's no way to consistently get camera information from the cli?

I have a niche usecase in that on [CadHub](https://cadhub.xyz/dev-ide/openScad) I'm using a three.js scene, and the scene's camera to requests openscad image renders from specific perspective using the `--camera` param. This works well, but I'd also like to be able to allow users to reset the view, or use the `viewall` flag, however unless I can get information back from openscad, then my three.js scene and the openscad one will become out of sync.

nophead, 2021-07-01:
I would also like to be able to get the camera distance set by --viewall on the command line so that I could know how big the scene is.

thehans, 2021-07-02:
Sorry, I glossed over your steps to reproduce, but @t-paul just showed me a link to the dev ide in IRC (moved to #openscad on irc.libera.chat btw) so I understand where you're coming from now.  Very cool app so far!

I don't think `--debug` is the best way to go about this, but perhaps a new flag such as `--console-status` could be added to replicate the status bar info to stdout.

Irev-Dev, 2021-07-02:
😁 All good @thehans. Glad you like it.

I'm hesitant to ask for new features since the CadHub usecase is very isolated, but hopefully it will help support the community, but also @nophead would be able to use `--console-status` as well.

My original question's been answered, should I close the issue, or maybe change the title into a feature request.

t-paul, 2021-07-03:
Yes, a new issue might be cleaner to collect what that output should contain initially. As people have often requested to also get bounding box info, that could be included too. I suppose it would help if this would easily be parsable while still being readable, so... JSON?

t-paul, 2021-09-12:
Closed via #3877.

closed 2021-09-12 by t-paul