openscad/openscad #3828: PNG of transparent object not correct

Parkinbot, 2021-07-02:
Using 2021.06.24.ci8242 on Windows 8.1 I get strange bitmap dumps in connection with transparency

`
color("white", 0.5)
cylinder(r=10, h=30, $fn = 60); 
`

[image]

Parkinbot, 2022-08-06:
Am I the only one having this effect? Animations with transperencies are not possible any more since early 2021 or so. 

`cylinder(r=60, h=100, center = true); `
`color("white",.3)rotate([90, 0]) cylinder(r=50, h=200, center = true); `

[image]

t-paul, 2022-08-06:
Looks normal on Linux with 2022.06.26.nightly (git ea8950b) using Ctrl+Shift+C

[image]

There's some changes sitting in #4261 which might help as it's moving the animation export to an offscreen buffer decoupling it from the viewport, e.g. allowing for different/higher resolutions too.

t-paul, 2022-08-06:
From #4002
> still happening with 2022.05.04 on Windows 10 Pro

jordanbrown0, 2024-11-07:
Still happening in 2024.10.28 on Windows 11.  It appears that a significant factor is that the transparency makes it through to the copied image.

I saw it with % objects.

```
color([0.9,0.9,0.9]) sphere(10);
translate([20,0,0]) %sphere(10);
```
System screen shot:
[image]

Copy viewport image:
[image]

Screen shot of c-v-i image pasted onto a white background (in Thunderbird):
[image]

And here's what I get if I paste that c-v-i image into a page with a green background (again, in Thunderbird) and then take a screen shot:
[image]

jordanbrown0, 2024-11-07:
That is, the copied image is **itself** transparent.