openscad/openscad #3829: Preview does not update when tab becomes active after closing other tab

gingerbeardman, 2021-07-02:
1. Open two documents each in a tab
2. Close the front most tab
3. Rearmost tab will become active

Expected
- preview should update to show current tab model

Actual
- preview shows model of tab that was just closed

t-paul, 2021-07-02:
There's no plan to change that behavior at this point. That might become a topic again if there's some kind of project (= multi-file) handling.

closed 2021-07-02 by t-paul

MichaelAtOz, 2021-07-02:
1. Open a main file which has `use <...>` or `include <...>` of a number of libraries.
2. Open said libraries in other tabs.

3. switch between libraries.

Preview should NOT be done.

Note, you can set `Design/Automatic-reload-and-preview` in the main file.
So when you `File/Save` a library, or tha main file, the Preview will update.

jordanbrown0, 2021-07-02:
People usually think of rerunning the preview on switching.  That's a problem if the program you're switching to is expensive to preview, or behaves pathologically. Might it be better to just save the results of the preview (if any) and switch *them* out when you switch tabs, rather than rerunning the preview processing?

gingerbeardman, 2021-07-03:
Thanks for feedback.

My workaround is to press Cmd+S to save every time I get to a new tab, even if there are no changes, as this refreshes the preview.

MichaelAtOz, 2021-07-03:
> My workaround is to press Cmd+S to save every time I get to a new tab, even if there are no changes, as this refreshes the preview.

That is a bug that came from some changes in Auto-reload&preview. #3539 & related.

gingerbeardman, 2021-07-04:
@MichaelAtOz then my request is to not fix this bug for now 😉

@jordanbrown0 exactly my thoughts, it's not that I want a new preview to render, it's just I want to see the preview as it last was when i viewed that specific tab. Saving/restoring the window contents bitmap is a great idea.

MichaelAtOz, 2021-07-04:
I did mention that this would be an issue when Tabs were introduced.
It is in the eye of the be-user.

It would be relatively simple to capture THE/A _photo-like picture_ of a tabs last previous-view.
But I think you would be instantly unhappy with that.

If you wish to have _fully 3D manipulative view with rotation and zoom_ of all tabs, that has overheads.
From my experience, you will not get too many tabs before you hit memory limits.

If you want multiple full-preview, you need to run multiple separate OpenSCAD processes.
Integrating that into a MDI/GUI is non trivial, particularly for multiple platforms.

So, yes, for now, Preview does not update. It was not designed to.

jordanbrown0, 2021-07-04:
Agree that it isn't just capturing the bitmap; it has to capture the model.  (I don't know whether that means capturing a CSG tree, or triangle soup, or what.)

But you don't need multiple instances to do this.  You can just open multiple windows, like in the pre-tab days.  That's what I always do.

gingerbeardman, 2021-07-04:
OK, then I would like to request that the preview be blanked when switching tabs.

That way it removes the visual confusion about which model I am editing and prompts me to do a new preview.

See #3834