openscad/openscad #3837: Segfault when destroying windows on exit

icecream95, 2021-07-11:
https://github.com/openscad/openscad/blob/cff035686cc9505d114115646e4d98edac307512/src/openscad.cc#L857

When `gui` deletes all windows on exit, it iterates through the windows using a range for loop, but this is unsafe because deleting a window can cause other items in the window list to be deleted.

I think the problem is that the iterator to the end of the window list is invalidated when items are removed from it, and so the iterator used by the range for hits the end of the set before it equals the end iterator.

Looking at the `QSet` documentation, I think this line could be rewritten like this (untested):

```c++
const auto &windows = scadApp->windowManager.getWindows();
while (!windows.empty())
        delete *windows.begin();
```

The segfault only seems to happen when using the quit menu option, either through File->Quit or Ctrl-Q, but not when just closing the window.

TheJKM, 2021-07-25:
There already is Merge Request #3784 covering this issue, however the solution I proposed there is rather brutal.
I just tested your solution and it works, so I will update the Merge Request. Thank you for your solution!

t-paul, 2021-07-25:
Closed via #3846. Thanks!

closed 2021-07-25 by t-paul